microsoft/AirSim #104: bash on windows arm-none-eabi-gcc --version Exec format error

eiselesr, 2017-03-13:
Hi, I'm trying to run with the SIL setup but I'm running into some trouble with bash on windows approach suggested here: https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md

Everything seems to be fine until the check step:
`arm-none-eabi-gcc --version`

The output is: 
`bash: /home/demo/gcc-arm-none-eabi-5_4-2016q2/bin/arm-none-eabi-gcc: cannot execute binary file: Exec format error`

Which seems correct since if I run 
`file arm-none-eabi-gcc`

I get: 
`arm-none-eabi-gcc: ELF 32-bit LSB  executable, Intel 80386, version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 2.6.8, stripped` 

suggesting that arm-none-eabi-gcc is a 32 bit file which is not compatible with bash on windows according to https://github.com/Microsoft/BashOnWindows/issues/390

So, what are my options?

Side note. Earlier I was trying to do this in ubuntu 16.04 and failing on the 
`sudo apt-get install gcc-4.6-base:i386` step, as that appears to be a ubuntu 14 library.
I'm not sure what question to ask, but I guess I want to know if someone has been able to run the SIL from ubuntu 16.04 or only 14.04.

lovettchris, 2017-03-13:
Ah, yes, sorry about that I forgot to mention arm-none-eabi-gcc does not work in BashOnWindows, but you shouldn't need that to build the SITL version of the code.    The reason you install all that is to keep the build system from complaining that you are missing those tools.

And change this "sudo apt-get install gcc-4.6-base:i386" to this "sudo apt-get install gcc-**5.4**-base:i386"

I'll update the docs.

lovettchris, 2017-03-13:
Docs are updated.

eiselesr, 2017-03-13:
Thanks!
Was the comment 
> And change this "sudo apt-get install gcc-4.6-base:i386" to this "sudo apt-get install gcc-5.4-base:i386"

in reference to installing on ubuntu 16.04?

lovettchris, 2017-03-25:
I've added new [PX4 build instructions](https://github.com/Microsoft/AirSim/blob/master/docs/px4.md).  Let me know if this is still not working, thanks.

closed 2017-03-25 by lovettchris

ShilinGuo520, 2017-05-06:
I have also encountered the same，I‘m try ‘sudo apt-get install gcc-arm-none-eabi’，but the ld Link problems , so I decision compiled from source.

lovettchris, 2017-05-08:
As I mentioned above, you do not need gcc-arm-none-eabi to build the SITL version of PX4.  BashOnWindows does not support 32 bit apps on a 64 bit machine, which is why it can't run the gcc-arm-none-eabi cross-compiler.  The BashOnWindows team said they have no plans to support 32 bit Linux apps.  But like I said, you don't need that cross compiler to build the SITL PX4 app.

SolinGuo, 2017-05-12:
I have compiled from source in bash on win 10 environment,Based on the latest source code（gcc-arm-none-eabi-5_4-2016q3-20160926）,The same environment should be available.
Link:
https://github.com/SolinGuo/arm-none-eabi-bash-on-win10-

lovettchris, 2017-05-15:
AWESOME !!  Thanks for doing that.  You rock!

romain-chiap, 2017-05-19:
Hi,

I extracted the file as given here: 
https://github.com/Microsoft/AirSim/blob/master/docs/px4.md 
but I don't know how to export the PATH=...

I tried: 
export PATH=/mnt/c/Users/Romain/Documents/McGill/Pixhawk/gcc-arm-none-eabi-5_4-2017q2/bin

But when I try to use the make command I got this error:
The command could not be located because '/usr/bin' is not included in the PATH environment variable.
make: command not found

I guess I messed up something with the PATH. I hope you can help

Romain

romain-chiap, 2017-05-19:
Hello,

I followed the steps here:
https://github.com/Microsoft/AirSim/blob/master/docs/px4.md

But I'm stuck at the last paragraph. How do I export PATH ? do I just type 
export PATH="myfolder/gcc-arm-none-eabi-5_4-2017q2/bin"
Could you explain it in more detail please ?

Thanks

Romain

lovettchris, 2017-05-19:
http://www.wikihow.com/Change-the-Path-Variable-in-Linux

romain-chiap, 2017-05-24:
OK for those who will need to do the same,
I used the line of code from ARM embedded tools section:
https://github.com/Microsoft/AirSim/blob/master/docs/px4.md 

I used:
exportline="export PATH=$HOME/gcc-arm-none-eabi-5_4-2017q2/bin:\$PATH"
if grep -Fxq "$exportline" ~/.profile; then echo nothing to do ; else echo $exportline >> ~/.profile; fi
. ~/.profile

(be careful it's 2017q2 here)

Then when checking with:
echo $PATH
it should appear in the list

Thanks to both of you ! it's building on BashOnWindows now 

Romain

lovettchris, 2017-05-25:
Great!