microsoft/AirSim #106: SIL: could not find conntected px4 flight conttroller

eiselesr, 2017-03-13:
Hi, 
I have the SITL version of PX4 running in Bash on Windowss

```
$ ./build_posix_sitl_default/src/firmware/posix/px4 ./posix-configs/SITL/init/lpe/iris
commands file: ./posix-configs/SITL/init/lpe/iris

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [dataman] Unkown restart, data manager file 'rootfs/fs/microsd/dataman' size is 11797680 bytes
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
```
However when I start the unreal project by pressing the play button I get the two errors:
`vehicle controller could not be started, please check your settings.json`
`Could not find a connected px4 flight controller`

The unreal project becomes unresponsive (i've been terminating it with the task manager) and no messages appear in the px4 console. My current settings.json is:

```
{
  "Pixhawk": {
    "LocalHostIp": "127.0.0.1",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}
```

Any hints as to what's going on, or is missing?

lovettchris, 2017-03-13:
Change "UseSerial" to false.

eiselesr, 2017-03-13:
That was it. Thanks!

One other unrelated question, if you have time. 
I'm controlling the quadcopter with an xbox controller, and after arming the quadcopter in QGroundControl I can control quadcopter in the unreal project. My question is about QGround Control, there is a warning log that is constantly speaking saying
```
Fail safe mode on
Fail safe mode off
manual control lost
position flight mode
```
and other warning log messages. Is this normal? In SIL mode can it be more stable? 

And a potential update for the READEME, it appears to be necessary to click the "arm" button in QGround Control to be able to control the quadcopter but I did not see that in the documentation.

lovettchris, 2017-03-14:
Yes, arming the drone is a required step, it carries over from the world of real drones where you want to be extra safe.  I'll add more info about that in the docs.  The [PX4 ](http://dev.px4.io) website will give you more info too.  Note: the PX4 app is also a command line interface, type "help" there to see available commands.  For example, you could do "commander arm" and "commander takeoff", etc.

But as for the QGroundControl warnings, this does indicate a problem.  I will do some debugging...  Perhaps QGroundControl  is not getting all the messages it needs in the SITL case because I forgot to join the second SITL MavLinkConnection on port 14556 to the QGroundControl  connection... I think I know the fix...

lovettchris, 2017-03-15:
Still working on it, I fixed one bug, which is that we needed two mavlink connections to control the SITL PX4 app.  But there's still one remaining bug, which is the local position is not getting initialized to 0,0,0 when you arm the drone, so local position is way off which explains why QGroundControl is twitchy, and why when you take off the drone flies away.  Still looking for a fix for that...

lovettchris, 2017-03-15:
Looks like it might be a PX4 bug: https://github.com/PX4/Firmware/issues/6823
You can try changing the default GPS location to Zurich to match PX4 default and see if that helps.
47.397742, 8.545594, 488m.
This is what jMavSim is using, and so it works ok with SITL...

sytelus, 2017-03-16:
Another fix Chris has found today:

> PX4 SITL mode needs to be configured to get the home location correct. Run the following in the PX4 console window so that the origin matches that which is setup in AirSim AVehiclePawnBase::HomeLatitude and HomeLongitude.
> param set LPE_LAT 47.641468
> param set LPE_LON -122.140165
>  
> Now close Unreal app, restart ./build_posix_sitl_default/src/firmware/posix/px4 and re-start the unreal app.

closed 2017-03-16 by sytelus

Timebutt, 2017-03-16:
Hi @sytelus, I'm not sure if I understand exactly what you are suggesting and I think others will struggle as well. Where exactly do I enter these commands? In the bash screen where my PX4 SITL is running, before starting an Unreal simulation, or in the same bash after starting a simulation in Unreal:

[image]

Both don't seem to work, as neither state seems to be waiting for a command to be entered ... 
Thanks for the clarification!

Timebutt, 2017-03-16:
Update: I have found in the [documentation](https://dev.px4.io/starting-building.html) on the PX4 website, that apparently a PX4 console application is installed if you install the Windows PX4 Toolchain. If somebody followed the SITL instructions, normally they won't have installed this console (I don't find it in my cloned repository in any case). Is there a way of doing it without this console, or do we have to install the PX4 toolchain?

lovettchris, 2017-03-16:
That screen shot you included above is the NuttX shell presented by the PX4 app that you are running in a BashOnWindows console.  So you don't need the windows PX4 toolchain.  IN that window above you can type in "help" to see a list of commands.  And you can type in this command "param set LPE_LAT 47.641468" and this command "param set LPE_LON -122.140165" then you may need to restart the PX4 app by typing Ctrl+C and repeat "./build_posix_sitl_default/src/firmware/posix/px4 ./posix-configs/SITL/init/lpe/iris".  Now the SITL mode drone should be happier with takeoff and other autonomous flying commands from QGroundControl and from DroneSHell.

Geekking, 2017-05-26:
Hi, @eiselesr . Where do you put the settings.json file?  I have install the PX4 SITL in another Ubuntu Mahines, and run the Block app in an windows machine, is it possible for me to control the drone on that ubuntu machine?  Thanks.

lovettchris, 2017-05-27:
The PX4 app has some simple commands like "commander arm" and "commander takeoff" type "?" to see them all.  But you can also run DroneShell on the windows side to control AirSim and you can now also control it using [Python](https://github.com/Microsoft/AirSim/blob/master/docs/python.md).