microsoft/AirSim #107: Linking error on MacOS

crubier, 2017-03-14:
Hello, and first thank you for this amazing piece of software. 

I know MacOS is not supported (yet). But I also know some people were able to make it work. 

However, I face this seemingly simple linking error when compiling the xcodeproj on MacOS, and I have not been able to resolve it:

```
Setting up Mono
Building MyProjectEditor...
2017-03-14 02:22:51.229 defaults[679:17007533] 
The domain/default pair of (com.apple.dt.Xcode, IDEBuildOperationMaxNumberOfConcurrentCompileTasks) does not exist
Running command : Engine/Binaries/DotNET/UnrealBuildTool.exe MyProjectEditor Mac Development /Users/vincent/Code/Unreal Projects/MyProject/MyProject.uproject
Compiling with MacOSX SDK 10.12
Parsing headers for MyProjectEditor
  Running UnrealHeaderTool "/Users/vincent/Code/Unreal Projects/MyProject/MyProject.uproject" "/Users/vincent/Code/Unreal Projects/MyProject/Intermediate/Build/Mac/MyProjectEditor/Development/MyProjectEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
Reflection code generated for MyProjectEditor in 4.050003 seconds
Performing 4 actions (8 in parallel)
[1/4] Compile Module.AirSim.cpp
[2/4] Compile AirSim.generated.cpp
[3/4] Link UE4Editor-AirSim.dylib
Undefined symbols for architecture x86_64:
  "common_utils::FileSystem::getUserDocumentsFolder()", referenced from:
      common_utils::FileSystem::getAppDataFolder() in Module.AirSim.cpp.o
  "msr::airlib::MavLinkDroneController::MavLinkDroneController()", referenced from:
      msr::airlib::Px4QuadX::createMavController(std::__1::unique_ptr<msr::airlib::DroneControllerBase, std::__1::default_delete<msr::airlib::DroneControllerBase> >&, msr::airlib::SensorCollection&) in Module.AirSim.cpp.o
  "common_utils::FileSystem::createDirectory(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >)", referenced from:
      common_utils::FileSystem::getLogFileNamePath(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, bool) in Module.AirSim.cpp.o
      common_utils::FileSystem::getAppDataFolder() in Module.AirSim.cpp.o
  "msr::airlib::MavLinkDroneController::initialize(msr::airlib::MavLinkDroneController::ConnectionInfo const&, msr::airlib::SensorCollection const*, bool)", referenced from:
      msr::airlib::Px4QuadX::createMavController(std::__1::unique_ptr<msr::airlib::DroneControllerBase, std::__1::default_delete<msr::airlib::DroneControllerBase> >&, msr::airlib::SensorCollection&) in Module.AirSim.cpp.o
  "msr::airlib::Settings::settings_", referenced from:
      MavMultiRotorConnector::startApiServer() in Module.AirSim.cpp.o
      ASimModeBase::BeginPlay() in Module.AirSim.cpp.o
      msr::airlib::Settings::loadJSonFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in Module.AirSim.cpp.o
      msr::airlib::Px4QuadX::getConnectionInfo() in Module.AirSim.cpp.o
  "msr::airlib::RpcLibServer::RpcLibServer(msr::airlib::DroneControllerCancelable*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, unsigned short)", referenced from:
      MavMultiRotorConnector::startApiServer() in Module.AirSim.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[4/4] Link UE4Editor-MyProject.dylib
clang: error: no such file or directory: '/Users/vincent/Code/Unreal Projects/MyProject/Plugins/AirSim/Binaries/Mac/UE4Editor-AirSim.dylib'
ERROR: UBT ERROR: Failed to produce item: /Users/vincent/Code/Unreal Projects/MyProject/Plugins/AirSim/Binaries/Mac/UE4Editor-AirSim.dylib
Total build time: 41.56 seconds
Command /Users/Shared/Epic Games/UE_4.14/Engine/Build/BatchFiles/Mac/Build.sh failed with exit code 5
```

Any pointers on what the solution could be ?

crubier, 2017-03-14:
This was when trying to compile on Commit https://github.com/Microsoft/AirSim/commit/0b31c00f56f6defb9f1b34b5575a3872b572224f

lovettchris, 2017-03-14:
Looks like it is not finding the AirLib library (which should have been built by cmake/AirLib/CmakeLists.txt).

The locating of that library is controlled by a file called AirSim.Build.cs (this is kind of like cmake for Unreal).  You should see a line like this:

AddLibDependency("AirLib", Path.Combine(AirSimPath, "lib"), "AirLib", Target, false);

I'm guessing this line of code is wrong for your platform:
        string PlatformString = (Target.Platform == UnrealTargetPlatform.Win64) ? "x64" : "x86";

Try printing the value of Target.Platform  so we can do the right thing here for OSX...  I'm guessing you want x64 here...  we probably should also throw an exception in this AddLibDependency method if the specified library can't be found...

crubier, 2017-03-14:
Ok, I added the following lines in `AirSim.build.cs`

```csharp
    private void SetupCompileMode(CompileMode mode, TargetInfo Target)
    {
        System.Console.WriteLine("==============================");
        string PlatformString = (Target.Platform == UnrealTargetPlatform.Win64) ? "x64" : "x86";
        System.Console.WriteLine(PlatformString);
        System.Console.WriteLine("==============================");
        switch (mode)
        {
```

This gave me the following result during compilation:

```
==============================
x86
==============================
```

crubier, 2017-03-14:
Ok thank you for your help @lovettchris , this helped me troubleshoot the few problems that were needed to fix this.

#109 fixes this Issue, I will let you review this.

commit f9ceea2f (2017-03-14): Try to fix #107
closed 2017-03-14 by lovettchris