microsoft/AirSim #108: Pixhawk not arming in Unreal

garethdicker, 2017-03-14:
Pixhawk running on HIL Quadrotor X, all set up radio etc, arming works.

Settings are standard:
[image]

But can't arm/control/kill in Unreal MyProject.

Only difference I can find from the video tutorials and what I've done is:
1) I'm running Windows 7
2) VS keeps telling me my project is out of date and to rebuild it. It then searches for libraries, seems to find them, and tells me "Ready"
[image]

any ideas? Let me know what kind of debugging info you need.

Thanks!

lovettchris, 2017-03-14:
Thanks for the detail you already provided.  Do you happen to have a PX4 FTDI system console cable handy?  Would be interesting to see what it is saying.  But if not, run this test:

MavLinkTest.exe -serial:*,115200

Let's see if this can find and chat with the Pixhawk.  Then when it says "ready" type in "arm" and see what it says.  If it cannot find a Pixhawk, it might be because the COM port works differently on Windows 7.  So you could also try the specific COM port number, like "COM6" instead of the asterix, so it would be:

MavLinkTest.exe -serial:COM6,115200

And then if that works, try doing the same in your settings.json. for the "SerialPort" setting, so it would read something like this:

    "SerialPort": "COM6",

Please attach the full output of MavLinkTest so I can debug it from there.

garethdicker, 2017-03-14:
Thanks for the quick reply, Chris!

I just had to put "SerialPort": "COM5" - all good.

closed 2017-03-14 by garethdicker

lovettchris, 2017-03-14:
Cool, glad it worked.  I'm curious to know what the VID/PID is of that device, since it means my auto-discovery code didn't work.  You can find that under the device Properties/Details/Device instance path... if you have time... thanks.