microsoft/AirSim #111: Unreal Editor crashes when hit play (Windows)

Alejolas, 2017-03-16:
Hi all,

Thanks a lot for this project. So I just bought a new build to run this properly, Intel I7 6800K, 32GB DDR4 RAM, GTX 1080. Installed everything following the documentation and videos, my problem is: every time I hit play, Unreal Editor crashes. When I debug it, here's what I get: 

[image]
[image]

Any help would be appreciated.

lovettchris, 2017-03-16:
Can you post the full call stack ?  Could be a setup issue.  Can you tell us more about your setup? which OS, Pixhawk hardware versus SITL mode, etc, thanks.

Alejolas, 2017-03-16:
Thanks for the reply Chris! 

I still don't have any Pixhawk hardware or drone... I just ordered the stuff from Amazon, I just got the computer, sorry if there some things in Spanish:

[image]
[image]
[image]
[image]
[image]
[image]

lovettchris, 2017-03-16:
Thanks, looks like it is crashing in the implementation of delete of a MavLinkNode which is not good.  But since you don't have pixhawk hardware, can you change your ~/Documents/AriSim/settings.json so that it has "Serial" set to "false".  Then I'd expect you would be looking at the [sitl instructions](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md) so you can get a flying drone.

lovettchris, 2017-03-16:
Note also that the master branch was just changed to switch to Unreal 4.15.  So if you have 4.14 you will need to upgrade.

Alejolas, 2017-03-16:
Oh, I don't have any controller of any kind, just keyboard and mouse. Should that be the problem?

lovettchris, 2017-03-16:
The [sitl instructions ](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md) show you how to get a drone that flies.  But we don't yet have support for keyboard/mouse flying.  But you could do some autonomous flying as showing in the [Unreal AirSim Setup Video ](https://youtu.be/1oY8Qu5maQQ).  Depends on what your goals are here.

Alejolas, 2017-03-16:
Yes, for the moment I just want to do autonomous fly while I get the other hardware, but first let's see if I can get pass through that MavLinkNode error that I'm having and crashing my Unreal Editor. I'm going to try again following all the steps. Thank you.

lovettchris, 2017-03-16:
Ok, let me know how it goes.  if you still run into trouble after following all the steps, then it might also help to enable the debugger to "break" on all C++ exception and see what you can find that way.  I'm guessing the delete is crashing because other setup code was skipped.  We have a bug there obviously, but we'll need to do some more debugging to track it down.

closed 2017-03-16 by sytelus

stvnmoon, 2017-03-19:
I'm having this issue also, only I do have HIL but get this stacktrace after running:
`>	UE4Editor-AirSim.dll!std::_LaunchPad<std::unique_ptr<std::tuple<void (__cdecl mavlinkcom_impl::MavLinkNodeImpl::*)(void) __ptr64,mavlinkcom_impl::MavLinkNodeImpl * __ptr64>,std::default_delete<std::tuple<void (__cdecl mavlinkcom_impl::MavLinkNodeImpl::*)(void) __ptr64,mavlinkcom_impl::MavLinkNodeImpl * __ptr64> > > >::_Run(std::_LaunchPad<std::unique_ptr<std::tuple<void (__cdecl mavlinkcom_impl::MavLinkNodeImpl::*)(void),mavlinkcom_impl::MavLinkNodeImpl *>,std::default_delete<std::tuple<void (__cdecl mavlinkcom_impl::MavLinkNodeImpl::*)(void),mavlinkcom_impl::MavLinkNodeImpl *> > > > * _Ln) Line 247	C++
`

MavLinkTest.exe works fine:

`PS C:\Users\steve\work\AirSim\x64\Release> .\MavLinkTest.exe -serial
Auto Selecting COM port: PX4 FMU (COM3)
Connecting to serial port COM3, baudrate=115200
isLocalControlSupported failed: Two second timeout waiting for response to mavlink command MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES
Downloading drone parameters so we know how to control it properly...
STATUS: sev=6, 'APM:Copter V3.4.6 (e707341b)'
STATUS: sev=6, 'PX4: de6b667d NuttX: 8c965992'
STATUS: sev=6, 'Frame: QUAD'
STATUS: sev=6, 'PX4v2 00350031 3432470E 31323533'
Ready...
### command 176 result: MAV_RESULT_UNSUPPORTED
System status:
    onboard sensors present:
    MAV_SYS_STATUS_SENSOR_DIFFERENTIAL_PRESSURE - 0x10 differential pressure
    MAV_SYS_STATUS_SENSOR_GPS - 0x20 GPS
    MAV_SYS_STATUS_SENSOR_VISION_POSITION - 0x80 computer vision position
    MAV_SYS_STATUS_SENSOR_RC_RECEIVER - 0x10000 rc receiver
    MAV_SYS_STATUS_SENSOR_3D_GYRO2 - 0x20000 2nd 3D gyro
    onboard sensors enabled:
    MAV_SYS_STATUS_SENSOR_ABSOLUTE_PRESSURE - 0x08 absolute pressure
    MAV_SYS_STATUS_SENSOR_DIFFERENTIAL_PRESSURE - 0x10 differential pressure
    MAV_SYS_STATUS_SENSOR_GPS - 0x20 GPS
    MAV_SYS_STATUS_SENSOR_OPTICAL_FLOW - 0x40 optical flow
    MAV_SYS_STATUS_SENSOR_VISION_POSITION - 0x80 computer vision position
    MAV_SYS_STATUS_SENSOR_EXTERNAL_GROUND_TRUTH - 0x200 external ground truth (Vicon or Leica)
    MAV_SYS_STATUS_GEOFENCE - 0x100000 geofence
    MAV_SYS_STATUS_TERRAIN - 0x400000 Terrain subsystem health
    CPU load 24388
    battery voltage 20037 millivolts
    battery current 20294 milliamps
    drop_rate_comm = 17234
    errors_comm = 69
    battery_remaining = 53
Connected:
    Version=3
    Type=MAV_TYPE_QUADROTOR-  Quadrotor
    Autopilot=MAV_AUTOPILOT_ARDUPILOTMEGA, ArduPilotMega / ArduCopter, http://diydrones.com
    State=MAV_STATE_STANDBY - System is grounded and on standby. It can be launched any time.
    Base mode:
    MAV_MODE_FLAG_STABILIZE_ENABLED - system stabilizes electronically its attitude (and optionally position). It needs however further control inputs to move around.
    MAV_MODE_FLAG_MANUAL_INPUT_ENABLED - remote control input is enabled.
    Custom mode=0
    VEHICLE SYSTEM ID: 1
    VEHICLE COMPONENT ID: 1
STATUS: sev=2, 'PreArm: Compass not calibrated'
STATUS: sev=2, 'PreArm: Accels not calibrated'
STATUS: sev=2, 'PreArm: Check FS_THR_VALUE'
STATUS: sev=2, 'PreArm: Throttle below Failsafe'`

I'm at a complete loss. I've stepped through until RpcLibServer.cpp line 116. Executing this crashes Unreal as described above with the stacktrace above.

`pimpl_->server.async_run(4);   //4 threads`

[image]

[image]

Any assistance would be appreciated. I have followed all instructions and cannot get it to run.

lkumar93, 2017-03-20:
I am getting the same issue but with HIL. I have my pixhawk connected to my PC in HIL mode . RC receiver is connected to the pixhawk.  In the SITL mode, I get an error saying " Computer load temporarily too high for real-time simulation". My Hardware looks something like this. 16gb RAM, Nvidia GTX 1060- 6GB GPU Memory, 128 GB SSD,. I am running the latest version of AirSim on Unreal 4.15 . Do you think 16GB RAM is the problem ?

lovettchris, 2017-03-22:
I am able to run smaller worlds with 16gb no problem.  What is the CPU ?  I've seen this message in SITL also "Computer load temporarily too high for real-time simulation" but I'm still able to fly, so I doubt this is the root issue.  

One thing with HIL mode is you must set this Airframe using QGroundControl then click "Apply and Restart"

[image]

lovettchris, 2017-03-22:
And for HIL mode your settings should look like this:

```
{
  "Pixhawk": {
    "LocalHostIp": "127.0.0.1",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}

```

lkumar93, 2017-03-22:
Thanks a lot for the reply. I am using Intel i7 6th gen processor. Also, pixhawk is on HIL mode. I haven't loaded any environment into unreal engine as I thought CPU load might get too high and my settings are the same as yours. And although I connect my Taranis Fr Sky to PC through USB. When I launch the project I am getting RC Controller not detected on USB

lovettchris, 2017-03-22:
Ok, great, i7 is fine.  That message is coming from AFlyingPawn::detectUsbRc.  Perhaps there are addition steps required to get your RC Controller to appear as a device that Unreal is happy with. There are some remote control instructions at the [bottom of this page](https://github.com/Microsoft/AirSim/blob/master/docs/prereq.md).

I believe we currently only support "xinput" devices (as per SimJoyStick.cpp) and there may be a missing step to convert your USB joystick into something that emulates xinput.  For example [x360ce](http://www.x360ce.com/) might be what you need there.

I also added [detailed new Unreal setup instructions](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_demo.md).

lkumar93, 2017-03-23:
I still get the same problem even after creating xinput.dll using x360ce.
Even thought the RC is not being detected by USB, it is still bound to the
receiver connected to pixhawk, so I dont think this is the problem for
unreal engine crash.

lovettchris, 2017-03-23:
hang on, why can't you connect your RC to your Pixhawk then ?  You do not need to use RC through USB to PC.  AirSim can get the RC controls via pixhawk as I show in [this video](https://www.youtube.com/watch?v=HNWdYrtw3f0&feature=youtu.be).

lkumar93, 2017-03-23:
Yes. As of now my RC is connected to the pixhawk directly. So the problem
must be something else.

lovettchris, 2017-03-23:
Ah, I see now, so the message about RC and USB is something you have to ignore in your case.  I agree it is confusing.  I will try and fix that so you don't get the confusing message.

But now I realize you haven't actually yet specified on this thread what you probably actually is.  Can you describe the symptoms you are seeing and the goal you are trying to reach ?

lkumar93, 2017-03-23:
I am getting the same error as seen in the snapshots of the 1st post of this thread. The GUI crashes after reading settings.json from the documents. The debugger says the crash occurs at MavLinkNodeImpl.

lovettchris, 2017-03-23:
I see thanks for clarifying.  Ok, I just made this [detailed unreal setup doc](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_demo.md) and followed every step there and it works fine, so this is indeed a mystery.  It would help me if you could paste the text of each C++ exception call stack you see before the crash while running the game in Visual Studio debugger.  Is that possible?  

Also, to ensure I never have any "stale bits" involved in my unreal projects I run this little script: 

```
pushd %~dp0
rd /s/q Binaries
rd /s/q Intermediate
rd /s/q Saved
rd /s/q Plugins\AirSim\Binaries
rd /s/q Plugins\AirSim\Intermediate
rd /s/q Plugins\AirSim\Saved
popd
```

then I update AirSim bits, and then I "regenerate the visual studio project", load it in VS and hit F5.

lkumar93, 2017-03-23:
Surprisingly it seems to be working right now. Previously I had launched the project file manually from unreal engine and that didn't work. This time I opened the solution file and put it in debug mode on Visual studio and then hit F5 , which launched the engine and then I was able to run it. However the drone is not flying as I increase the throttle after arming it. The gui says takeoff detected but It doesn't really move. The propellers are spinning though. Btw , the pixhawk is on altitude mode.

lovettchris, 2017-03-23:
Great, that's progress then.  Better to take off without Altitude mode - Altitude mode tries to hold the current altitude which is probably fighting the takeoff.  You also need to ensure Pixhawk is in HIL mode as I showed in the QGroundControl screen shot earlier in this thread.  Sometimes "rebooting" the pixhawk is necessary to reset any residual state it had from previous crashes or weird behavior.

CaoYongshengcys, 2019-01-12:
> The gui says takeoff detected but It doesn't really move. The propellers are spinning though

I came across the same problem,  takeoff detected but It doesn't really move. How do you fix it