microsoft/AirSim #112: Compilation errors in Linux using latest checkins

inkjet, 2017-03-16:
It appears that [this checkin](https://github.com/Microsoft/AirSim/commit/40eb4a1911620ec474d8e6835bcb702904c258aa#diff-ec3b8e3650c9ba6afdf77a6e534b956c) adding ROSFlight functionality is causing my AirSim compilation with Ubuntu 16.04 to fail.  I had success with previous AirSim versions.

Here is the compiler output -- I wonder if these were warnings on Windows, but errors using clang 3.8 on Linux:

```
[498/915] Compile Module.GeometryCache.cpp
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:15:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/MultiRotorConnector.cpp:3:
/home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/vehicles/configs/PX4ConfigCreator.hpp:29:16: error: 
      moving a local object in a return statement prevents copy elision [-Werror,-Wpessimizing-move]
        return std::move(config);
               ^
/home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/vehicles/configs/PX4ConfigCreator.hpp:29:16: note: 
      remove std::move call here
        return std::move(config);
               ^~~~~~~~~~      ~
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:15:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/MultiRotorConnector.cpp:4:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/vehicles/configs/RosFlightQuadX.hpp:8:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/RosFlightDroneController.hpp:13:
/home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/AirSimRosFlightBoard.hpp:105:19: error: 
      no matching conversion for functional-style cast from 'const char [50]' to 'std::exception'
            throw std::exception("cannot write motor output for index > motor count");
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ThirdParty/Linux/LibCxx/include/c++/v1/exception:91:29: note: candidate constructor (the implicit copy constructor) not
      viable: no known conversion from 'const char [50]' to 'const std::exception' for 1st argument
class _LIBCPP_EXCEPTION_ABI exception
                            ^
ThirdParty/Linux/LibCxx/include/c++/v1/exception:94:31: note: candidate constructor not viable: requires 0 arguments,
      but 1 was provided
    _LIBCPP_INLINE_VISIBILITY exception() _NOEXCEPT {}
                              ^
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:15:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/MultiRotorConnector.cpp:4:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/vehicles/configs/RosFlightQuadX.hpp:8:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/RosFlightDroneController.hpp:13:
/home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/AirSimRosFlightBoard.hpp:175:15: error: 
      no matching conversion for functional-style cast from 'const char [38]' to 'std::exception'
        throw std::exception("Diff pressure sensor is not available");
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ThirdParty/Linux/LibCxx/include/c++/v1/exception:91:29: note: candidate constructor (the implicit copy constructor) not
      viable: no known conversion from 'const char [38]' to 'const std::exception' for 1st argument
class _LIBCPP_EXCEPTION_ABI exception
                            ^
ThirdParty/Linux/LibCxx/include/c++/v1/exception:94:31: note: candidate constructor not viable: requires 0 arguments,
      but 1 was provided
    _LIBCPP_INLINE_VISIBILITY exception() _NOEXCEPT {}
                              ^
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:15:
```

sytelus, 2017-03-16:
Thanks for pointing this out. The warning was because of std::move. I've removed it and tested in Ubuntu 14.04 as well. Please get latest and try it out!

inkjet, 2017-03-16:
@sytelus Thank you for the quick response.  However, I'm still getting casting errors with the latest checkin when compiling the plugin with UE 4.13:

```
[495/909] Compile Module.AutomationWorker.cpp
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:15:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/MultiRotorConnector.cpp:4:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/vehicles/configs/RosFlightQuadX.hpp:8:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/RosFlightDroneController.hpp:13:
/home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/AirSimRosFlightBoard.hpp:105:19: error: 
      no matching conversion for functional-style cast from 'const char [50]' to 'std::exception'
            throw std::exception("cannot write motor output for index > motor count");
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ThirdParty/Linux/LibCxx/include/c++/v1/exception:91:29: note: candidate constructor (the implicit copy constructor) not
      viable: no known conversion from 'const char [50]' to 'const std::exception' for 1st argument
class _LIBCPP_EXCEPTION_ABI exception
                            ^
ThirdParty/Linux/LibCxx/include/c++/v1/exception:94:31: note: candidate constructor not viable: requires 0 arguments,
      but 1 was provided
    _LIBCPP_INLINE_VISIBILITY exception() _NOEXCEPT {}
```

and later on the errors get clipped due to the sheer volume:

```
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:15:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/MultiRotorConnector.cpp:4:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/vehicles/configs/RosFlightQuadX.hpp:8:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/RosFlightDroneController.hpp:17:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/firmware.hpp:8:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/estimator.hpp:7:
/home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/param.hpp:316:52: error: 
      ISO C++11 does not allow conversion from string literal to 'char *' [-Werror,-Wwritable-strings]
        init_param_int(static_cast<param_id_t>(i), "DEFAULT", 0);
                                                   ^
/home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/param.hpp:349:42: error: 
      ISO C++11 does not allow conversion from string literal to 'char *' [-Werror,-Wwritable-strings]
    init_param_int(PARAM_BOARD_REVISION, "BOARD_REV", 2); // Major board revision of naze32/flip32 | 1 | 6

/home/inkjet/Documents/AirSim/Unreal Projects/Quad/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/param.hpp:374:45: error: 
      ISO C++11 does not allow conversion from string literal to 'char *' [-Werror,-Wwritable-strings]
    init_param_float(PARAM_PID_ROLL_RATE_P, "PID_ROLL_RATE_P", 0.070f); // Roll Rate Proportional Gain | 0.0 | 1000.0

...
                                            ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
```

sytelus, 2017-03-16:
Above error is occurring in ROSFlight code that I recently integrated. It's because std::exception doesn't have constructor to take char*. If you replace std::exception by std::runtime_error, the error would go away. My linux m/c is toast right now so if you had like to make these changes and submit PR, I would be happy to take it. Alternatively, I can do these changes and push sometime next week.

inkjet, 2017-03-18:
@sytelus I updated the ROSIFlight .hpp files to use the `std::runtime _error`as you suggested, and that fixed the first set of errors.  However, compilation still fails because of a large amount of conversion errors in param.hpp.  Examples below -- any suggestions on how to fix these?

```
[402/759] Compile Module.InputBindingEditor.cpp
In file included from /home/srodkey/Documents/Valkyrie/Unreal Projects/Quad-new-airsim/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:15:
In file included from /home/srodkey/Documents/Valkyrie/Unreal Projects/Quad-new-airsim/Plugins/AirSim/Source/MultiRotorConnector.cpp:4:
In file included from /home/srodkey/Documents/Valkyrie/Unreal Projects/Quad-new-airsim/Plugins/AirSim/Source/AirLib/include/vehicles/configs/RosFlightQuadX.hpp:8:
In file included from /home/srodkey/Documents/Valkyrie/Unreal Projects/Quad-new-airsim/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/RosFlightDroneController.hpp:17:
In file included from /home/srodkey/Documents/Valkyrie/Unreal Projects/Quad-new-airsim/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/firmware.hpp:8:
In file included from /home/srodkey/Documents/Valkyrie/Unreal Projects/Quad-new-airsim/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/estimator.hpp:7:
/home/srodkey/Documents/Valkyrie/Unreal Projects/Quad-new-airsim/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/param.hpp:316:52: error: 
      ISO C++11 does not allow conversion from string literal to 'char *' [-Werror,-Wwritable-strings]
        init_param_int(static_cast<param_id_t>(i), "DEFAULT", 0);
                                                   ^
/home/srodkey/Documents/Valkyrie/Unreal Projects/Quad-new-airsim/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/param.hpp:349:42: error: 
      ISO C++11 does not allow conversion from string literal to 'char *' [-Werror,-Wwritable-strings]
    init_param_int(PARAM_BOARD_REVISION, "BOARD_REV", 2); // Major board revision of naze32/flip32 | 1 | 6
                                         ^
/home/srodkey/Documents/Valkyrie/Unreal Projects/Quad-new-airsim/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/param.hpp:350:37: error: 
      ISO C++11 does not allow conversion from string literal to 'char *' [-Werror,-Wwritable-strings]
    init_param_int(PARAM_BAUD_RATE, "BAUD_RATE", 921600); // Baud rate of communication with onboard computer | ...
```

sytelus, 2017-03-18:
Thanks for making progress! Basically, we need to turn off these warnings for GCC. To do this, open up StrictMode.hpp and then insert following line after line #32:

```
_Pragma("GCC diagnostic ignored \"-Wwritable-strings\"")                   \
```

As you might see, that section of code basically tells GCC which warnings to ignore. So we just add above new warnings in the existing list. This list of warnings gets turned off for selected sections in our code base, specifically any external code for which we don't have control. 

Let nail this down!

inkjet, 2017-03-19:
@sytelus We're getting close!  The `_Pragma` update eliminated the issue I previously mentioned. But now there is a new compilation error - some header files cannot be found.  This one is puzzling, since we hadn't previously seen those errors. Are files not being copied to the correct location?

The compiler output:

```
[406/759] Compile Module.ProceduralMeshComponent.cpp
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad-gcc-update/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:3:
/home/inkjet/Documents/AirSim/Unreal Projects/Quad-gcc-update/Plugins/AirSim/Source/AirBlueprintLib.cpp:7:10: fatal error: 
      'common/common_utils/Utils.hpp' file not found
#include "common/common_utils/Utils.hpp"
         ^
1 error generated.
[407/759] Compile Module.LocalizationService.cpp
[408/759] Compile Module.HTML5TargetPlatform.cpp
[409/759] Compile Module.ViewportSnapping.cpp
[410/759] Compile Module.Profiler.cpp
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad-gcc-update/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/AirSim.generated.cpp:10:
In file included from ../../../Unreal Projects/Quad-gcc-update/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/AirSim.generated.dep.h:15:
In file included from ../../../Unreal Projects/Quad-gcc-update/Plugins/AirSim/Source/VehiclePawnBase.h:3:
../../../Unreal Projects/Quad-gcc-update/Plugins/AirSim/Source/VehicleConnectorBase.h:3:10: fatal error: 
      'common/UpdatableObject.hpp' file not found
#include "common/UpdatableObject.hpp"
         ^
1 error generated.
```

sytelus, 2017-03-22:
My guess is that we probably have to also include directories in PublicIncludePaths? To try this,

Open AirSim.Build.Cs.
Insert below line after line #65:

`PublicIncludePaths.Add(Path.Combine(AirSimPath, "include"));`

inkjet, 2017-03-23:
@sytelus  I rolled in the update you suggested, and that fixed the header include issue, thanks.  However, there are a couple of new errors that pop up during compilation:

```
[495/908] Compile Module.GeometryCache.cpp
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad-newAirSimBuild/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:15:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad-newAirSimBuild/Plugins/AirSim/Source/MultiRotorConnector.cpp:4:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad-newAirSimBuild/Plugins/AirSim/Source/AirLib/include/vehicles/configs/RosFlightQuadX.hpp:8:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad-newAirSimBuild/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/RosFlightDroneController.hpp:17:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad-newAirSimBuild/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/firmware.hpp:12:
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad-newAirSimBuild/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/controller.hpp:9:
/home/inkjet/Documents/AirSim/Unreal Projects/Quad-newAirSimBuild/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/mixer.hpp:41:17: error: 
      extra qualification on member 'write_servo'
    void Mixer::write_servo(uint8_t index, int32_t value);
         ~~~~~~~^
In file included from /home/inkjet/Documents/AirSim/Unreal Projects/Quad-newAirSimBuild/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:15:
/home/inkjet/Documents/AirSim/Unreal Projects/Quad-newAirSimBuild/Plugins/AirSim/Source/MultiRotorConnector.cpp:24:15: error: 
      no matching conversion for functional-style cast from 'const char [63]' to 'std::exception'
        throw std::exception("ConfigType is not supported in MultiRotorConnector::initialize");
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ThirdParty/Linux/LibCxx/include/c++/v1/exception:91:29: note: candidate constructor (the implicit copy constructor) not
      viable: no known conversion from 'const char [63]' to 'const std::exception' for 1st argument
class _LIBCPP_EXCEPTION_ABI exception
                            ^
ThirdParty/Linux/LibCxx/include/c++/v1/exception:94:31: note: candidate constructor not viable: requires 0 arguments,
      but 1 was provided
    _LIBCPP_INLINE_VISIBILITY exception() _NOEXCEPT {}
                              ^
[496/908] Compile Niagara.generated.cpp
2 errors generated.
```

The second error seems like it can be fixed by changing `std::exception` to `std::runtime_error` as you had previously suggested.  However, I'm unsure how to proceed the error related to the `write_servo` function.

madratman, 2017-04-01:
@inkjet @sytelus I was facing same issues on Linux. 
@inkjet There are basically 3 issues :
- replace by `std::exception` by `std::runtime_error` wherever the compiler cries, which you seem to have already done
- the extra qualification error you have in the previous comment can be simply fixed by removing the `Mixer::` in the `write_servo` line. It's a small bug of multiple scoping inside the class itself.

Your lines 39-41 here (https://github.com/Microsoft/AirSim/blob/40eb4a1911620ec474d8e6835bcb702904c258aa/AirLib/include/controllers/rosflight/firmware/mixer.hpp#L41) should be just :

```
private:
    void write_motor(uint8_t index, int32_t value);
    void write_servo(uint8_t index, int32_t value);
```

- Then they are multiple errors of `ISO C++11 does not allow conversion from string literal to 'char *'` inside `AirLib/include/controllers/rosflight/RosFlightDroneController.hpp` as you mentioned. I think typecasting to `char *` manually should work, 

You can see the changes I made here in my fork https://github.com/madratman/AirSim/commit/d6955ea642940f2ef769fc6d7c468bfb538efe4c#diff-85b48f089b02420d5c806ab6c433521e

@sytelus  if this looks okay, I can make a PR.
However, this didn't end my woes of making it run on UE in Linux, it almost begins to run, when in the end, it cries:

```
[2017.04.01-02.02.13:320][  0]LogHAL: Linux SourceCodeAccessSettings: NullSourceCodeAccessor
[2017.04.01-02.02.15:885][  0]LogUdpMessaging: Initializing bridge on interface 0.0.0.0:0 to multicast group 230.0.0.1:6666.
[2017.04.01-02.02.16:907][  0]LogLinux:Warning: dlopen failed: /home/vaibhav/Documents/Unreal_Projects/airsim_ue/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so: undefined symbol: _ZN9rosflight7Sensors7gravityE
[2017.04.01-02.02.16:907][  0]LogModuleManager:Warning: ModuleManager: Unable to load module '/home/vaibhav/Documents/Unreal_Projects/airsim_ue/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so' because the file couldn't be loaded by the OS.
[2017.04.01-02.02.25:956][  0]LogExit: Preparing to exit.
[2017.04.01-02.02.27:069][  0]LogObj: Freed 0b from 0 cluster array pools.
[2017.04.01-02.02.27:069][  0]LogExit: Object subsystem successfully closed.
[2017.04.01-02.02.27:107][  0]LogModuleManager: Shutting down and abandoning module HotReload (181)
```

lovettchris, 2017-04-01:
Thanks, I picked up your fixes already, no need for a pull request.  I also removed the static from the rosflight sensors gravity field which might solve the undefined symbol: _ZN9rosflight7Sensors7gravityE.

madratman, 2017-04-01:
Thanks! Ah, I made around a minute before you commented anyway :D 
I ll try this now.
Also #135  seems related. I am not sure what exaclty he changed. Let's ask there

inkjet, 2017-04-01:
Thanks for keeping this thread alive!  I'll also try the latest check-in now.

madratman, 2017-04-01:
@inkjet no problem! let me know how it goes as I am facing issues in UE now:
@lovettchris, I was able to get it running in UE after a few more removals of static constexpr as can be seen in https://github.com/Microsoft/AirSim/pull/140. 
Switched to game mode in UE -> no issues
If I try to remove Player positions -> UE crashes
If I try to click "Play" (with or without SimGameMode) -> UE crashes.

I am able to "Play" the Landscape Mountain environments, however I see these relevant error and warning both when I click Play on Landscape Mountain, or with the AirSim plugin :

```
[2017.04.01-03.22.33:500][ 44]LogBlueprintUserMessages: Early PlayInEditor Detection: Level '/Game/Maps/LandscapeMap.LandscapeMap:PersistentLevel' has LevelScriptBlueprint '/Game/Maps/LandscapeMap.LandscapeMap:PersistentLevel.LandscapeMap' with GeneratedClass '/Game/Maps/LandscapeMap.LandscapeMap_C' with ClassGeneratedBy '/Game/Maps/LandscapeMap.LandscapeMap:PersistentLevel.LandscapeMap'
[2017.04.01-03.22.33:500][ 44]BlueprintLog: New page: Pre-Play auto-recompile
[2017.04.01-03.22.33:501][ 44]LogPlayLevel: [PlayLevel] Compiling LandscapeMap before play...
[2017.04.01-03.22.35:901][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error Spawn node  SpawnActor NONE  must have a  Class  specified.
[2017.04.01-03.22.35:901][ 44]LogBlueprint:Warning: [Compiler PersistentLevel.LandscapeMap] Warning Could not find a variable named "bIsOutOfBounds" in 'LandscapeMap'.
Make sure 'LandscapeMap' has been compiled for  Set bIsOutOfBounds
[2017.04.01-03.22.35:901][ 44]LogBlueprint:Warning: [Compiler PersistentLevel.LandscapeMap] Warning Can't connect pins  Return Value  and  Object : This cast has an invalid target type (was the class deleted without a redirect?).
[2017.04.01-03.22.35:901][ 44]LogBlueprint:Warning: [Compiler PersistentLevel.LandscapeMap] Warning Can't connect pins  Object  and  Return Value : This cast has an invalid target type (was the class deleted without a redirect?).
[2017.04.01-03.22.35:901][ 44]LogBlueprint:Warning: [Compiler PersistentLevel.LandscapeMap] Warning Could not find a variable named "bIsOutOfBounds" in 'LandscapeMap'.
Make sure 'LandscapeMap' has been compiled for  Set bIsOutOfBounds
[2017.04.01-03.22.35:901][ 44]LogBlueprint:Warning: [Compiler PersistentLevel.LandscapeMap] Warning Can't connect pins  Return Value  and  Object : This cast has an invalid target type (was the class deleted without a redirect?).
[2017.04.01-03.22.35:902][ 44]LogBlueprint:Warning: [Compiler PersistentLevel.LandscapeMap] Warning Can't connect pins  Object  and  Return Value : This cast has an invalid target type (was the class deleted without a redirect?).
[2017.04.01-03.22.35:902][ 44]LogBlueprint:Warning: [Compiler PersistentLevel.LandscapeMap] Warning Can't connect pins  Object  and  Return Value : This cast has an invalid target type (was the class deleted without a redirect?).
[2017.04.01-03.22.35:902][ 44]LogBlueprint:Warning: [Compiler PersistentLevel.LandscapeMap] Warning Can't connect pins  Return Value  and  Object : This cast has an invalid target type (was the class deleted without a redirect?).
[2017.04.01-03.22.35:903][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error This blueprint (self) is not compatible with ' Target ', therefore that pin must have a connection.
[2017.04.01-03.22.35:903][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error This blueprint (self) is not compatible with ' Target ', therefore that pin must have a connection.
[2017.04.01-03.22.35:903][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error This blueprint (self) is not compatible with ' Target ', therefore that pin must have a connection.
[2017.04.01-03.22.35:903][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error This blueprint (self) is not compatible with ' Target ', therefore that pin must have a connection.
[2017.04.01-03.22.35:903][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error This blueprint (self) is not compatible with ' Target ', therefore that pin must have a connection.
[2017.04.01-03.22.35:904][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error  Using VR  : unable to determine expected signature - is the delegate pin connected?
[2017.04.01-03.22.35:904][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error Event Dispatcher has no property  Bind Event to Using VR
[2017.04.01-03.22.35:904][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error  Respawn player_Event  : unable to determine expected signature - is the delegate pin connected?
[2017.04.01-03.22.35:904][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error Event Dispatcher has no property  Bind Event to Respawn player
[2017.04.01-03.22.35:904][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error The property associated with  Is Out Of Bounds  could not be found
[2017.04.01-03.22.35:904][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error Event Dispatcher has no property  Call Respawn player
[2017.04.01-03.22.35:904][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error The property associated with  Is Out Of Bounds  could not be found
[2017.04.01-03.22.35:905][ 44]LogBlueprint:Error: [Compiler PersistentLevel.LandscapeMap] Error Failed to create property K2Node_CreateDelegate_OutputDelegate from  <None>  due to a bad or unknown type (Dele [... cut ...]
[2017.04.01-03.23.35:081][144]LogALAudio: Closing ALAudio device : OpenAL Soft
^C[2017.04.01-03.23.53:159][396]LogLinux: FLinuxPlatformMisc::RequestExit(bForce=false, ReturnCode=130)
[2017.04.01-03.23.53:159][396]LogLinux: FLinuxPlatformMisc::RequestExit(0)
[2017.04.01-03.23.53:159][396]LogGenericPlatformMisc: FPlatformMisc::RequestExit(0)
[2017.04.01-03.23.53:396][397]LogALAudio: Closing ALAudio device : OpenAL Soft
[2017.04.01-03.23.53:671][397]LogExit: Preparing to exit.
^C[2017.04.01-03.23.53:773][397]LogLinux: FLinuxPlatformMisc::RequestExit(1)

```

madratman, 2017-04-01:
I get similar compiler errors and warnings when running with or without AirSim. 

Without AirSim : https://www.youtube.com/watch?v=bTas8PQ3KzI&feature=youtu.be
With AirSim: https://www.youtube.com/watch?v=gXtWDvugrzU&feature=youtu.be

Log without AirSim:https://gist.github.com/madratman/60a445535f20985f97f2a35f6d9ba3a6
Log with AirSim: https://gist.github.com/madratman/e6e3bac7d6b3bc32ea04fd1fad6caa5f#file-airsim_ue

madratman, 2017-04-05:
@lovettchris sorry for bugging again, any ideas about why this is happening? I don't have a clue on how to  to debug this.

AlexandreBorowczyk, 2017-04-07:
@inkjet Did the compilation worked for you with the latest version? 

I'm still getting the following errors
```
/home/inkjet/Documents/AirSim/Unreal Projects/Quad-new-airsim/Plugins/AirSim/Source/AirLib/include/controllers/rosflight/firmware/param.hpp:316:52: error: 
      ISO C++11 does not allow conversion from string literal to 'char *' [-Werror,-Wwritable-strings]
        init_param_int(static_cast<param_id_t>(i), "DEFAULT", 0);
```
Do we need to add the (char*) casting as did @madratman ?

madratman, 2017-04-08:
@a2 haven't looked at the latest commits, but you can try this branch in my fork https://github.com/madratman/AirSim/tree/gigabyte_working or better just check the diffs for `char *` here - https://github.com/Microsoft/AirSim/pull/140/files and try again? 
Also, if you get it working, can you please share if you're getting the crash as shown in the video and logs in my previous comment? Thanks!

AlexandreBorowczyk, 2017-04-09:
@madratman I've managed to get everything compiling. I don't have the compilation issue but it seems I am now stuck at the same place as you; UE4Editor crash when I press the play button. One additional thing I've noticed if I switch to SimGameMode and try to save the project, it crashes.

`[2017.04.09-19.11.41:694][945]LogLinux: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000002

[2017.04.09-19.11.41:694][945]LogLinux: Fatal error!

[Callstack]  00  0x00007fd7775738bf  FLinuxPlatformStackWalk::CaptureStackBackTrace(unsigned long long*, unsigned int, void*)
[Callstack]  01  0x00007fd77742dc95  FGenericPlatformStackWalk::StackWalkAndDump(char*, unsigned long, int, void*)
[Callstack]  02  0x00007fd77752b671  FLinuxCrashContext::CaptureStackTrace()
[Callstack]  03  0x00007fd76c574340  CommonLinuxCrashHandler(FGenericCrashContext const&)
[Callstack]  04  0x00007fd777530572  PlatformCrashHandler(int, siginfo_t*, void*)
[Callstack]  05  0x00007fd77d357390  /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7fd77d357390]
[Callstack]  06  0x00007fd7774bf768  FMallocBinned::Private::FreeInternal(FMallocBinned&, void*)
[Callstack]  07  0x00007fd704d5ec3a  std::__1::basic_filebuf<char, std::__1::char_traits<char> >::~basic_filebuf()
[Callstack]  08  0x00007fd704d59131  ASimModeWorldBase::~ASimModeWorldBase()
[Callstack]  09  0x00007fd776bfe1a6  IncrementalPurgeGarbage(bool, float)
[Callstack]  10  0x00007fd776c0121e  CollectGarbageInternal(EObjectFlags, bool)
[Callstack]  11  0x00007fd776c02227  CollectGarbage(EObjectFlags, bool)
[Callstack]  12  0x00007fd76da029f3  UEditorEngine::OnPreSaveWorld(unsigned int, UWorld*)
[Callstack]  13  0x00007fd76da01466  UEditorEngine::Save(UPackage*, UObject*, EObjectFlags, wchar_t const*, FOutputDevice*, FLinkerLoad*, bool, bool, unsigned int, ITargetPlatform const*, FDateTime const&, bool)
[Callstack]  14  0x00007fd76da026b2  UEditorEngine::SavePackage(UPackage*, UObject*, EObjectFlags, wchar_t const*, FOutputDevice*, FLinkerLoad*, bool, bool, unsigned int, ITargetPlatform const*, FDateTime const&, bool)
[Callstack]  15  0x00007fd76da8466f  UEditorEngine::Exec_Obj(wchar_t const*, FOutputDevice&)
[Callstack]  16  0x00007fd76da8b07c  UEditorEngine::Exec(UWorld*, wchar_t const*, FOutputDevice&)
[Callstack]  17  0x00007fd76e49a9b2  UUnrealEdEngine::Exec(UWorld*, wchar_t const*, FOutputDevice&)
[Callstack]  18  0x00007fd76dd5a26c  /home/aborowczyk/Documents/epicgames_ws/UnrealEngine/Engine/Binaries/Linux/libUE4Editor-UnrealEd.so(+0x125c26c) [0x7fd76dd5a26c]
[Callstack]  19  0x00007fd76dd5d7e9  FEditorFileUtils::SaveLevel(ULevel*, FString const&, FString*)
[Callstack]  20  0x00007fd76dd5f582  FEditorFileUtils::SaveCurrentLevel()
[Callstack]  21  0x0000000000440ec4  TBaseStaticDelegateInstance<void ()>::ExecuteIfSafe() const [Runtime/Core/Public/Delegates/DelegateInstancesImpl.h, line 1027]
[Callstack]  22  0x00007fd7703641c2  FUICommandList::ExecuteAction(TSharedRef<FUICommandInfo const, (ESPMode)0>) const
[Callstack]  23  0x00007fd77044af42  SToolBarButtonBlock::OnClicked()
[Callstack]  24  0x00007fd7704c5e6e  TBaseSPMethodDelegateInstance<false, SToolBarButtonBlock, (ESPMode)0, FReply ()>::Execute() const
[Callstack]  25  0x00007fd7704fbf93  SButton::OnMouseButtonUp(FGeometry const&, FPointerEvent const&)
[Callstack]  26  0x00007fd770342172  FSlateApplication::RoutePointerUpEvent(FWidgetPath&, FPointerEvent&)
[Callstack]  27  0x00007fd770302699  FSlateApplication::ProcessMouseButtonUpEvent(FPointerEvent&)
[Callstack]  28  0x00007fd770348d32  FSlateApplication::OnMouseUp(EMouseButtons::Type, FVector2D)
[Callstack]  29  0x00007fd77751bf1a  FLinuxApplication::ProcessDeferredMessage(SDL_Event)
[Callstack]  30  0x00007fd77751b78d  FLinuxApplication::AddPendingEvent(SDL_Event)
[Callstack]  31  0x00007fd77751b65c  FLinuxPlatformMisc::PumpMessages(bool)
[Callstack]  32  0x00000000004232f0  FEngineLoop::Tick() [Runtime/Core/Public/UObject/NameTypes.h, line 638]
[Callstack]  33  0x000000000042b923  GuardedMain(wchar_t const*) [/home/aborowczyk/Documents/epicgames_ws/UnrealEngine/Engine/Source/Runtime/Launch/Private/Launch.cpp, line 62]
[Callstack]  34  0x00007fd76c5750ca  CommonLinuxMain(int, char**, int (*)(wchar_t const*))
[Callstack]  35  0x00007fd76b8ed830  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7fd76b8ed830]
[Callstack]  36  0x0000000000415e29  /home/aborowczyk/Documents/epicgames_ws/UnrealEngine/Engine/Binaries/Linux/UE4Editor(_start+0x29) [0x415e29]

`

inkjet, 2017-04-10:
Good news, bad news:

@AlexandreBorowczyk Using the [latest checkin](https://github.com/Microsoft/AirSim/commit/97d17ab25a3b87b4d7141e6841548313eb9d2f77) (as of April 7th), I was able to get the plugin to compile and `libUE4Editor-AirSim.so` to successfully link with my project.  So for me, nothing breaks.  I'm using Ubuntu 16.04, clang 3.8, and the libc++ 3.9 libraries.

@lovettchris The odd part is that when I launch AirSim using `SimGameMode`, I get a blank screen in the editor window.  I can quit out of everything and get back to my environment (LandscapeMountains), but can't seem to successfully start the sim.  Anyone else experience this?

The console output says everything is loading correctly, but could this be related to the warning about SpawnActor?

<pre><code>
[2017.04.10-14.17.26:970][936]LogLoad: Game class is 'SimGameMode_C'
[2017.04.10-14.17.27:044][936]LogWorld: Bringing World /Game/Maps/UEDPIE_0_LandscapeMap.LandscapeMap up for play (max tick rate 0) at 2017.04.10-10.17.27
[2017.04.10-14.17.27:071][936]LogWorld: Bringing up level for play took: 0.099912
<b>[2017.04.10-14.17.27:071][936]LogSpawn:Warning: SpawnActor failed because no class was specified</b>
[2017.04.10-14.17.27:071][936]LogGameMode:Display: Match State Changed from EnteringMap to WaitingToStart
[2017.04.10-14.17.27:072][936]LogGameState: Match State Changed from EnteringMap to WaitingToStart
[2017.04.10-14.17.27:072][936]LogGameMode:Display: Match State Changed from WaitingToStart to InProgress
[2017.04.10-14.17.27:079][936]LogGameState: Match State Changed from WaitingToStart to InProgress
</code></pre>

lovettchris, 2017-04-10:
Are you also using Unreal 4.15 ?  When I launch Landscape Mountains the log says this, and the picture comes up fine:

LogLoad: Game class is 'SimGameMode_C'
LogWorld: Bringing World /Game/Maps/UEDPIE_0_LandscapeMap.LandscapeMap up for play (max tick rate 0) at 2017.04.10-11.39.48
LogWorld: Bringing up level for play took: 0.127557
LogContentBrowser: Native class hierarchy updated for 'MovieSceneCapture' in 0.0018 seconds. Added 11 classes and 0 folders.
LogGameMode:Display: Match State Changed from EnteringMap to WaitingToStart
LogGameState: Match State Changed from EnteringMap to WaitingToStart
LogGameMode:Display: Match State Changed from WaitingToStart to InProgress
Cmd: r.CustomDepth 3
r.CustomDepth = "3"
LogGameState: Match State Changed from WaitingToStart to InProgress
PIE: Info Play in editor start time for /Game/Maps/UEDPIE_0_LandscapeMap 0.248
LogBlueprintUserMessages: Late PlayInEditor Detection: Level '/Game/Maps/LandscapeMap.LandscapeMap:PersistentLevel' has LevelScriptBlueprint '/Game/Maps/LandscapeMap.LandscapeMap:PersistentLevel.LandscapeMap' with GeneratedClass '/Game/Maps/LandscapeMap.LandscapeMap_C' with ClassGeneratedBy '/Game/Maps/LandscapeMap.LandscapeMap:PersistentLevel.LandscapeMap'
LogRenderer: Reallocating scene render targets to support 1280x720 NumSamples 1 (Frame:5248).

Perhaps you should check that your  map is not corrupted?  My Content\Maps folder contains this:

[image]

inkjet, 2017-04-12:
@lovettchris I'm using Unreal 4.14.3, as 4.15 wasn't recommended for the first builds of AirSim.  Would this issue be related to #155?   I will try with 4.15 later today.

Also, my Content/Maps folder contains a single file:

[image]

The map works for the "Hanglider" game mode, so it appears to be valid.

madratman, 2017-04-12:
I have 4.15 and have both files with similar sizes. and yes, handglider / without airsim works

AlexandreBorowczyk, 2017-04-12:
Same, I have 4.15. However, I choose to use a basic C++ project.

inkjet, 2017-04-13:
@lovettchris Confirmed that the project crashes when I select "SimGameMode" and press play using Unreal Editor 4.15.1.

blaind, 2017-04-15:
Same here, tested on both 4.15.0 and 4.15.1. I compiled airsim in docker container though (with newer libc++), and run it on host with libc++ 3.7.X, not sure if this could be the cause.

EliaTarasov, 2017-04-16:
Has anybody made any progress so far on this issue?

AlexandreBorowczyk, 2017-04-16:
I've used the QtCreator Debugger to try to get more insight into the problem. I'm posting a screenshot of here it seems catch in the plugin.
[image]

EliaTarasov, 2017-04-16:
I have the off-topic question but: how did you manage to compile your test project with Qt at all?
The generated project *.pro file using qmake as a build system, so it means you have to compile Qt Framework with clang (to create qmake) to be able to select Qt Kits. AFAIK there is no default installation for Qt with LLVM.
Does it mean you build Qt with clang support from the source?

lovettchris, 2017-04-16:
He's just using the QT debugger, didn't say he compiled the code with QT.  

Alexandre, thanks for the stack trace, I've been studying the ASimModeBase class defined in SimModeBase.?pp and it has these private members:

    bool is_recording;
    std::ofstream record_file;
    int record_tick_count;

So the crash appears to be related to this "record_file", but this file is not opened by default, it should be in the default state, destructing it should be simple.  You could double check that the startRecording meethod is never called.   So I'm guessing there is some other memory corruption problem going on that just happens to show up at this time.    

I have an idea on something that could be helpful.  Can you add a ASimModeBase::~ASimModeBase descructor and put a breakpoint on it and take a look at the members of this class and see if they look ok.  Also put a breakpoint on the constructor and see how many times it gets called.  This is what I see on Windows, notice the record_file has a null buffer:

[image]

lovettchris, 2017-04-16:
And since this is in standard library code the other thing to check is how did you get the libcxx on your machine?  Did you use the ~/cmake/getlibcxx.sh script and build it locally using your clang ?

EliaTarasov, 2017-04-16:
@lovettchris Yes, i understand that he uses Qt debugger, the question was how did he setup Qt project and how was he able to launch it? I watched this video https://www.youtube.com/watch?v=RnINgMbJx5Q
but there the Qt project was already being setup (i didn't understand how exactly) with Clang Kit that uses qmake. And that was exaclty my question.
Sorry for that, that's a little off-topic question.

lovettchris, 2017-04-16:
can we move that to a separate issue so I can focus on the crash problem here?  Let's call it "Debugging AirSim on Linux" and get input from multiple folks on the best way to do that.

EliaTarasov, 2017-04-16:
Done.

lovettchris, 2017-04-17:
This might be the same issue as issue #144.  I checked in a possible workaround for the crash.  See [commit 43baa4f](https://github.com/Microsoft/AirSim/commit/43baa4f8f514c09fd0b38ff484006d4d408daf74).
This disables the RPC server via settings.json.  This seems to to be a workaround for the crash.

madratman, 2017-04-17:
@lovettchris. Just tried with 43baa4f, same results. I have the log [here](https://gist.github.com/madratman/0a3a8ffffcea2706fd0a0f043075e667)

madratman, 2017-04-17:
@lovettchris. thanks for looking into this again. Just tried with 43baa4f, same result - it crashes. I have the log [here](https://gist.github.com/madratman/0a3a8ffffcea2706fd0a0f043075e667) if it helps.

clovett, 2017-04-17:
Ok, rats, this fixed a similar crash on Windows, do I was hoping it would help.  Thanks for trying it out.

AlexandreBorowczyk, 2017-04-18:
@lovettchris I've gave a quick try at setting breakpoints in ASimModeBase's constructor and destructor. 
It seems the constructor is called 3 time when I open the project. Moreover, it's called again when I switch the gamemode to SimGameMode. However at this point my PC freeze and I can go further. I've tried by looking at the destructor only, when it's called record_file is not NULL. I will try to dig deeper tomorrow. As for libcxx I've used the script to get it.

lovettchris, 2017-04-18:
Yes, this is normal behavior, I see the same thing on Windows.   Multiple ASimModeBase objects come and go, so this seems to be normal behavior for the UE editor.  Note that record_file  is not a pointer, but it is an empty ofstream, so it should contain null buffers inside it.  I'll have to offer a prize to who ever finds this annoying bug :-)    

PS: I also updated the rpclib bits, can you pull the latest and try that too ?

AlexandreBorowczyk, 2017-04-18:
Hahah yeah it's a head scratcher. I went over the linux build instructions and I dont think I've did the steps regarding libjsoncpp0 and i have Ubuntu 16.04. Could that have a impact?

sytelus, 2017-04-19:
Closing as latest compilation seems to be working fine.

closed 2017-04-19 by sytelus