microsoft/AirSim #113: Possible bug - Random number generation of Eigen::Vector3f with AirLib

jespestana, 2017-03-16:
I extracted the random number generation part of AirLib for a personal project. I am always a little bit concerned with how to seed the random generators properly, and decided to give a try to your library.

I might have found a bug: the float gaussian random generator works perfectly well, however the random generator of Eigen::Vector3f always results in a vector with the same randomly generated number on the three components of the vector. Is this the correct expected behaviour? (I guess not)

This could be important since these same routines are used to simulate the onboard sensors of the MAV, for instance the IMU sensor. 
Can somebody check if whether they experience the same issue in their computer? Thanks!

*** Example code below ***

Commit from which I have extracted the random generation routines for Float and Vector3f:
From repo: https://github.com/Microsoft/AirSim
commit 54d95ea67b74df74dec25b26d07d9ac13353a31b
Author: Shital Shah <shitals@microsoft.com>
Date:   Thu Mar 16 01:16:53 2017 -0700

Minimal code example:
`
// noise generation - AirLib from AirSim - Microsoft Research
#include "AirLib/common/Common.hpp"

int main(int argc, char** argv){

  msr::airlib::RandomGeneratorGausianR gauss_dist_f = msr::airlib::RandomGeneratorGausianR(0, 1);
  msr::airlib::RandomVectorGaussianR gauss_dist_v3f = msr::airlib::RandomVectorGaussianR(0, 1);

  std::cout << "Float32 random number generation:" << std::endl;
  std::cout << "gauss_dist_f:" << gauss_dist_f.next() << std::endl;
  std::cout << "gauss_dist_f:" << gauss_dist_f.next() << std::endl;
  std::cout << "gauss_dist_f:" << gauss_dist_f.next() << std::endl;
  std::cout << "gauss_dist_f:" << gauss_dist_f.next() << std::endl;
  std::cout << "gauss_dist_f:" << gauss_dist_f.next() << std::endl << std::endl;

  std::cout << "Eigen::Vector3f random number generation:" << std::endl;
  std::cout << "gauss_dist_v3f:" << gauss_dist_v3f.next().transpose() << std::endl;
  std::cout << "gauss_dist_v3f:" << gauss_dist_v3f.next().transpose() << std::endl;
  std::cout << "gauss_dist_v3f:" << gauss_dist_v3f.next().transpose() << std::endl;
  std::cout << "gauss_dist_v3f:" << gauss_dist_v3f.next().transpose() << std::endl;
  std::cout << "gauss_dist_v3f:" << gauss_dist_v3f.next().transpose() << std::endl << std::endl;

  return 0;
}
`

Output:
`Float32 random number generation:
gauss_dist_f:1.22192
gauss_dist_f:-0.516964
gauss_dist_f:0.869636
gauss_dist_f:0.721333
gauss_dist_f:1.58856

Eigen::Vector3f random number generation:
gauss_dist_v3f:1.22192 1.22192 1.22192
gauss_dist_v3f:-0.516964 -0.516964 -0.516964
gauss_dist_v3f:0.869636 0.869636 0.869636
gauss_dist_v3f:0.721333 0.721333 0.721333
gauss_dist_v3f:1.58856 1.58856 1.58856
`

sytelus, 2017-03-16:
This is excellent find! Yes, you are correct there is a bug. Each of the component of vector is actually using different instance of random generator (as it should). However, each random generator by default initializes from same seed so they produce same numbers! The reason we have random generator initializing from fixed seed by default is to enable reproducible (this behavior can easily be changed).

In any case, I've fixed the issue in this commit: https://github.com/Microsoft/AirSim/commit/b176ce1df1323a7cbf50ba0aae1e3a94a7b77ab8. So please feel free to get latest.

Thanks for reporting the bug!

jespestana, 2017-03-16:
Your reply was very informative, it really hits true compared to what I noticed while using the random number generation code.

I just rerun my code on the current latest commit and it works. Therefore I close the issue with this reply.

Current AirSim commit:
`
commit 2872e2a51a3723c01f23951e8df82e7e9840ad33
Author: Chris Lovett <clovett@microsoft.com>
Date:   Thu Mar 16 04:04:37 2017 -0700
`

Output of my code:
`
Float32 random number generation:
gauss_dist_f:1.22192
gauss_dist_f:-0.516964
gauss_dist_f:0.869636
gauss_dist_f:0.721333
gauss_dist_f:1.58856

Eigen::Vector3f random number generation:
gauss_dist_v3f:   0.3064  -1.30311 -0.757752
gauss_dist_v3f: 0.156066 -0.264848 0.0896679
gauss_dist_v3f:-0.424386   0.64198  0.811979
gauss_dist_v3f: -0.56804 0.0712087  0.497168
gauss_dist_v3f:-0.204547 -0.652074 -0.666402
`

Thanks for checking and fixing the bug so fast!!

closed 2017-03-16 by jespestana