microsoft/AirSim #114: Could not arm the drone by the shell

zf223669, 2017-03-16:
i start the shell and input the command "arm", but it could not do anything?

lovettchris, 2017-03-16:
can you tell us more about your setup?  Pixhawk hardware, or SITL mode? which operating system, and where are you entering the command?  from QGroundControl or from DroneShell ?  Do you have a flight controller configured?

zf223669, 2017-03-17:
Hello，I used  the pixhawk hardware with frsky RC, my operating system is windows10, i entered the command in the droneShell. The setting.json is below:
{
  "AirControlCompID": 1,
  "AirControlSysID": 134,
  "ExtRendererCompID": 1,
  "ExtRendererSysID": 167,
  "ExternalSimPort": 14588,
  "LocalHostIp": "10.1.9.10",
  "LogViewerHostIp": "10.1.9.10",
  "LogViewerPort": 14388,
  "Pixhawk": {
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true
  },
  "QgcHostIp": "10.1.9.10",
  "QgcPort": 14550,
  "SimCompID": 42,
  "SimSysID": 142
}
[image]

sytelus, 2017-03-18:
Are you able to fly manually by RC? Also, if you are using latest bits, you will need to upgrade to 4.15 as 4.14 is no longer supported because of incompatibility. Upgrade steps are easy. See here for details: https://github.com/Microsoft/AirSim/issues/116#issuecomment-287512253

closed 2017-04-19 by sytelus