microsoft/AirSim #115: An idea of the development of the "Error Injection"

fmzhu, 2017-03-16:
Have anyone thinking of the development of the Simulation of the "Error Injection", e.g. Sensor Failures(Bit-Flips), Software Failures and Network Failures. Perhaps that'll be good for the collection of Data under " Unnormal Situation".

lovettchris, 2017-03-16:
I'm very interested in this, I've been thinking about a smarter "anomaly detection" stack that analyzes every sensor stream so it can detect these sorts of things, so having a way to do fault injection would be a great way to test that.  I would love to be able to detect, for example, that a given sensor is picking up something unusual today that it hasn't seen before, so might not want to fly until we look into it.  Or, the accelerometer is picking up new vibration, perhaps we have a chipped propeller, and so on.  I'd love to see this sort of thing developed and test in AirSim then applied to the real PX4 firmware.  Another thing to test along the same lines is the Pixhawk has redundant IMU's, so what happens when one goes bad?  Is the sensor fusion algorithm they have there doing the right thing.  So adding that to the simulation would be great too.

fmzhu, 2017-04-12:
thank you very much for your nice suggestion! Sorry about the late response, because I'm on an vocation last month! So where do you think we can start? e.g. add some noise to the sensor? Would you like to give me some suggestion!

lovettchris, 2017-04-13:
AirSim already models a small amount of sensor noise on every sensor because all real sensors have some noise.  The trick then is creating an algorithm that can detect unusual patterns in the sensor streams that cannot be explained by noise.  First, can you detect a broken propeller by changes in the accelerometer signal on a real Pixhawk flight controller?  If you can reliably detect something is wrong, versus normal noise and vibration that happens when a drone is flying then you have something useful.  Modelling all that in the simulator then would also be interesting way to demonstrate this idea.  The real anomaly detection may require higher level reasoning than just looking at accelerometer.  For example, a broken propeller might also change the way the drone responds to controls, for example, perhaps the pitch, yaw, roll (attitude) is not responding like it did before, and so on.  So there's tons of things to try here and it might help to read up on streaming analytics and anomaly detection.

closed 2017-04-19 by sytelus