microsoft/AirSim #116: Build error, win 10 vs2015 update 3 community.

Maxfashko, 2017-03-17:
Hello. I used the sample assembly from this video https://www.youtube.com/watch?v=1oY8Qu5maQQ&feature=youtu.be. The following errors occurred during the build (debug x64):

Error LNK2038 detected a mismatch for "_ITERATOR_DEBUG_LEVEL": value "0" does not match the value "2" in main.obj DroneShell C: \ Users \ maxfa \ Code \ AirSim \ DroneShell \ AirLib.lib (client.obj)
Error LNK2038 detected a discrepancy for "RuntimeLibrary": the value of "MD_DynamicRelease" does not match the value of "MDd_DynamicDebug" in main.obj DroneShell C: \ Users \ maxfa \ Code \ AirSim \ DroneShell \ AirLib.lib (client.obj) 1
Error LNK2038 detected a mismatch for "_ITERATOR_DEBUG_LEVEL": value "0" does not match the value "2" in main.obj DroneShell C: \ Users \ maxfa \ Code \ AirSim \ DroneShell \ AirLib.lib (rpc_error.obj) 1
Error LNK2038 detected discrepancy for "RuntimeLibrary": MD_DynamicRelease value does not match MDd_DynamicDebug value in main.obj DroneShell C: \ Users \ maxfa \ Code \ AirSim \ DroneShell \ AirLib.lib (rpc_error.obj) 1
Error LNK2038 detected a mismatch for "_ITERATOR_DEBUG_LEVEL": value "0" does not match the value "2" in main.obj DroneShell C: \ Users \ maxfa \ Code \ AirSim \ DroneShell \ AirLib.lib (response.obj) 1
Error LNK2038 detected a discrepancy for "RuntimeLibrary": the value of "MD_DynamicRelease" does not match the value of "MDd_DynamicDebug" in main.obj DroneShell C: \ Users \ maxfa \ Code \ AirSim \ DroneShell \ AirLib.lib (response.obj) 1
Warning LNK4098 default library "MSVCRT" is inconsistent with the use of other libraries; Use the / NODEFAULTLIB parameter: library DroneShell C: \ Users \ maxfa \ Code \ AirSim \ DroneShell \ LINK 1
If you comment out the lines in build.cmd:

REM msbuild /p:Platform=x64 /p:Configuration=Debug AirSim.sln
REM if ERRORLEVEL 1 goto :buildfailed

Assembly is performed without errors.
After completing the steps to create a project, unreal gives an error:

The following modules are missing or build with a different engine version:
UE4Editor-AirSim.dll
Would you like to rebuild them now?

Timebutt, 2017-03-17:
Hi @Maxfashko, have you tried cleaning (Build->Clean Solution) before building in Visual Studio? I sometimes have build errors too when importing newly built versions of AirSim, cleaning beforehand gets rid of those.

If not, did you choose to rebuild the UE4-Editor-AirSim.dll?

Maxfashko, 2017-03-17:
@Timebutt, 

> Hi @Maxfashko, have you tried cleaning (Build->Clean Solution) before building in Visual Studio?

No, I have not tried it.

> If not, did you choose to rebuild the UE4-Editor-AirSim.dll?

Yes, the project is going. In the unreal there is no directory containing the plugin AirSim. When you run the project in unreal, nothing happens

Timebutt, 2017-03-17:
So did cleaning help you build the project or not? I don't understand how you can get to the Unreal project without having built the Visual Studio project. Does this mean you got post the build error?

Maxfashko, 2017-03-17:
@Timebutt , Yes, instead of the file I used AirSim.sln in the directory and collected Debug after cleaning.
But now nothing is still happening when I press to play in the project. Maybe there are no scene cameras installed?

Timebutt, 2017-03-17:
Ok, so you got the project built and are now in the Unreal environment, check. It's absolutely possible that there is no starting position defined in your current map. You can check this out by typing 'start' in the top right input box, it should show you all the currently defined starting positions. See this cropped screenshot:

[image]

The [instruction video](https://youtu.be/1oY8Qu5maQQ?t=294)(the link starts at the relevant part) also clearly shows how you should position this starting point for optimal result.

Don't forget to set your _World Settings_ to _SimGameMode_, otherwise the drone won't spawn.

Maxfashko, 2017-03-17:
@ Thimebutt, Yes, I have a starting position. And after clicking on the playlist, I see:
[image]

Timebutt, 2017-03-17:
So, what's exactly the problem you're having now? Did you check the instructional video and _World Settings_ configuration?

Maxfashko, 2017-03-17:
The fact is that I did not find the description of the _World Settings_. I want to achieve the same result as on the video (flight, segmentation cameras, depth, etc)
[image]
[image]

sytelus, 2017-03-18:
Hi Folks, this seems to be because you are using 4.14. We just upgraded everything to 4.15 couple of days ago. This required some changes in BP_CameraDirector but unfortunately it turned out that if you modify these in 4.15 then they become invisible (and unusable!) in 4.14. That is, written BPs in 4.15 are no longer compatible to 4.14. Even worse was that there are no error messages and you will basically see nothing happening when you press Play button. This was disappointing however there is no known way to go back and we don't want to get in the hassle of of maintaining two versions. So we suggest that every one upgrade to 4.15. I've also put in some code to display warning message in latest version. Upgrading to 4.15 is easy: [here are the steps] (https://github.com/Microsoft/AirSim/blob/master/docs/unreal_upgrade.md).

Maxfashko, 2017-03-18:
@sytelus,  thanks. Really, I have 4.14.3 version.
I try to reinstall unreal.

Maxfashko, 2017-03-18:
@sytelus, thanks. Reinstallation helped in my case. Now I got the following:

[image]

Can I control the drone with a keyboard or mouse? Without specialized tools?

Timebutt, 2017-03-20:
As far as I know, there is no way to control the drone with mouse or keyboard at this instant. You'll have to either use an RC controller wirelessly connected through a Pixhawk device or directly through USB, or some kind of [joystick/game controller](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md). 

It would also be very hard to actually control a drone using a keyboard: the control values would vary from 0 to max, nothing in between. While I do agree it would be a nice-to-have feature to test if your setup is correctly running, it will in no way come close to actually flying a drone using a dedicated controller. You need very accurate control to be able to fly a drone correctly.

closed 2017-03-20 by Maxfashko