microsoft/AirSim #118: Semantics or labels in AirSim object segmentation scenes?

Thusheep, 2017-03-17:
Hi. I would like to have some further information from your Demo video of AirSim. 

For the middle images with the realtime object segmentation, what do the segmentations contain and return? Do you use semantics or any geometrical segmentation or are there labels like tree, road, roof etc? I am just seeing this colourful separations but cannot refer to anything.

Also is someone familiar with that problem: Since my update from Windows 8 to Windows 10 I get this error message "DX11 feature level 10.0 is required to run the engine" eventhough DX12 is installed. I tried reinstalling UE4 and updating the drivers but nothing works.

My system information:
OS: Win10 64-bit
Processor: Intel(R) Xeon(R) CPU E5-2697  @ 2.7GHz (24CPUs)
Memory: 32GB RAM
DirectX version: DirectX 12

Thank you.

sytelus, 2017-03-18:
In Unreal environment you can have bunch of objects getting rendered as things like trees, houses, roof on house and so on. Each of these object can be assigned an integer value corresponding to index in a color pallet. So if you assigned 1 to tree objects in your environment, they would be rendered as green (for example) in object segmentation window. The way you assign these integer values defines the meaning of each color and it depends on what you want to do, specific environment you have and objects you have in that environment. By default, currently, we assign random ints to objects we discovered in environment (because we have no idea what environment you are using). So you want to oveeride that behaviour in FlyingPawn::setupStencilIDs() function or assign proper values in StencilID property in object browser window in Unreal Editor. In future, we will have APIs available so you can set stencil IDs via Python without having to mess with Unreal.

sytelus, 2017-03-18:
Re: DX11 issue: https://answers.unrealengine.com/questions/194326/dx11-feature-level-100-is-required-to-run-the-engi.html

sytelus, 2017-04-19:
I've written more detailed answer on how to change labels here: https://github.com/Microsoft/AirSim/issues/153

closed 2017-04-19 by sytelus