microsoft/AirSim #120: Build Plugin Problems (Get Error)

Focke-Wulf, 2017-03-19:
Hello All:
I am a student and find this interesting project!
I am now trying to follow the steps to set up the Environment.
from this Vedio..https://www.youtube.com/watch?v=HNWdYrtw3f0&t=277s

However i have  tech issue when i trying to run Build.cmd to Build my Plugins file..
the Errors are following:

 SafetyEval.cpp
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Users
\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : warning C4819: The file contains a character that cannot be represented in t
he current code page (936). Save the file in Unicode format to prevent data loss [C:\Users\panze\Documents\GitHub\AirSim\AirLib\AirLi
b.vcxproj]
  RpcLibServer.cpp
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Users
\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : warning C4819: The file contains a character that cannot be represented in t
he current code page (936). Save the file in Unicode format to prevent data loss [C:\Users\panze\Documents\GitHub\AirSim\AirLib\AirLi
b.vcxproj]
  RpcLibClient.cpp
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Users
\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : warning C4819: The file contains a character that cannot be represented in t
he current code page (936). Save the file in Unicode format to prevent data loss [C:\Users\panze\Documents\GitHub\AirSim\AirLib\AirLi
b.vcxproj]
  ObstacleMap.cpp
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Users
\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : warning C4819: The file contains a character that cannot be represented in t
he current code page (936). Save the file in Unicode format to prevent data loss [C:\Users\panze\Documents\GitHub\AirSim\AirLib\AirLi
b.vcxproj]
  MavLinkDroneController.cpp
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Users
\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : warning C4819: The file contains a character that cannot be represented in t
he current code page (936). Save the file in Unicode format to prevent data loss [C:\Users\panze\Documents\GitHub\AirSim\AirLib\AirLi
b.vcxproj]
  DroneControllerBase.cpp
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Users
\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]
c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : warning C4819: The file contains a character that cannot be represented in t
he current code page (936). Save the file in Unicode format to prevent data loss [C:\Users\panze\Documents\GitHub\AirSim\AirLib\AirLi
b.vcxproj]
  Generating Code...

"C:\Users\panze\Documents\GitHub\AirSim\AirSim.sln" (default target) (1) ->
"C:\Users\panze\Documents\GitHub\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
"C:\Users\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj" (default target) (3) ->
(ClCompile target) ->
  **

> c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Use
> rs\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]
>   c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Use
> rs\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]
>   c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Use
> rs\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]
>   c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Use
> rs\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]
>   c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Use
> rs\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]
>   c:\eigen\eigen3\eigen3\eigen\src\core\arch\cuda\half.h : error C2220: warning treated as error - no 'object' file generated [C:\Use
> rs\panze\Documents\GitHub\AirSim\AirLib\AirLib.vcxproj]

**

    6 Warning(s)
    6 Error(s)

Time Elapsed 00:01:18.93
#### Build failed

Focke-Wulf, 2017-03-19:
Can someone help me to solve this problem? 
I really confuse why this happened?

lovettchris, 2017-03-20:
To upgrade to 4.15, you might want to first right click your *.uproject file and select this option:
[image]

This will allow you to switch the project to 4.15.  But I see you are using Landscape Mountains, and unfortunately, this one does not upgrade, you have to switch your default unreal version to 4.15 in the Epic Games Launcher, under the Library tab select this:
[image]

Then download the landscape mountains again and re-integrate the AirSim plugin again. Then it should be good.

Focke-Wulf, 2017-03-21:
Thanks for the help! I am now fix this problem!

closed 2017-03-21 by Focke-Wulf

lovettchris, 2017-03-21:
Great!