microsoft/AirSim #121: Can't Lunch Drone in Unreal Engine when try to test Flying

Focke-Wulf, 2017-03-19:
Hi All
I just come back to use 4.14 and follow the set up environment and build the Plugins.. every thing goes fine...

but when i click the play button...

Nothing happened... no controls and no Drone lunch into the environment...
here is my floder
[image]

this is the test environment...
[image]

when i click the play button....it only shows like this..
[image]

how should i fix this ?
Thanks!

Timebutt, 2017-03-20:
You'll need to switch to Unreal Editor 4.15, 4.14 is no longer supported. See  https://github.com/Microsoft/AirSim/issues/116#issuecomment-287512253.

Focke-Wulf, 2017-03-20:
When i am trying to rebuild the project from 4.14 to 4.15 
it shows ...
[image]

i have no idea why ....

saihv, 2017-03-20:
Instead of clicking on 'Yes' when Unreal asks whether to rebuild the dll, did you try 'Generate Project Files' and then building directly from Visual Studio as mentioned in the instructions?

lovettchris, 2017-03-20:
Note, you have to use 4.15 if you are on the latest AirSim bits because the drone asset in the blueprint only works on 4.15.  To use 4.14 you would have to use the "Unreal4.14" branch that created, but that branch will not be getting updates to the code.

As for the screen shot you included, you might try changing PlayerStart location to be closer to the ground as I show in this [airsim setup video](https://www.youtube.com/watch?v=1oY8Qu5maQQ&feature=youtu.be).

Focke-Wulf, 2017-03-21:
Thanks for the help. i am now fix this problem!!!

closed 2017-03-21 by Focke-Wulf