microsoft/AirSim #123: DroneShell commands not working with RosFlight

saihv, 2017-03-20:
I was trying to test the new RosFlight mode in the updated UE4.15 AirSim code. I changed the vehicle configuration from Pixhawk to RosFlight in `createVehicle()` [line 120 of SimModeMultiRotor.cpp], and I can see the RosFlight relevant updates on the screen once the sim is started. Problem is that DroneShell commands do not work. Arm doesn't result in the motors spinning; and any take off or waypoint command causes an exception to be thrown in the RosFlightDroneController.hpp, which complains that the variable `kinematics_` contains NULL.

sytelus, 2017-03-22:
Well, first congratulations for catching this new hidden feature! This is highly experimental right now. Basically the idea is that we want to have a "stand-alone mode"  so you don't need to do complicated set up with PX4. The ROSFlight is very basic firmware which only has ability to do attitude control. To enable full APIs, I still need to write the code to add position and velocity control. For now my focus is to just make ROSFlight work on manual flights with joystick. It currently does work but behaviour is still bit unstable so I need to debug that. Once that part is done, I'm planning to either extend ROSFlight with higher level controllers or use integrate ros_copter that some of it. This will then have all the functionality required by DroneControllerBase so APIs would start working as well. BTW, I'm also working with Simon Levy for integrating Hackflight.

saihv, 2017-03-22:
Ah I see, thanks for the update!