microsoft/AirSim #124: Can't   fly

jjy201314, 2017-03-21:
it  shows the errors:
[image]

the  settings.json  is:
{
  "Pixhawk": {
    "LocalHostIp": "127.0.0.1",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "COM3",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14560,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}
how   can I do?

lovettchris, 2017-03-22:
For some reason I cannot see the screen shot you tried to upload.  All I can see is:

[image]

Perhaps GitHub or someone is blocking it.  Can you try and share the screen shot using some other place?

jjy201314, 2017-03-23:
thank you very much,lovettchris.  Maybe   something   wrong   happened,  I  will  try  other  ways  to  show  you   my  errors  .

closed 2017-03-23 by jjy201314