microsoft/AirSim #125: Trouble with cmake for linux build

gmgremillion, 2017-03-22:
I just need clarification as to which cmake directory is referred to by these [instructions ](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md)

> Now go to **cmake** folder and run following commands:
> 
> ```
> mkdir build
> cd build
> cmake -D CMAKE_C_COMPILER=gcc-6 -D CMAKE_CXX_COMPILER=g++-6 -D CMAKE_BUILD_TYPE=Debug ..
> make
> ```
> 

I tried creating a 'build' directory in '~/AirSim/cmake' and ran the cmake command above and received the following error:

```
-- found AIRSIM_ROOT=/home/greg/AirSim
CMake Error at CMakeLists.txt:9 (add_subdirectory):
  The source directory

    /home/greg/AirSim/external/rpclib

  does not contain a CMakeLists.txt file
```

saihv, 2017-03-22:
I would just directly run the 'build.sh' script. The reason you're getting that error is because the rpclib submodule was probably not checked out, and the command to do that is included in the build.sh script. I think those instructions are outdated, the project requires clang instead of g++ anyway.

sytelus, 2017-03-22:
Hi folks, please feel free to contribute the updates! We love PRs :).

gmgremillion, 2017-03-22:
Running the build.sh script in AirSim/cmake runs successfully, but from [these video instructions for building in Windows](https://www.youtube.com/watch?v=HNWdYrtw3f0&feature=youtu.be), it seems I should probably be running the build.sh script in AirSim, instead. However, this errors, stating that clang++ 3.9 is necessary, and a `sudo apt-get install clang++-3.9` fails as it's `unable to locate the package clang++-3.9`

Am I right in assuming the build.sh I should be running is in the AirSim directory, not the AirSim/cmake directory? If so, is there a repository I need to add to get clang++-3.9?

lovettchris, 2017-03-22:
yes, apologies for the confusion, I'm still using cmake/build.sh to get gcc built binaries for testing.  But the "clang" build is what you need for Unreal.    I have checked in some fixes to the top level build.sh so it works with clang 3.9.  After installing clang 3.9 you will also need the latest libc++.  I have made some fixes in the top level build.sh and updated the [Linux build docs](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md).

saihv, 2017-03-22:
Ah yes, I don't think clang 3.9 is available through the standard channels in 14.04 or 16.04. You would have to try this: 

`sudo add-apt-repository ppa:ubuntu-toolchain-r/test`
`sudo apt-get update`
`sudo apt-get install clang-3.9 clang++-3.9`

lovettchris, 2017-03-22:
By the way, I've never used Unreal on Linux, I assume you are following [these Unreal instructions](https://wiki.unrealengine.com/Building_On_Linux#Clang)?

saihv, 2017-03-22:
IIRC, those manual instructions are only for versions below 4.7. For 4.14/15, Unreal provides a Setup.sh that takes care of all dependencies and building, but for Ubuntu 16.04, the script uses clang-3.8.

gmgremillion, 2017-03-22:
@lovettchris Yes, I'm using those instructions and have successfully built Unreal. I'm running through your updated Linux build document now.

I've been trying the following commands that @saihv and your instructions suggest

```
sudo apt-get update
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt-get install clang-3.9 clang++-3.9
```

but I keep getting the following errors

```
E: Unable to locate package clang-3.9
E: Couldn't find any package by regex 'clang-3.9'
E: Unable to locate package clang++-3.9
E: Couldn't find any package by regex 'clang++-3.9'
```

lovettchris, 2017-03-22:
Assuming you are on Ubuntu 16 ? can you try this then?

wget -O - http://apt.llvm.org/llvm-snapshot.gpg.key | sudo apt-key add - 
sudo apt-add-repository "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-3.9 main" 
sudo apt-get install clang-3.9 clang++-3.9

gmgremillion, 2017-03-22:
Ah, no. I'm on 14.04.

lovettchris, 2017-03-22:
There's some more instructions for 14.04 [on this page](http://askubuntu.com/questions/787383/how-to-install-llvm-3-9).  Can you find something that works there then I'll update the docs with that...

gmgremillion, 2017-03-23:
I am able to install clang 3.8, but when trying to install 3.9 via `sudo apt-get install clang-3.9` I get the following errors:

```
The following packages have unmet dependencies:
 clang-3.9 : Depends: libclang1-3.9 (= 1:3.9.1~svn288847-1~exp1) but it is not going to be installed
             Depends: libjsoncpp1 but it is not installable
             Depends: libllvm3.9 but it is not going to be installed
             Depends: libclang-common-3.9-dev (= 1:3.9.1~svn288847-1~exp1) but it is not going to be installed
             Recommends: llvm-3.9-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

I've tried installing the listed dependencies individually, e.g., `sudo apt-get install libllvm3.9`, which then yields

```
The following packages have unmet dependencies:
 libllvm3.9 : Depends: libtinfo5 (>= 6) but 5.9+20140118-1ubuntu1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

and subsequently `sudo apt-get install libtinfo5` yields

```
libtinfo5 is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
```

I've also tried adding the following to `/etc/apt/sources.list` 

```
deb http://apt.llvm.org/trusty/ llvm-toolchain-trusty-3.9 main
deb-src http://apt.llvm.org/trusty/ llvm-toolchain-trusty-3.9 main
```

and running the following with no luck

`wget -O - http://apt.llvm.org/llvm-snapshot.gpg.key|sudo apt-key add -`

I'm not sure how to resolve this. Any suggestions?

lovettchris, 2017-03-23:
Ah, I see you have entered the apt-get dependency hell.  The error usually means you have something install that depends on something that conflicts with something you are trying to install.  So it did not install the new thing because that would break some existing thing.  Usually the fix for that is removing stuff that you don't need.  I would start with removing clang 3.8.

gmgremillion, 2017-03-23:
I've tried about everything I can find online to deal with this issue (removing 3.8, synaptic, aptitude, etc.) and nothing has resolved this issue. I was able to use synaptic to install successively and successfully 3.3, 3.4, 3.5, 3.6, 3.8, and finally 5.0. However, 3.9 still gave the "held broken packages" error. 

I currently have clang 5.0 installed and was trying to move forward with the [instructions](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md), assuming 5.0 would be sufficient. Is this a misguided assumption?

Currently, I'm trying to run the below command but am unsure of where this .sh file is located. Can you advise?

> ```
> ./cmake/getlibcxx.sh
> ```

Thank you.

lovettchris, 2017-03-23:
5.0 should work, they rarely remove features from compilers :-) I haven't tried it, but give it a go.

I see I forgot to checkin my getlibcxx script, sorry about that.  Since you have 5.0 now, try the instructions on this page: [http://libcxx.llvm.org/docs/BuildingLibcxx.html](http://libcxx.llvm.org/docs/BuildingLibcxx.html).

gmgremillion, 2017-03-23:
I realized that 5.0 is **not** supported by Unreal while trying to run the Setup.sh referred to in [their instructions](https://wiki.unrealengine.com/Building_On_Linux).

If 3.8 doesn't work with AirSim, then it seems I'm between a rock and hard place. :-/

lovettchris, 2017-03-23:
Hmm. Sounds like we really need to get 3.9 working on your machine somehow...  Sorry we aren't further along in Linux support.  You are blazing the trail here.  [Vincent Lecrubier ](https://github.com/crubier) did get it working on OSX, and I think he was using clang 3.9 which I think is what comes with XCode 8.  I do know that in a clean Linux VirtualBox, I can install clang 3.9 and 'build.sh' works.  I have not yet tried to get that working build of AirSim working with Unreal, for that I need a native linux box with real graphics card and matching drivers....

lovettchris, 2017-03-25:
And the Travis CI build is now passing.  See ~/.travis.yml.

lovettchris, 2017-04-06:
closing due to inactivity, please open new thread if you still have trouble, thanks.

closed 2017-04-06 by lovettchris