microsoft/AirSim #126: Unreal Editor crashes when hit play (Windows) - New Issue

stvnmoon, 2017-03-22:
I tried to address this in #111 but no one is answering and it is a genuine request for help.

@lovettchris 

I'm having this issue also, only I do have HIL but get this stacktrace after running:
`>	UE4Editor-AirSim.dll!std::_LaunchPad<std::unique_ptr<std::tuple<void (__cdecl mavlinkcom_impl::MavLinkNodeImpl::*)(void) __ptr64,mavlinkcom_impl::MavLinkNodeImpl * __ptr64>,std::default_delete<std::tuple<void (__cdecl mavlinkcom_impl::MavLinkNodeImpl::*)(void) __ptr64,mavlinkcom_impl::MavLinkNodeImpl * __ptr64> > > >::_Run(std::_LaunchPad<std::unique_ptr<std::tuple<void (__cdecl mavlinkcom_impl::MavLinkNodeImpl::*)(void),mavlinkcom_impl::MavLinkNodeImpl *>,std::default_delete<std::tuple<void (__cdecl mavlinkcom_impl::MavLinkNodeImpl::*)(void),mavlinkcom_impl::MavLinkNodeImpl *> > > > * _Ln) Line 247	C++
`

MavLinkTest.exe works fine:

```
PS C:\Users\steve\work\AirSim\x64\Release> .\MavLinkTest.exe -serial
Auto Selecting COM port: PX4 FMU (COM3)
Connecting to serial port COM3, baudrate=115200
isLocalControlSupported failed: Two second timeout waiting for response to mavlink command MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES
Downloading drone parameters so we know how to control it properly...
STATUS: sev=6, 'APM:Copter V3.4.6 (e707341b)'
STATUS: sev=6, 'PX4: de6b667d NuttX: 8c965992'
STATUS: sev=6, 'Frame: QUAD'
STATUS: sev=6, 'PX4v2 00350031 3432470E 31323533'
Ready...
### command 176 result: MAV_RESULT_UNSUPPORTED
System status:
    onboard sensors present:
    MAV_SYS_STATUS_SENSOR_DIFFERENTIAL_PRESSURE - 0x10 differential pressure
    MAV_SYS_STATUS_SENSOR_GPS - 0x20 GPS
    MAV_SYS_STATUS_SENSOR_VISION_POSITION - 0x80 computer vision position
    MAV_SYS_STATUS_SENSOR_RC_RECEIVER - 0x10000 rc receiver
    MAV_SYS_STATUS_SENSOR_3D_GYRO2 - 0x20000 2nd 3D gyro
    onboard sensors enabled:
    MAV_SYS_STATUS_SENSOR_ABSOLUTE_PRESSURE - 0x08 absolute pressure
    MAV_SYS_STATUS_SENSOR_DIFFERENTIAL_PRESSURE - 0x10 differential pressure
    MAV_SYS_STATUS_SENSOR_GPS - 0x20 GPS
    MAV_SYS_STATUS_SENSOR_OPTICAL_FLOW - 0x40 optical flow
    MAV_SYS_STATUS_SENSOR_VISION_POSITION - 0x80 computer vision position
    MAV_SYS_STATUS_SENSOR_EXTERNAL_GROUND_TRUTH - 0x200 external ground truth (Vicon or Leica)
    MAV_SYS_STATUS_GEOFENCE - 0x100000 geofence
    MAV_SYS_STATUS_TERRAIN - 0x400000 Terrain subsystem health
    CPU load 24388
    battery voltage 20037 millivolts
    battery current 20294 milliamps
    drop_rate_comm = 17234
    errors_comm = 69
    battery_remaining = 53
Connected:
    Version=3
    Type=MAV_TYPE_QUADROTOR-  Quadrotor
    Autopilot=MAV_AUTOPILOT_ARDUPILOTMEGA, ArduPilotMega / ArduCopter, http://diydrones.com
    State=MAV_STATE_STANDBY - System is grounded and on standby. It can be launched any time.
    Base mode:
    MAV_MODE_FLAG_STABILIZE_ENABLED - system stabilizes electronically its attitude (and optionally position). It needs however further control inputs to move around.
    MAV_MODE_FLAG_MANUAL_INPUT_ENABLED - remote control input is enabled.
    Custom mode=0
    VEHICLE SYSTEM ID: 1
    VEHICLE COMPONENT ID: 1
STATUS: sev=2, 'PreArm: Compass not calibrated'
STATUS: sev=2, 'PreArm: Accels not calibrated'
STATUS: sev=2, 'PreArm: Check FS_THR_VALUE'
STATUS: sev=2, 'PreArm: Throttle below Failsafe'
```

I'm at a complete loss. I've stepped through until RpcLibServer.cpp line 116. Executing this crashes Unreal as described above with the stacktrace above.

`pimpl_->server.async_run(4);   //4 threads`

[image]

[image]

Any assistance would be appreciated. I have followed all instructions and cannot get it to run.

Additional Info:

I've tried a number of settings.json settings and this is my latest:
```
{
  "AirControlCompID": 1,
  "AirControlSysID": 134,
  "ExtRendererCompID": 1,
  "ExtRendererSysID": 167,
  "ExternalSimPort": 14588,
  "LocalHostIp": "127.0.0.1",
  "LogViewerHostIp": "127.0.0.1",
  "LogViewerPort": 14388,
  "Pixhawk": {
    "LocalHostIp": "127.0.0.1",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "Model": "Generic",
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  },
  "QgcHostIp": "127.0.0.1",
  "QgcPort": 14550,
  "SimCompID": 42,
  "SimSysID": 142
}
```

lovettchris, 2017-03-22:
The settings should look like this:
```
{
  "Pixhawk": {
    "LocalHostIp": "127.0.0.1",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}

```

lovettchris, 2017-03-22:
I also added new [detailed unreal setup instructions](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_demo.md).

stvnmoon, 2017-03-22:
Thanks. I will go through that in detail to see if that resolves my issue.

stvnmoon, 2017-04-02:
Hi,

@lovettchris Sorry for the delay. I was able to get they system running with your new detailed instructions so Unreal no longer crashes, however the sim doesn't actually fly even though it recognizes takeoff, landing, flight modes, etc as I move my RC controls. If I give throttle for a while the text indicates takeoff and when I remove throttle, after a while it indicates landing all while the simulator world shows no activity apart from the quad sitting there jiggling around a bit... I didn't have much time to look it since. Does this sound familiar at all?

lovettchris, 2017-04-03:
Interesting, this could be caused by several issues, can you follow the [logging instructions](https://github.com/Microsoft/AirSim/blob/master/docs/logging.md), then I'll take a look and see if I can pinpoint the problem.

tklinterg, 2017-04-03:
Hi, I have a similar issue, but the difference is that I get the crashing when trying use the SITL mode (but it never gets this far). 

When I hit the play button in the unreal game editor then Unreal stops wokring for some unknow reason.

The process stops working at line 247 in a xthtread file 

[image]

Please let me know if more information is needed.

Best regards

lovettchris, 2017-04-05:
Can you open a new issue on that, including full stack where it crashes?  
I'd like to keep the original issue posted by Steven who got to the point where it was not crashing for him.

Steven, I made some fixes that might help with what you were seeing, please try again and open a new issue if you still have a problem, with logs as per [logging instructions](https://github.com/Microsoft/AirSim/blob/master/docs/logging.md) thanks.

closed 2017-04-05 by lovettchris