microsoft/AirSim #127: Some questions about cameras

JohnDing1995, 2017-03-26:
Is the *stereo camera mode* supported at present?I still can't find the api interface to add a second camera.
Besides, where can I *modify the camera parameter*? I can't find it in the source code of the unreal plugin"Airsim" . Because I need two cameras to estimated the depth,and these cameras,which is attached to UAV,  should heading to the different direction.So,I need to change some factors of camera, like the heading or field of view.
Thanks!

sytelus, 2017-04-19:
DJI has different model to enable HITL simulation. We are currently working with few DJI folks to see if we can integrate this with AirSim. Stay tuned!

closed 2017-04-19 by sytelus

IeiuniumLux, 2018-02-01:
@sytelus Any update on this?  I have one of this [Lightbridge-2](https://www.dji.com/lightbridge-2) units and I can give it a try if AirSim now supports it.

chaitanyantr, 2019-02-08:
i am too waiting for that

demoniusrex, 2019-03-05:
I am working on a multi-rotor API implementation for DJI Onboard-SDK  that should support M100 with a few tweaks. All I have to test on is a N3 flight controller. 

You can check out the work in progress here [AirSim-Onboard](https://github.com/demoniusrex/AirSim-Onboard) [New multi-rotor API](https://github.com/demoniusrex/AirSim-Onboard/blob/master/Rosie/linux/OnboardLib/include/vehicles/multirotor/firmwares/dji/OnboardMultirotorApi.hpp)

ilyak93, 2022-08-28:
@sytelus, was there any success with this ?