microsoft/AirSim #128: Broken movement commands

jamuus, 2017-03-26:
I've been working with both a pixhawk and the sitl successfully on a packaged version of the game built on a commit sometime in the last 1-2 weeks. Attempting to control using a build from the most recent commits causes issues when performing multiple calls to moveByVelocity and moveToPosition where the first call seems to succeed but later calls will fail, sometimes reporting the need to call requestControl. requestControl is called before attempting any offboard commands and adding another call in between movement commands doesn't resolve the errors.

Any pointers to figure out whats changed?

lovettchris, 2017-03-27:
Hi, 

That message usually means the PX4 controller has automatically switched modes (stopping offboard control from continuing).  When this happens the MavLinkVehicleImpl::handleMessage method notices the mode change in the next heartbeat messages, and it resets the state of offboard control in order to communicate to you that you don't have it any more.  There are several reasons that the PX4 can stop offboard control:
1. the next offboard message was too slow causing a timeout.  The offboard messages must be sent at least 20 times a second. 
2. the user flipped a "flight mode" switch on the RC controller, thereby "taking control" back from offboard control.  This is obviously an important safety feature that we must honor to avoid crashing a real drone.  
3. you called a non-offboard control method (like arm, disarm, takeoff, hover, land).  These will all interrupt "offboard control", and require you to call requestControl again.
4. you called releaseControl.

Hope this helps!

lovettchris, 2017-03-27:
Having said all that, this is the "spec", I reviewed the code and did find a [bug](https://github.com/Microsoft/AirSim/commit/a5805bb4bb3c91653c09c84a71c7e92f02a53fbc).  So please sync and try again, thanks.

jamuus, 2017-03-27:
The latest commits seem to have fixed the main problem, thanks! I do still get an exception about needing to request control again sometimes but this doesn't seem to prevent control.

jamuus, 2017-03-27:
Actually I'm still having some issues. Sometimes it works and I can keep issuing velocity commands, other times the velocity commands hang for a second or so before returning and only causing the quad to briefly apply the velocity before returning to a resting attitude.

It seems the velocity commands are battling with something, when it is working and the desired velocity is too low it wont move very fast at all, as if a move to home is being triggered between calls, resisting the direction of movement. I've noticed that I'm a `INFO  [tone_alarm] positive` in the px4 console for each command sent, which doesn't occur on my older working build.

lovettchris, 2017-03-27:
That sounds familiar to a problem we had a while ago.  Can you reproduce the problem using DroneShell? If so, please include the list of DroneShell commands you are sending and the following [logging information](https://github.com/Microsoft/AirSim/blob/master/docs/logging.md).  Thanks.

jamuus, 2017-03-27:
Here is the logs for a run by a test script, calling arm;takeoff;requestcontrol;gotoposition; then looping movebyvelocity.

https://www.dropbox.com/s/7k2pzd94nudhtmg/21_31_28.zip?dl=0

the quad seems to try to move to position but doesn't quite make it, then moves onto the velocity calls but doesn't move in the right direction and steadily falls.

jamuus, 2017-03-30:
Any luck with this?

jamuus, 2017-04-01:
On the latest commits it seems DroneShell is working correctly but hellodrone does not. The position estimation on the controller seems to freeze on the blocking api calls and the drone will start to drift. Quitting hellodrone and calling a new command in droneshell fixes the position updating.

jamuus, 2017-04-02:
I believe the first issue I was happening was due to the fact that [offboard mode is released](https://github.com/Microsoft/AirSim/blob/644951b55c757a418f5b8287c694799a9658fd4d/AirLib/src/controllers/MavLinkDroneController.cpp#L1018) after every is_loop_command=true command in the destructor of [CallLock](https://github.com/Microsoft/AirSim/blob/02faf6767f9f0c618cbd874af935da276ef40b58/AirLib/include/controllers/DroneControllerCancelable.hpp#L255). This doesn't seem like the best way of making sure offboard mode is set given the api requires calling requestControl to do the same thing.

lovettchris, 2017-04-03:
Problem is if we don't call requestControl then the PX4 will attempt to land the drone which we don't want either.  An explicit releaseControl puts the drone into "previous mode" which could be position hold.  So if you set that mode using mavlinkvehicle::setmode or using an RC control, then the transition between move commands should be relatively smooth.  In my testing so far this seems to work fine, but I think I found a related bug.  It seems that when doing back-to-back commands the previous command is not getting cancelled properly, so we might be have commands fighting each other for control, still debugging... I've also seen some weirdness in our RpcLibClient/RpcLibServer where the command called by client ends up executing something different on the server, for example, I was just trying to "land" the drone and the server was trying to execute moveOnPath instead.  So these bugs might be the real problem here...

jamuus, 2017-04-03:
I made a quick and dirty fix by just commenting out the release code and the problems I was having have disappeared. Requesting control before each command is okay as if offboard is already set then this is a noop. The controller will automatically release offboard control if it doesn't receive an offboard command for 0.5s.

Do we really want to be fighting another control mode? I'm using quite small timeouts such that the control is significantly degraded by releasing each time. The px4 offboard api requires [polling offboard commands](https://dev.px4.io/ros-mavros-offboard.html) which must be happening here somewhere. Wouldn't it be better to propagate the timeout behavior into the AirSim api?

lovettchris, 2017-04-04:
I pushed the change that removes the releaseControl in endOffboardMode.

closed 2017-04-04 by jamuus