microsoft/AirSim #129: build.cmd problem (Windows, VS2015 update 3)

GVDubrovskiy, 2017-03-27:
Dear all,

Usually I work under Linux, but because one library was not available for Ubuntu 14 (it supports only 5.9 version for libtinfo5, while AirSim intallation requires >=6). I decided to install under Windows.

I follow steps in build.md, but I cannot run build.cmd

It gives me a list of errors (it looks that I didn't install CMake well, but I don't know how to fix it: I specified that it should make some "variables" in "path")

-------------------------------------------------------------------------------------------------------------
**Errors:**

C:\Program Files (x86)\Microsoft Visual Studio 14.0\AirSim\AirSim>build.cmd
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "C:/Program Files (x86)/Microsoft Visual Studio 14.0/AirSim/AirSim/external/rpclib/build/CMakeFiles/CMakeOutput.log".
See also "C:/Program Files (x86)/Microsoft Visual Studio 14.0/AirSim/AirSim/external/rpclib/build/CMakeFiles/CMakeError.log".
Microsoft (R) Build Engine version 14.0.25420.1
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: ALL_BUILD.vcxproj
Microsoft (R) Build Engine version 14.0.25420.1
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: ALL_BUILD.vcxproj
#### Build failed

-------------------------------------------------------------------------------------------------------------
Thank you for your time!

clovett, 2017-03-27:
Assuming you have visual studio 2015  with the C++ feature installed, you should run build.cmd from a "Developer Command Prompt for VS2015" so that cmake can find your C++ compiler.

GVDubrovskiy, 2017-03-27:
Thank you for your reply! I do run "build.cmd" from a "Developer Command Prompt for VS2015"
[image]

lovettchris, 2017-03-27:
Ok, thanks for confirming that.  Can you also check that you have enabled the C++ compiler feature of Visual Studio ?  From the Add/Remove programs dialog, select Visual Studio and click Modify.  Then check that these boxes are all checked, and if not, check them and click Next so the C++ features are installed.

[image]

jjy201314, 2017-03-27:
hi,  lovettchris,   my question   have  sent to your mailbox.  Can  you   help  me   solve  the    problem   troubled  me   a long  time   if  you  have  time  .  thank   you    very  much . @lovettchris

clovett, 2017-03-27:
I'm sorry, but i not receive you email.  Someone is blocking your emails?

GVDubrovskiy, 2017-03-27:
Thank you for your comments! 
I do have these VC++ features enabled. It still behaves badly. I just deleted VS2017 and tried again (see the screenshoot below).
When I run without administrative rights it says about "permission error". I don't know how to start execution with administrative rights (like sudo in Linux) from Windows prompt.

I start a new "Developer Command Prompt for VS2015" as administrator. Then, it shows the following (see the screenshoot below).

Now I'm going to erase all mentioning of Visual Studio from my laptop, 
restart, 
install VS2015 again, 
restart, 
install CMake (3.7), 
do all steps in build.md

I don't think it will help, but I will try.
Thank you for your time!!

[image]

athersec, 2017-03-27:
Hi! I had this problem, too.
Final solution is unstall the VS2017, and restall the VS2015. It works.
PS: I had tried repairing my VS2015, restall cmake and so on.However, they didn't work.

athersec, 2017-03-27:
Hi! I had this problem, too.
Final solution is uninstall the VS2017, and reinstall the VS2015. It works.
So, it works, when I try only
1. VS2017 and cmake 3.8
or
2. VS2015 and cmake 3.8.
But, if you try vs2017 first, and change to vs2015, it will crash.
If you try vs2015 first, and change to vs2017, and delete the files "extern" .It works.
Somebody says, if you clean the cmake's buffering, it will work again.But I don't know how to do it. So I have to choose to reinstall.
PS: I had tried:
1. Uninstall the vs2017, and repair VS2015. It doesn't work.
2. Reinstall camke. It doesn't work.

athersec, 2017-03-27:
> So, eventually, you have installed only VS2017 and cmake 3.7?
>
> Initially I had VS2017 and cmake 3.8, but it didn't work and I tried installing VS2015.

If you use vs2017, you must include tools vc++ 2015.3 v140. Just look like this.

GVDubrovskiy, 2017-03-27:
Thanks a lot for all your comments!
I erased all VSs and reinstalled the VS2015 (x86) with cmake 3.7 (x86).

I didn't find how to install VS2015 (x64), so I installed (x86) from https://www.microsoft.com/en-us/download/details.aspx?id=48146
Because my OS is (x64), I installed cmake 3.7(x64), but now I installed (x86) from here: https://cmake.org/download/ and it works (it least, the build.cmd says that the build was successful and didn't show any errors).

Working config:
Windows 10 (Education) x64
VS2015 (x86) with VC++
Cmake 3.7 (x86)

Thanks a lot!
I'm closing the topic.

closed 2017-03-27 by GVDubrovskiy

lovettchris, 2017-03-27:
Indeed it appears that cmake added support for VS 2015 in their [3.7 release](https://cmake.org/cmake/help/v3.7/release/3.7.html).

mljack, 2018-01-18:
That's due to the missing rc.exe file in windows 10 sdk folders. (I guess it's installed with vs2017)
set path=C:\Program Files (x86)\Windows Kits\10\bin\10.0.15063.0\x64;%path%
or other sdk folder which has the rc.exe file
then build.cmd won't complains any more.

Shwethakrishnaa, 2019-06-13:
C:\Users\SMURAKR\Desktop\Unity\AirSim-master>REM //---------- copy binaries and include for MavLinkCom in deps ----------

C:\Users\SMURAKR\Desktop\Unity\AirSim-master>msbuild AirLibWrapper\AirsimWrapper.sln  /target:Clean /target:Build  /property:Configuration=Release /property:Platform=x64
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: AirLibWrapper\AirsimWrapper.sln

C:\Users\SMURAKR\Desktop\Unity\AirSim-master>if ERRORLEVEL 1 goto :buildfailed

C:\Users\SMURAKR\Desktop\Unity\AirSim-master>echo(

C:\Users\SMURAKR\Desktop\Unity\AirSim-master>echo #### Build failed - see messages above.  1>&2
#### Build failed - see messages above.

C:\Users\SMURAKR\Desktop\Unity\AirSim-master>exit /b 1

I use windows 10, Visual studio 2017. I have set the environment path for MSBuild.exe
I am having this error for a long time, can someone help me with this?