microsoft/AirSim #131: UE4 editor is broken when running packaged content

athersec, 2017-03-28:
It works well in debuggame model and debuggame editor model.But the drone just goes crazy, so I want to package it or run in development environment.
But when I run in development environment, the UE4 editor stops working if i hit "play".Just like this:
[image]
[image]

If I run in debugame editor model, and package it , when I run the ".exe", it's also broken.

PS: I use VS2015 + UE4.15.1 + cmake 3.8 +win10 x64
i5 + 12G memory + ssd

Is there anyone knowing solution? Thank you for your help!

lovettchris, 2017-04-05:
This can happen if your PC is starving the serial input stream from pixhawk due to lack of CPU power.  One way to confirm this is the issue is to look at the mavlink streams in [PX4 Log Viewer](http://www.lovettsoftware.com/LovettSoftware/Downloads/Px4LogViewer/Px4LogViewer.application) connected via socket and the port defined for "LogViewerPort" in your ~/Documents/AirSim/settings.json.  For example, if the hiresimu messages are not coming in at a high rate giving you many points on the graph per second, then the serial input is getting starved.

athersec, 2017-04-14:
Thanks for your answer and sorry for delay.
Now the airsim plugin won't crash in the editor, even though it can't take off.
However, it still crashes as "UEGame stop working".
I try to connect PX4 Log Viewer to "LogViewerPort", But, I'm sorry I don't know which label can tell me the situation of "CPU power". Can you tell me more things ? Thanks for your kind! 

Operating System: Windows 10 64bit
CPU: Intel Core i5
GPU: Nvidia GTX 1080
RAM: 8 gb
Flight Controller: Pixhawk v2
Remote Control: Walkera Devo10

lovettchris, 2017-04-14:
With PX4 Log Viewer click this button on the top right:
[image]

Then select Socket and localhost, click "Connect" button at the bottom of that panel.  You should see a picture of the drone pose:
[image]

Then click the "record" button:
[image]

and you should see a bunch of mavlink messages show up in the panel on the left.  open "hiresimu_t" and select "xacc", when the xacc has filled the screen you should see something like this:
[image]

Right click that graph and select "export .csv" and attach the file to this thread.

athersec, 2017-04-15:
Thank you! You are very nice!
This issue can be closed now.

closed 2017-04-15 by athersec