microsoft/AirSim #132: Linking errors on Linux build

sparviero19, 2017-03-28:
Hello,

I'm trying to install AirSim on Linux following [these instructions](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md). Unfortunately, I'm experiencing these linking errors when building the libraries:

```
"std::experimental::filesystem::v1::path::compare(std::experimental::filesystem::v1::path const&) const":
/usr/include/c++/v1/experimental/filesystem:959: not defined reference to "std::experimental::filesystem::v1::path::__compare(std::__1::basic_string_view<char, std::__1::char_traits<char> >) const"
"std::experimental::filesystem::v1::path::compare(std::experimental::filesystem::v1::path const&) const":
/usr/include/c++/v1/experimental/filesystem:959: not defined reference to "std::experimental::filesystem::v1::path::__compare(std::__1::basic_string_view<char, std::__1::char_traits<char> >) const"
"std::experimental::filesystem::v1::directory_iterator::operator*() const":
/usr/include/c++/v1/experimental/filesystem:1928: not defined reference to "std::experimental::filesystem::v1::directory_iterator::__dereference() const"
```
Googling these errors pointed me on solutions based on adding a `-lc++experimental` flag in the cmake file. However, despite this seems to be [already managed](https://github.com/Microsoft/AirSim/blob/master/cmake/cmake-modules/CommonSetup.cmake) by the AirSim cmake setup script, it still gives me the above errors.

I'm using the [build.sh](https://github.com/Microsoft/AirSim/blob/master/build.sh) script in the root directory. Building is successful when testing the gcc build script in the `cmake/` directory.

Any hint to solve this problem?

lovettchris, 2017-03-28:
Yes, it is possible you don't have the experimental libc++ library installed, try running the script 

`./cmake/getlibcxx.sh`

sparviero19, 2017-03-28:
Sorry for not mentioning this before, I already ran the `./cmake/getlibcxx.sh` script, thus, I have the experimental libc++ library, but I still get those errors

lovettchris, 2017-04-05:
Do you have this file installed:
````
root@ULYSSES:/mnt/d/git/AirSim# ls -al  /usr/lib/libc++experimental.a
-rw-r--r-- 1 root root 1933448 Mar 22 22:31 /usr/lib/libc++experimental.a
````

sytelus, 2017-04-19:
Please reopen if latest code + instruction still doesn't work.

closed 2017-04-19 by sytelus

curantil, 2017-07-21:
I ran ./cmake/getlibcxx.sh, but I still don't have /usr/lib/libc++experimental.a and thus the build.sh doesn't work. 

Maybe the issue should be reopened until the fix is confirmed? There are popping up duplicates ( #355 and #360 )