microsoft/AirSim #134: Multiple camera views from the drone

lkumar93, 2017-03-30:
It would be great if some one could let me know, if its possible to have multiple cameras on the drone and extract frame from each one of it. For a research project that I am working on, the drone has 2 cameras, one facing forward and the other one facing bottom. I am trying to do deep learning for navigation. It would be great if some one could shed light on this. Ideally it would be great if we could extract 360 degree vision from the drone.

lkumar93, 2017-04-06:
Any help would be greatly appreciated.

sytelus, 2017-04-19:
We have multiple camera support internally but its not fully exposed yet via APIs. We are planning to start work on this but earliest availability for this would be still at least couple of months away. Meanwhile, please feel free to browse the code and contribute in this area!

closed 2017-04-19 by sytelus

fuenwang, 2018-01-03:
Hi @sytelus  ,
Will you work on 360 degree camera? Or just for multiple views? Because I am very interesting in 360 view environment, if you are not planning to do this, I can work on it.