microsoft/AirSim #135: Cannot select On Linux Building

mocibb, 2017-03-30:
I got newest version(68efe37ea56a27d0) of github, and have successful built and run AirSim after remove RosFlightQuadX dependance (MultiRotorConnector.cpp) which can cause some system error when Unreal Engine runs.
But i cannot set Game Mode to SimGameMode, it crashed when saving map.
there are backtraces i got from gdb debug.
[image]
I cannot solve this problem, because I'm not familar with Unreal Engine.

madratman, 2017-04-01:
hey @mocibb, I am facing apparently the same issue:

Can you share exactly what you changed in `MultiRotorConnector.cpp`. 
I guess you changed `vehicle_params_.reset(new msr::airlib::RosFlightQuadX());` and `        static_cast<RosFlightQuadX*>(vehicle_params_.get())->initializePhysics(&environment_, &vehicle_.getKinematics());`
but that doesn't cause any problems?

mocibb, 2017-04-10:
I just remove something that related to RosFlightQuadX, because cause symbol not found error when I run UE.
[MultiRotorConnector.txt](https://github.com/Microsoft/AirSim/files/909503/MultiRotorConnector.txt)
Since the issue I've ran into is tracked by #112 I will close this one.

closed 2017-04-10 by mocibb