microsoft/AirSim #136: makefile not found when run build.sh in the root dir

chenmin1107, 2017-03-31:
It seems that the makefile was put into the cmake folder after run build.sh. But it should be in the build_debug folder.

madratman, 2017-04-01:
Hey, can you share the exact error you got?
For me `build.sh` worked fine after the changes about 7 days back in the docs

chenmin1107, 2017-04-01:
Hi madratman,

my build.sh ran until line "65", which is make command. The error is that there is no makefile in the build_debug/ dir. 

The makefile is in the cmake/ folder.

What I do is to ran a make in the cmake/ folder. Then copy over the output/ folder back to build_debug folder again. Then ran the rest of the build.sh lines.

There must be an easier way to do that.

lovettchris, 2017-04-04:
It is possible you got those files by running cmake\build.sh in cmake.  Can you do this in cmake folder:

rm -rf  CMakeCache.txt  Makefile  CMakeFiles

then rerun the root level build.sh ?

chenmin1107, 2017-04-05:
Yes, you are right. Thanks a lot!

lovettchris, 2017-04-05:
Found a [bug in build.sh](https://github.com/Microsoft/AirSim/commit/b34078e720e928270dd7d8a23ea2c0d99324f226) - it was supposed to delete these files for you.  Now it does.

closed 2017-04-05 by lovettchris