microsoft/AirSim #137: How to install the "cmake" on Windows

Mzhouyi, 2017-03-31:
I have download the "cmake-3.8.0-rc4-win64-x64.msi" and install it on my computer, but when I Run `build.cmd` from the command line, it show that "cmake was not found! First install the latest version from https://cmake.org/."how  can I install the cmake and build the rpclib? please

lovettchris, 2017-03-31:
Make sure the path to your cmake installation bin folder is added to your System Control panel, Advanced System Settings, Environment Variables, Path, like this:

[image]

lovettchris, 2017-04-04:
Please re-open this if this was not the issue, thanks.

closed 2017-04-04 by lovettchris