microsoft/AirSim #138: Quadcopter Not Taking Off in Airsim Simulation

exploke, 2017-03-31:
I have gotten the Airsim environment up and running on linux, but I am unable to make the quadcopter take off. I have followed every step on the Airsim tutorial, and all my software for the Unreal Engine, Airsim, Pixhawk, and Ubunutu is up to date to the tutorial specifications. First, I attempted to use a controller to fly the quadcopter; the controller would be able to arm the quadcopter, but when I thrusted up, nothing would happen. Upon checking the RC inputs on QGroundControl and calibrating the settings multiple times, there was no change in the simulation. With the controller not working, I attempted to use DroneShell to test out different commands, but I ran into the same issues. I was able to arm the quadcopter, but taking off did not work. I either got _drone hasn't came [sic] to expected z of -3.000000 within time 15.000000 sec within error margin_  or _rpclib: function 'takeoff' (called with 1 arg(s)) threw an exception_ when I prompted the "takeoff" command. Using the "getimage" command did save pictures to the Airsim folder (which I verified), so I know the Unreal Simulation is able to connect with the DroneShell. I also observed that using the "disarm" command did not make the propellers stop spinning. Does anybody know if I am doing something wrong, and how to fix the errors?

Timebutt, 2017-04-03:
You seem to be experiencing exactly the same issues I reported in issue #85. Just pulled and built the latest AirSim version to test this again, and I can confirm the issues persist. _GetImage_ works fine, _takeoff_ throws the same error you mention. I too wait for a solution ;)

lovettchris, 2017-04-04:
I added some [documentation in sitl.md](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md) under **Check the Home Position** before trying to takeoff, can you check if that is the issue you are running into here?

vatsagandhi, 2017-04-11:
I am getting the same error as above using the actual PixHawk Hardware. Could it be an issue with how it is set up with QGroundControl? I followed all the steps in the tutorial  and its firmware is up to date so I do not know what could be causing this issue.

lovettchris, 2017-04-11:
When you run DroneShell.exe does it get past the "waiting for GPS location" step ?  if not I'm guessing your forgot to put your PX4 in HIL mode by selecting "HIL Quadrocopter X" in QGroundControl.  Also, please try the latest PX4 firmware (ignore my video that says version 1.4.4).  Use 1.6.0 instead.

vatsagandhi, 2017-04-11:
I was looking at the qgroundcontrol and it kept saying the accel, mag, and gyro calibration failed. I looked it up online and it's not showing me any solutions, so that may be our problem. I ran unreal and drone shell and saw that it was getting past the step of getting the gps location. Additionally I am running the latest PX4 firmware (1.6.0 px4fmu-v2_default.px4).

lovettchris, 2017-04-12:
Sounds like you need to select the simulation AirFrame as shown below:
[image]

exploke, 2017-04-12:
@lovettchris we already selected the HIL Quadcopter X

lovettchris, 2017-04-12:
Can you paste a screen shot of the Unreal simulator, you should see some text in the top left that looks like this:
[image]

vatsagandhi, 2017-04-12:
[image]

It says "taking off Detected" but the Quad never takes off. We get the same error when using Drone Shell in addition to Drone Shell spitting out ("drone hasn't came [sic] to expected z of -3.000000 within time 15.000000 sec within error margin or rpclib: function 'takeoff' (called with 1 arg(s)) threw an exception")

lovettchris, 2017-04-12:
How come the drone looks ghosted and has no shadow? Is the drone under the ice? Mine looks like this:

[image]

vatsagandhi, 2017-04-12:
It is not under the ice, I am using reduced graphics settings so it does not display the shadow of the quad. What should the altitude typically be for this map? I read that this simulator uses the NED frame so should z initially be positive or close to 0?
When I type position on drone shell, it displays Local position: x=0.0410362, y=-0.117092, z=11.0654
Global position: lat=47.6415, lon=-122.14, alt=127.71

lovettchris, 2017-04-12:
The z coordinate is a little high (NED coordinate system has negative z when drone is above ground).  This is what I get :

position: x=0.00353828, y=0.0042316, z=1.02402
Global position: lat=47.6415, lon=-122.14, alt=122.276

Does your drone do a little freefall before hitting the ground when you hit Play ?  Are you getting no movement at all ?  Can you also try "movetoposition -z -20" ?

vatsagandhi, 2017-04-12:
Yes it does do a free fall after we hit the play button but afterwards the drone does not take off at all (even when I type "movetoposition -z -20" . Additionally when I turned the graphics to maximum I see a shadow so I know the quad is not inside the ice. I moved the player start to a different location and when I type "pos" on DroneShell it says z = 0.271578 which is more reasonable than the previous position I was getting.

lovettchris, 2017-04-12:
Ok, try and lower your PlayerStart actor in the UE editor a little bit so the freefall is minimized.

lovettchris, 2017-04-13:
I just checked in a fix that might help with the case where local position 'z' coordinate is unusually high, please pull latest bits and try again.

lovettchris, 2017-04-13:
Actually, just found a proper fix, please try again...

exploke, 2017-04-14:
[image]
[image]

I updated to most current commit, and I am running the simulator on windows. As you can see from the pictures, it is displayed that drone is armed and taking off, but the drone itself is not going anywhere. Using "movetoposition -z -20" seems to be making no difference also. Do you know any more solutions to this issue?

lovettchris, 2017-04-14:
There is a "CPU" setting  under Edit/Editor Preferences, search for "CPU" and uncheck the box labelled "Use Less CPU when in Background".

Also, can you attach the PX4 log file mentioned when you ARM the drone, the above shows the path "/fs/microsd/log/sess021", you can download the log from PX4 using QGroundControl, or you can use the "get" command in MavLinkTest.exe, like this:
> get /fs/microsd/log/sess021/* d:\temp\logs

Then zip the appropriate log and attach it to this issue...

vatsagandhi, 2017-04-14:
[Log 1.zip](https://github.com/Microsoft/AirSim/files/922540/Log.1.zip)

I used QGroundControl to save the log. When I opened QGroundControl though, the display was showing the drone was fluctuating between manual and arm mode before I put in any commands. After attempting to arm the drone, the display on QGroundcontrol was fluctuating between land, takeoff, disarm. The log file I posted is of those occurrences.

lovettchris, 2017-04-14:
This is the QGroundControl mavlink log, it is good, but I was looking also for the PX4 log (PX4 has an sdcard with it's own logs) which you can download using QGroundControl using this toolbar button:
[image], then click 
[image]
on the right side, to see the logs, pick the most recent one and download it.

This QGroundControl mavlink log is interesting it does show the following telling status messages:
````
no RC
failsafe mode on
````

If this is the problem (you have no RC) then you can tell PX4 to ignore that problem by setting the following PX4 parameters:

NAV_RCL_ACT 0

You can use QGroundControl to set this parameter, the value zero means "disabled":

[image]

exploke, 2017-04-14:
I changed the NAV_RCL_ACT to 0 or disabled, and tried running the simulation again. This time, when I went to QGroundControl, the drone was not fluctuating between modes, but takeoff was being denied when I tried. I'm attaching the PX4 log you said to download on the comment above of these occurrences.

[log_0_UnknownDate.zip](https://github.com/Microsoft/AirSim/files/922627/log_0_UnknownDate.zip)

lovettchris, 2017-04-14:
Ok, that is progress, PX4 only denies takeoff if it is not happy with GPS signal being sent by simulator.  So here's some more things to try:
1. kill QGroundControl
2. reboot pixhawk
3. run simulator
4. find MavLinkTest.exe, (should be in MavLinkCom\MavLinkTest\build\x64\Debug\)
5. run MavLinkTest -server:127.0.0.1:14550 -logdir:d:\temp
6. wait for "Home Position" message, then enter the following commands from the MavLinkTest console.
6. type "pos" you should see a valid GPS location 
7. enter "arm" and "takeoff 10"
8. wait 20 seconds, "land", "disarm", "x".

Then send me the resulting logs from your d:\temp\log folder.

exploke, 2017-04-14:
While or after running the simulator, we go into the MavLinkCom\MavLinkTest\build\x64\Debug folder and attempt to open MavLinkTest, but the window that opens immediately closes.

lovettchris, 2017-04-14:
Run it from a DOS command prompt and capture any error message it prints.  I assume here that you are running with ~/Documents/AirSim/settings.json set to "serial":true, so the Simulator is connected to PX4 serial port and then the simulator is publishing to 14550 as per:

    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,

So then MavLinkTest -server:127.0.0.1:14550 will pick up this QGC stream.  Which means if you are running QGC it will not work, which is why I listed "kill QGroundControl" above.

exploke, 2017-04-15:
Attached is the picture of the error we get when running MavLinkTest. we killed QGroundControl but MavlinkTest still does not work.
[image]

lovettchris, 2017-04-15:
Note: It is very unusual that anyone would install a git repro into this location:
[image]

This is the Visual Studio install location.  I normally create a folder at the root of my hard drive named "git" then I clone AirSim in there so I end up with d:\git\airsim.  Can you move AirSim to a location like this and try again?

I'm also guessing you do not have a "D:" drive at all, which explains the crash.  Change this argument:

-logdir:c:\temp

This will cause MavLinkTest to write log files to your C:\temp\log folder.

Then if it still crashes can you run MavLinkTeste.exe with the same command line using Visual Studio and attach the stack trace where it is crashing, thanks.

exploke, 2017-04-15:
[logsMavLinkTest.zip](https://github.com/Microsoft/AirSim/files/923832/logsMavLinkTest.zip)
Thank you very much for the help, MavLinkTest worked and I am attaching the log files from MavLinkTest in this comment.

lovettchris, 2017-04-17:
Thanks for the log, using [PX4 Log Viewer ](http://www.lovettsoftware.com/LovettSoftware/Downloads/Px4LogViewer/Px4LogViewer.application) I can load the *input.mavlink log and it shows something very weird on the Altitude:

[image]

Why is the altitude plummeting to -81 meters below ground every 30 seconds or so, very weird.

The heading is also drifting a lot, was the drone spinning around by any chance?

[image]

exploke, 2017-04-17:
The propellers spin in a very peculiar way; they will start spinning back and forth even before I arm the drone. I also notice the 3 bottom camera views sometimes wobble.

exploke, 2017-04-19:
I re-downloaded and updated all the files. I tried with a different pixhawk and ran into the same problems.This time when I try to run the simulator, the drone shell is unable to get a GPS location which may be a possible explanation to the weird graphs. I am attaching a screenshot of what our unreal environment looks like along with the drone shell. The FPS is usually at 60 before running the simulation, but drops to 15 when running the simulation, and it drops to 3 when running the drone shell as well.

[image]

clovett, 2017-04-19:
Yep, 3 is not enough to fly properly.  Try mouse click on airsim canvas and type "0" to remove the camera views, that should help.

clovett, 2017-04-19:
I would also move off that bridge, it has weird collision mesh that is up higher than the bridge appears.

exploke, 2017-04-21:
I lowered the graphics. moved the drone to the location specified on the github tutorial, and removed the camera views. My FPS went up to about 60 before and during running the simulation. Running the simulation now, the propellers are not fluctuating around anymore; rather, the propellers go slower and faster depending on how much throttle I give from my RC controller. However, the drone is still not going anywhere. Also, the propellers do not stop spinning when I issue a disarm command from my RC controller. I then attempted to use the drone shell, but the FPS dropped to about 3 while I had the program running. The interesting thing I found was that the drone shell kept saying waiting for the GPS location, an issue I was not having before. Is it possible that is why my drone is not going anywhere, because the drone cannot pick up the GPS location?

[image]
[image]

lovettchris, 2017-04-21:
See "[Unreal editor is slow when it is not the active window](https://github.com/Microsoft/AirSim/blob/master/docs/faq.md)"

exploke, 2017-04-21:
Unchecking the "Use Less CPU when in Background" checkbox did have a dramatic FPS rise of 120 while using the RC controller, and 70 FPS while using the DroneShell. Although, the drone is still not moving with RC controller, and the DroneShell is still '"Waiting for drone to report a valid GPS location..."

[image]

lovettchris, 2017-04-21:
Can you run MavLinkTest.exe -serial:*,115200, then when it is started type "params c:\temp\px4params.log" and attach the c:\temp\px4params.log file ?  I want to try the exact same parameters on my pixhawk and see what happens. Which version of px4 firmware are you using?  1.4.4 or the latest 1.6.0 ?

exploke, 2017-04-21:
I am using the latest version of the px4 firmware (1.6.0). Here is the px4params.log file.

[px4params.zip](https://github.com/Microsoft/AirSim/files/948407/px4params.zip)

lovettchris, 2017-04-21:
If you are getting 70fps then flying should be excellent, no problems there.

Your params are quite a bit different from mine, but I don't see anything obvious that should affect takeoff. 
[myparams.zip](https://github.com/Microsoft/AirSim/files/948529/myparams.zip)

Your radio trims are different of course, I assume you calibrated your radius using QGroundControl.

I also included the firmware I'm using which I built a couple days ago just to rule out that variable:
[px4fmu-v2_default.zip](https://github.com/Microsoft/AirSim/files/948542/px4fmu-v2_default.zip)

exploke, 2017-04-22:
Thank you very much, I used the firmware you provided in the comment earlier and the quadcopter lifts off the ground now. Flight is very stable. I am closing this thread. I really appreciate the help.

closed 2017-04-22 by exploke

lovettchris, 2017-04-22:
great, thanks for closing the issue.  Love to know why my firmware works and yours didn't, but we can leave that for another day.  Cheers.