microsoft/AirSim #139: Xbox controller not detected issue

kang235, 2017-03-31:
Greetings,
I have struggled for days and have an issue to use Xbox controller with sitl. Please see below.

1. The Unreal environment runs (screenshot 1). On the top left corner, it says "RC Controller on USB: Not detected".
[image]

2. I think my PX4 service is running correctly (screenshot 2). I can use the command "commander takeoff" to fly the drone in Unreal. 
[image]

3. When I click on "Run" in Unreal, QGroundControl shows "Unknown" under Vehicle Setup for 3 seconds (screenshot 3), then indicates "Communication Lost" (screenshot 4). So that I can not set up my Xbox controller anywhere in QGroundControl. 
[image]
[image]

Please let me know if other information is needed. Help is appreciated. Thank you.

Timebutt, 2017-04-03:
Did you follow the [instructions](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md#using-joystickgamepad-alternative-to-rc) and enable the 'Virtual Joystick' button in QGroundControl?

kang235, 2017-04-03:
@Timebutt Thanks for the reply. 

Yes, I followed the instruction, but since the Xbox controller is unrecognized, I cannot move to step 4 in "Using Joystick/Gamepad (Alternative to RC)". I enabled the "Virtual Joystick", please see my following screenshot.
[image]

saihv, 2017-04-03:
I have a similar controller problem, but it might not be the exact same cause, as I am not using a PX4 vehicle type. Essentially, for me, it looks like the XInputGetState function (Source/SimJoyStick/SimJoyStick.cpp) is returning a 'not connected' value all the time, even though the controller is connected: perhaps you can check if it's the same issue for you?

Timebutt, 2017-04-04:
Hi @kang235, do you see your Xbox controller listed as a valid device under _Devices and Printers_ in Windows (assuming you are on Windows, I couldn't deduce that from your posts) like in my screenshot? Keep in mind I am running an emulator to make my PS3 controller look like an Xbox 360 one, you should have the exact same Xbox 360 device listed however.

[image]

If not, you must first install the drivers required to correctly register your controller in Windows before proceeding.

kang235, 2017-04-05:
@saihv It seems this issue is not because of x360ce. But thanks anyway!

kang235, 2017-04-05:
@Timebutt Yes, I can see my controller (screenshot). I used 'joy.cpl' to make sure the controller is responding properly. Any other possibilities? Thanks.
[image]

Timebutt, 2017-04-12:
What controller are we talking about here exactly, are you using an Xbox 360 controller or a different/older type? I find it curious your controller is listed plain and simple as _controller_, whereas it should be _Xbox 360 controller for Windows_.

kang235, 2017-04-12:
@Timebutt I am using Xbox One controller (screenshot1). 
[image]

Yes, I agree. Please see my settings.json below. "128.210.171.233" is my local ip (using "127.0.0.1" causes connection issue with PX4 service), "128.210.171.141" is the PX4 service ip. 

{
  "Pixhawk": {
    "LocalHostIp": "128.210.171.233",
    "LogViewerHostIp": "128.210.171.233",
    "LogViewerPort": 14388,
    "Model": "Generic",
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "128.210.171.233",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "128.210.171.141",
    "SitlPort": 14556,
    "UdpIp": "128.210.171.141",
    "UdpPort": 14560,
    "UseSerial": false,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}

Timebutt, 2017-04-19:
Hi @kang235, are you running a specific setup that requires  you to host your SITL simulation on a different machine than your own local machine? A standard setup uses the 'local' 127.0.01 IP address, just for reference I added the content of my own _settings.json_ file below.

I'm just asking because it is possible you have either copied someone else's settings, or you may have changed the IP addresses to try and fix your problem. Is that the case? Or are you running the Posix simulation on a remote machine?

```
{
  "Pixhawk": {
    "LocalHostIp": "127.0.0.1",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "LogViewerSendPort": 14389,
    "Model": "Generic",
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": false,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}
```

kang235, 2017-04-19:
Hi @Timebutt Thanks for the reply.
Yes, I am running the SITL simulation on a remote linux machine (128.210.171.141). Unreal and Qgc are running locally. It seems my SITL simulation can respond correctly.

jonyMarino, 2021-06-01:
This seems to be a QGroundControl issue at that time.
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem.
Thanks!

closed 2021-06-01 by jonyMarino

tanyaapaul, 2023-02-05:
same issue here if any one found a solution please share with us