microsoft/AirSim #141: Xbox controller not detected by AirSim

saihv, 2017-04-03:
I am currently experimenting with the RosFlight mode currently available in AirSim, but I'm having problems connecting my Xbox controller to the sim. If I simulate the joystick stick values (hardcode them in the code, for example), I can see that the vehicle is responding to them, but if I connect an actual controller, it is never detected. Especially this line:  

`auto dwResult = XInputGetState(index, &controllers_[index].state);`

never results in a SUCCESS value, and I am not sure why: because I tested with another program I have which uses XInput similarly (has the exact same line of code for checking whether a controller is connected), and that works fine, so I can confirm that the controller itself has no issues. Is another part of AirSim perhaps blocking the Xbox controller input from working properly?

saihv, 2017-04-04:
Update: Removing the Dependencies/x360ce folder solved this issue for me. Looks like x360ce is overriding the default XInput dll's in some way and breaking native Xbox controller support.

lovettchris, 2017-04-04:
Oh, that's good to know, thanks for following up.

lovettchris, 2017-04-04:
I updated the docs to say "NOT" to use x360ce.

saihv, 2017-04-04:
It might also be worth mentioning that x360ce seems to be a problem only if the user already has an Xbox controller. I remember someone once posted that they were able to use a x360ce + PS3 controller combination successfully..

lovettchris, 2017-04-05:
[Done](https://github.com/Microsoft/AirSim/commit/f32725f32033a050e0e5b0089d7c7519171685db)

closed 2017-04-05 by lovettchris

lovettchris, 2017-04-06:
Done.