microsoft/AirSim #144: Airsim crashes when pressing PLAY using SITL mode

tklinterg, 2017-04-04:
Hi, I have a similar issue, but the difference is that I get the crashing when trying use the SITL mode (but it never gets this far).

When I hit the play button in the unreal game editor then Unreal stops wokring for some unknow reason.

The process stops working at line 247 in a xthtread file

[image]

Please let me know if more information is needed.

Best regards

lovettchris, 2017-04-06:
Hopefully you are using Unreal 4.15.  Can you provide details about which operating system you are using, and which build script you used to build it?  For examploe, if you are on Linux we recommend you use clang 3.9...

tklinterg, 2017-04-06:
Hi and thanks for the swift replay.
I'm currently trying to run the Airsim using a windows 10 OS. The Airsim is run through the Unreal 4.15 physics motor. I'm using the build scripts that is in the folder when downloading the git project under the build for Windows 10. Somehow I'm suspecting that the STIL PX4 flight controller is not communicating correctly with the unreal simulator.

Let me know if I can provide more information.

Thanks again for the patience and the help.

Best regards

lovettchris, 2017-04-08:
I checked in some fixes that will check your SITL setup and report a new error if the Unreal app is not talking to SITL.  These are the settings you will need in ~/Documents/AirSim/settings.json:

{
  "Pixhawk": {
    "LocalHostIp": "127.0.0.1",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "LogViewerSendPort": 14389,
    "Model": "Flamewheel",
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": false,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}

a610402863, 2017-04-10:
I  can't  make "make posix_sitl_default",it shows " command not found",I don't  know  what   can  i   do?

lovettchris, 2017-04-10:
See [PX4 development](https://github.com/Microsoft/AirSim/blob/master/docs/px4.md).

tklinterg, 2017-04-10:
Hi chris 

Tried with the new settings file. Strange this is that when trying to hit the play button in the Unreal editor, the same error still remains. 

I still see that there seems to be no connection between the SITL mode in px4 and the Unreal engine, as px4 is still waiting for the required input from the engine (or vice versa)

[image]
.

I'm sorry to say, but I'm rather a noob in this area, is there a way to generate a more through report then the pictures that I'm attaching to further pin point the problem ? 

Thanks in advance. 

Best regards

lovettchris, 2017-04-10:
Did you do step 6. of [SITL](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md) ?

namely: Now edit [settings file ](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md) with UdpIp address 127.0.0.1 and UdpPort 14560, set UseSerial to false and set SitlIp to 127.0.0.1 and SitlPort to 14556.

lovettchris, 2017-04-11:
Also, did you get past the crash you reported at the beginning of this thread?

tklinterg, 2017-04-12:
Hi Chris, 

Unfortunately I've not succeed in going pass the initial crash. The last step has also been included in the modification of the settings file. The crash happens in the exact same place that is in line 247 in the xthtread file.

Best regards

lovettchris, 2017-04-12:
Since you are on Windows 10, can you try this [pre-built binary](https://1drv.ms/f/s!AoS-ymEfLuCLpeRlqa8O5GF0pzgjog)?
The see the "Cooked" folder for the pre-built executable.  Let me know if this works, or not, thanks.

lovettchris, 2017-04-17:
I checked in a possible workaround for the crash.  See [commit 43baa4f](https://github.com/Microsoft/AirSim/commit/43baa4f8f514c09fd0b38ff484006d4d408daf74).
This disables the RPC server via settings.json.  This seems to to be a workaround for the crash.

tklinterg, 2017-04-18:
Hi Chris,

I downloaded the new version of the project and the work around seems to
work. I'm now able to get the drone when hitting play as well as the small
communications messages in the bash command prompt.

Thank you so much for all the help.

Best regards

closed 2017-04-18 by tklinterg

lovettchris, 2017-04-18:
awesome, thanks for trying it.