microsoft/AirSim #146: Issue compiling AirSim plugin in Linux

AlexandreBorowczyk, 2017-04-05:
Hi,

I've following the instruction to getting AirSim running on Linux. I've built UE4 and AirSim without any issues. However, I'm running into problems when trying to add the plugin to the project. Here is where I am:
- Created a new C++ basics code project
- Opened with UE4 Editor, Editor compiles some code for a while then opens the project
- Copied the AirSim plugin into the project
- Modified the project's uproject file to add the plugin.
- Tried to run: 
`./GenerateProjectFiles.sh -project="/home/user/Documents/Unreal\ Projects/MyProject/MyProject.uproject" -game -engine`
-Getting the following: 
`Couldn't find module rules for module 'AirSim'"`

lovettchris, 2017-04-05:
so check the folder layout, you should have home/user/Documents/Unreal\ Projects/MyProject/Plugins/AirSim/AirSim.uplugin ?

AlexandreBorowczyk, 2017-04-06:
Good catch the files seems to be corrupted. I've fixed it and the GenerateProjectFiles.sh worked.
What are my options to compile the plugin? Can I just start UE4 editor and let it do the work?

Thanks

lovettchris, 2017-04-06:
[This page ](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_demo.md) walks you through the specific steps required to get AirSim working in an Unreal project.

AlexandreBorowczyk, 2017-04-07:
Thx, I've made good progress. Since the issue I've ran into is tracked by https://github.com/Microsoft/AirSim/issues/112 I will close this one.

closed 2017-04-07 by AlexandreBorowczyk