microsoft/AirSim #148: ERROR: UBT ERROR:

qhl722, 2017-04-07:
Hi,

I'm trying to compile the Airsim in ubuntu 16.04. In the final stage of compiling the AirSim plugin got the following error:
 Failed to produce item: AirSim/Unreal/MyProject/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so

In addition, the Unreal engine got the error 
UnrealEngine/Engine/Binaries/Linux/libUE4Editor-Kismet.so: undefined symbol: _ZN26FGraphEditorDragDropAction9ConstructEv

cannot restart again. Any assistance is welcomed. Thank you.

lixiao89, 2017-04-08:
I'm getting the same error. Rebuilding Unreal solved the restarting issue (or simply backup a copy of the folder UnrealEngine/Engine/Binaries and recover afterwards). However no idea as to how to get libUE4Editor-AirSim.so. I'm attaching my error message.

[image]

lovettchris, 2017-04-08:
I think I just fixed these errors, can you try again?

As for the libUE4Editor-AirSim.so, this should just get built by the generated make files.  On windows this is automatically built from the generated vcxproj files as a result of the plugin being installed correctly.  Make sure the location of the AirSim plugin is correct.  The foldering containing *.uproject should contain Plugins/AirSim/ and the .uproject should reference the plugin with 
````
,
	"Plugins": [
		{
			"Name": "AirSim",
			"Enabled": true
		}
	]
````

lixiao89, 2017-04-08:
I'm building on Ubuntu 16.04. But the previous errors are resolved to the following new ones.
[image]

madratman, 2017-04-08:
@lixiao89 you can change that line in MultRotorConnector from `std::exception` to `std::runtime_error` and remove the semicolon at end of the `endif`. You might find discussion https://github.com/Microsoft/AirSim/issues/112 or this branch in my fork helpful https://github.com/madratman/AirSim/tree/gigabyte_working

anyway, made a PR @lovettchris. Still couldn't get it working as mentioned in #112 last comment :(

lovettchris, 2017-04-08:
Thanks, those are now fixed in [97d17ab2](https://github.com/Microsoft/AirSim/commit/97d17ab25a3b87b4d7141e6841548313eb9d2f77).  Weird that ./build.sh is not catching some of those errors.  What version of CLang are you using?  I guess the make files generated by Unreal have different compiler options...

qhl722, 2017-04-09:
@lovettchris Thank you for your reply, problem solved through change folder permission

lixiao89, 2017-04-11:
Thanks, I've also been able to build successfully using [97d17ab2](https://github.com/Microsoft/AirSim/commit/97d17ab25a3b87b4d7141e6841548313eb9d2f77).  Now moving on to the crash problem which I think is discussed in other issues. Also not sure how to import an environment (like LandscapeMountains) in Ubuntu since Unreal launcher is not available in Linux.

lovettchris, 2017-04-11:
Does this help ? [https://wiki.unrealengine.com/Linux_Demos](https://wiki.unrealengine.com/Linux_Demos).

lixiao89, 2017-04-11:
Not sure how I would go about loading these environments into UE4Editor. Do I need to unpack the .pak file to get the maps?

closed 2017-06-10 by lovettchris