microsoft/AirSim #154: Where are the depth and segmentation computations being done?

saihv, 2017-04-10:
I was wondering where the code that's responsible for converting the scene to depth map and segmentation images is in the project. I was able to track the source of that data to the Unreal render target assets but I am not sure which part of the code controls those views.

lovettchris, 2017-04-10:
Good question.  This is not setup using code, it is setup using the Unreal editor in the AirSim blueprint.  To see it you have to load the Unreal editor with AirSim plugin, select "View Options" enable "Plugin Content", open Content Browser, expand Air Sim Content, select Blueprints, then in the list of blueprints you see there, double click "BP_PIPCamera", and when this loads you will see "Components" listed there, with "DepthCaptureComponent", "SegmentationCaptureComponent" and "SceneCaptureComponent".  If you double click DepthCaptureComponent you will see the Details, for example, the field of view is 90.  These are the capture components.  You can see the code accessing these components with this code:
````
UAirBlueprintLib::GetActorComponent<USceneCaptureComponent2D>(this, TEXT("DepthCaptureComponent"));
````
Then we need a render target, and you can see the code doing this:
````
ConstructorHelpers::FObjectFinder<UTextureRenderTarget2D> depth_render_target_finder(TEXT("TextureRenderTarget2D'/AirSim/HUDAssets/DepthRenderTarget.DepthRenderTarget'"));
````
If you look in the UE editor for that asset you will see how the depth information is extracted from the captured scene using the special input called:

[image]

The segmentation material uses "CustomStencil", which renders the stencil colors setup in code (AFlyingPawn::setStencilIDs).

saihv, 2017-04-11:
Thanks a lot for the detailed answer, @lovettchris! The reason I brought this up is because I was trying to look into more efficient ways of logging image data (without affecting the game thread too much), and I've been hitting a few roadblocks. I was wondering if the code is accessing the raw pixel data somewhere to compute the depth and segmentation maps 'manually', but I guess that's not the case, as they're being rendered within Unreal similar to the scene.

As of now, I am debugging a multithreaded approach that allows for camera images to be logged on a separate thread (accessing GameThread->CameraDirector->getCamera() repeatedly from a dedicated logging thread at a certain frequency, obtaining the render targets and executing ReadPixels() in a non-blocking way). I would be interested to hear your thoughts if any.

lovettchris, 2017-04-11:
Yes, the actual rendering is done on the GPU using a custom shader generated by Unreal.  I've been searching around for an answer too, so I'm not ignoring you :-) Unfortunately, I'm not finding any easy answers.  I can capture the whole screen using Camtasia Studio just fine.  I could probably plug in at the DirectX level (see  D:\Program Files\Epic Games\UE_4.15\Engine\Source\Runtime\D3D12RHI\Private\D3D12Device.h) and setup a [SwapChain ](https://msdn.microsoft.com/en-us/library/windows/desktop/bb206356(v=vs.85).aspx) to do it.  Or go even lower level and use the [NVIDIA Capture SDK](https://developer.nvidia.com/capture-sdk) to do it... unfortunately none of these are simple... and all have cross-platform disadvantages...

saihv, 2017-04-11:
I see: thanks again for the info! Closing this question for now.

closed 2017-04-11 by saihv

saihv, 2017-04-14:
@lovettchris 

Just a quick update, I was finally able to get the multithreaded approach working for recording images to disk, not without some changes to the existing settings in AirSim though. First was to turn HDR off on the scene capture (this halves the size of the texture, removes stuttering in the rendering thread, but to improve image quality I had to bump up the gamma in the render target settings): and I also had to expose some class members that are currently set to private so that the new thread can access them. On my computer, I am seeing around 10 images saved per second while the game thread remains at ~120 FPS. 

If this is something you might be interested in, please let me know and I can submit my modifications as a pull request for review and more testing. I am still investigating if it's possible to optimize it further so that HDR can be safely enabled as well.

lovettchris, 2017-04-14:
Sounds awesome, please submit a pull request :-)