microsoft/AirSim #156: In windows, airsim start normally but immediately crash

qhl722, 2017-04-14:
Setup the windows environment follow the demo successfully, but the unreal crashed after load in the airsim plugin. Is it because did not connect to the px4 controller? Is the flight controller connection mandatory?

[image]

athersec, 2017-04-14:
My project also crash when I run ".exe". However, it works in the UE Editor. And I don't know why.

WorldEditors, 2017-04-14:
I have the similar problem,  it runs ok with other game mode, but immediately crush in "SimGameMode", it did not even show the drone, but crash right after I clicked the button

lovettchris, 2017-04-14:
Yes, there is a problem and I've been trying to debug it too, but it seems to only work if you select the build configuration "DebugGame Editor", in VS:
[image]

Then use F5 to launch the Unreal Editor under debugger, then make Unreal Editor use the same build configuration by selecting it here:

[image]

qhl722, 2017-04-15:
OK, problem solved by selecting the build configuration "DebugGame Editor". Can close now

closed 2017-04-15 by qhl722