microsoft/AirSim #157: How to actually run AirSim plugin on Linux?

EliaTarasov, 2017-04-14:
There is clear instruction how to run AirSim on Windows, but there is no similar manual for Linux.
1. Linux users cannot use the Launcher to follow instruction for Windows.
2. Linux users can download the same demo https://wiki.unrealengine.com/Linux_Demos, but it's unclear how to use it with AirSim.

Since it is impossible to create project from the scratch in a Windows way, i tried to create an empty project and somehow imported LandscapeMountains demo into it.

I tried to follow video instruction "AirSim Setup Video" but after i copied "Assets" and "Maps" directories from "/LandscapeMountains/Content/" and run UE4Editor, it crushed with error:
"BP_Boost_Ring.uasset. Invalid BufferCount=0. File is most likely broken"

Can anybody provide an example how to use AirSim on Linux?

AlexandreBorowczyk, 2017-04-14:
You can follow the instructions in [unreal_demo](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_demo.md).
You don't need to use the LandscapeMountains, I use the basic c++ project.
Just use

` rsync -a -t pathtoAirsim/Plugin pathtoUEproject `

to copy the plugin. And 

`./GenerateProjectFiles.sh -project="/home/user/Documents/Unreal\ Projects/MyProject/MyProject.uproject" -game -engine`

to generate the project files. Then you can build with the method you prefer. After that you can follow the guide as for Windows.

If you manage to get it working, please let me know, as I run into issue when comes time to "run".

EliaTarasov, 2017-04-15:
Thanks for the help!
The only reason why i tried LandscapeMountains was that it had .uproject file, and therefore it is possible to set it up close to Windows way. 
Yes, i generated project files, but it produces basically files for 2 IDE: .pro for Qt Creator and .workspace for the Eclipse. To use Qt Creator i have to compile Qt with Clang, and it looks too little too much for this particular task. 
May i ask what IDE do you use to compile generated project files?

AlexandreBorowczyk, 2017-04-15:
Hi, 
I'm not sure I understand the part with the .uproject, I'm not a expert in UE4 but I would expect all project to have one. For the "generated project file" script, it will also update the build instruction. Tto use QT creator you can should the steps [here](https://wiki.unrealengine.com/Running_On_Linux#Qt). However you should be able to build my typing `make MyProjectNameEditor`

EliaTarasov, 2017-04-15:
Hey, thanks for the help again.
Let me rephrase the question, maybe i messed it up.
To quick start, it's better to follow the "Walkthrough Demo Video" which is for Windows only.
The first step is to somehow download the LandscapeMountains and open it in UE4Editor.
However, my UE4Editor failed to load it and crashed with exception.
On the other hand, it can create an empty C++ project and load Airsim plugin successfully.
So there are 2 options: 
1. Something wrong with Linux examples for the current release branch of UE4Editor.
2. I did something wrong with my current build and i should fix that.

That's why i asked does anybody successfully load at least one of the default Linux demos?
And if not, then why?
For instance, the answer may be: "Because it doesn't matter and if AirSim loads int Editor ok and doesn't crush when Play is pressed then it's ok and you can start develop your app with this plugin"

AlexandreBorowczyk, 2017-04-15:
Ok, if I understand correctly you have the [same issue](https://github.com/Microsoft/AirSim/issues/112) a few of us are having: when you switch the game mode to SimGameMode, UE4Editor crash when you try to run?

EliaTarasov, 2017-04-15:
Yes, exactly the same.
But in addition, im unable to load any demo, while @madratman did it successfully.

EliaTarasov, 2017-04-15:
Relevant part of the crash log is here:

2017.04.15-18.02.51:418][351]LogPlayLevel: Creating play world package: /Temp/UEDPIE_0_Untitled_1
[2017.04.15-18.02.51:422][351]LogPlayLevel: PIE: StaticDuplicateObject took: (0.003838s)
[2017.04.15-18.02.51:423][351]LogAIModule: Creating AISystem for world Untitled_1
[2017.04.15-18.02.51:423][351]LogPlayLevel: PIE: World Init took: (0.001357s)
[2017.04.15-18.02.51:424][351]LogPlayLevel: PIE: Created PIE world by copying editor world from /Temp/Untitled_1.Untitled_1 to /Temp/UEDPIE_0_Untitled_1.Untitled_1 (0.005610s)
[2017.04.15-18.02.51:631][351]LogCrashTracker: 

[2017.04.15-18.02.51:632][351]LogLinux: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000002

[2017.04.15-18.02.51:633][351]LogLinux: Fatal error!

[Callstack]  00  0x00007f8f2da808bf  FLinuxPlatformStackWalk::CaptureStackBackTrace(unsigned long long*, unsigned int, void*)
[Callstack]  01  0x00007f8f2d93ac95  FGenericPlatformStackWalk::StackWalkAndDump(char*, unsigned long, int, void*)
[Callstack]  02  0x00007f8f2da38671  FLinuxCrashContext::CaptureStackTrace()
[Callstack]  03  0x00007f8f22a81340  CommonLinuxCrashHandler(FGenericCrashContext const&)
[Callstack]  04  0x00007f8f2da3d572  PlatformCrashHandler(int, siginfo_t*, void*)
[Callstack]  05  0x00007f8f33865390  /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7f8f33865390]
[Callstack]  06  0x00007f8f2d9cc768  FMallocBinned::Private::FreeInternal(FMallocBinned&, void*)
[Callstack]  07  0x00007f8eb709a4da  std::__1::basic_filebuf<char, std::__1::char_traits<char> >::~basic_filebuf()
[Callstack]  08  0x00007f8eb70949d1  ASimModeWorldBase::~ASimModeWorldBase()
[Callstack]  09  0x00007f8f2d10b1a6  IncrementalPurgeGarbage(bool, float)
[Callstack]  10  0x00007f8f2d10e21e  CollectGarbageInternal(EObjectFlags, bool)
[Callstack]  11  0x00007f8f2d10f227  CollectGarbage(EObjectFlags, bool)
[Callstack]  12  0x00007f8f29958f8d  UGameInstance::InitializeForPlayInEditor(int, FGameInstancePIEParameters const&)
[Callstack]  13  0x00007f8f2451fb35  UEditorEngine::CreatePIEGameInstance(int, bool, bool, bool, bool, float)
[Callstack]  14  0x00007f8f2451ad44  UEditorEngine::PlayInEditor(UWorld*, bool)
[Callstack]  15  0x00007f8f2450b0ca  UEditorEngine::StartQueuedPlayMapRequest()
[Callstack]  16  0x00007f8f23efe8ff  UEditorEngine::Tick(float, bool)
[Callstack]  17  0x00007f8f248d0e99  UUnrealEdEngine::Tick(float, bool)
[Callstack]  18  0x0000000000423de0  FEngineLoop::Tick() [Runtime/Core/Public/Stats/Stats2.h, line 1686]
[Callstack]  19  0x000000000042b923  GuardedMain(wchar_t const*) [/home/elia/Documents/Unreal/UnrealEngine/Engine/Source/Runtime/Launch/Private/Launch.cpp, line 62]
[Callstack]  20  0x00007f8f22a820ca  CommonLinuxMain(int, char**, int (*)(wchar_t const*))
[Callstack]  21  0x00007f8f21dfa830  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f8f21dfa830]
[Callstack]  22  0x0000000000415e29  ./UE4Editor(_start+0x29) [0x415e29]

EliaTarasov, 2017-04-16:
This one is closed since it duplicates  #112 [](url)

closed 2017-04-16 by EliaTarasov