microsoft/AirSim #158: The purple led of pixhawk flashes when the AirSim starts

athersec, 2017-04-16:
I can fly paper plane in the QGroundControl. But when I run the AirSim, purple led of pixhawk starts flashing, and airsim gives the message " data link#0 regained" and " GPS #0 regaind". So the drone can't arm and take off. The xacc of "px4 log" looks like this:
[image]

Anyone meets this problem ?
And how can I resolve this problem?
Any help will be appriciated!

lovettchris, 2017-04-16:
The xacc signal looks good actually, and "data link#0 regained" and " GPS #0 regained" are both expected messages that mean the simulation mode has resumed on the Pixhawk.  In otherwords the Pixhawk is getting simulated sensor data from AirSim.  What we need to do is debug why the "arm" and "takeoff" doesn't work.   You said you can fly paper plane in the QGroundControl, question is with what simulator?  Normally PX4 won't fly unless it is running on a real drone, or it is running in simulation mode with a real simulator connected providing simulated sensor data ?  Were you using jmavsim or some other simulator?  Or was AirSim connected at the time and QGroundControl was flying that drone?  To debug this further I could use the logs outlined in our [logging docs](https://github.com/Microsoft/AirSim/blob/master/docs/logging.md).

rennsix, 2017-04-17:
I have the same type of issue. I have the purple light flashing but I can arm and disarm. However, it never gets off the ground...See video.  https://youtu.be/xAJta5Yoivk

Still working on viewing the logs and what, if anything, those would say.

athersec, 2017-04-18:
@rennsix What's your equpments' information? 
and my equpments' information:
Operating System: Windows 10 64bit
CPU: Intel Core i5
GPU: Nvidia GTX 1080
RAM: 8 gb
Flight Controller: Pixhawk v2
Remote Control: Walkera dev10

rennsix, 2017-04-18:
Win7 pro
i7
GTX 1070
32 ram
Pix v2
3d robotics iris remote

lovettchris, 2017-04-18:
By the way, here's the [pixhawk led color definitions](https://pixhawk.org/users/status_leds).  I think the purple led is supposed to be "blue" and it means the pixhawk has lost the GPS lock.  In my case I get slow green LED blink when drone is disarmed, and a faster green blink when drone is flying.  Sometimes I also see the LED turn blue and rapid, but it is still flying ok.  So that might be a red herring.  We may need to check the firmware default GPS location versus simulator default location.  When AirSim is running you should be able to connect QGroundCOntrol and see what location it has.  You should see the Microsoft Building 99 on your map, with no flicker or jumping about, can you check if this is what you see?
[image]
You should also be able to arm, takeoff and fly the AirSim drone using QGroundControl... it would be interesting to see if that works.

lovettchris, 2017-04-18:
One more idea, try turning off the camera views - they might be taking up too much CPU/GPU.  Click on the AirSim canvas and press digit "0".  The other thing you should do is turn on the FPS tracker and let me know what FPS you are seeing when you click Play:
[image]

On LandscapeMountains I see 27 with all 3 views open, and 92 when they are off.  I also do this to ensure that AirSim still gets lots of CPU when it is not the active window:

[image]

rennsix, 2017-04-18:
Thanks. I'll test tomorrow. Any chance of an engine plugin?

lovettchris, 2017-04-18:
Not sure what the process is for engine plugins.  But if you mean publishing AirSim in their [marketplace](https://www.unrealengine.com/marketplace/content-cat/assets/codeplugins), it is on our list of things to do...

athersec, 2017-04-18:
@lovettchris I think I know the reason.
The FPS just  2 or 3, sometimes it can be 15. So my drone can't fly just because the hardware's property is too bad.
I'll change a better computer tomorrow.
Thanks all of you.

rennsix, 2017-04-20:
It's always a little embarrassing to admit when you haven't followed the instructions. As it turns out, while re-watching the AirSimDemo video, I noticed the part where Chris mentions that the latest PX4 Firmware has an issue with HIL mode. Then I noticed his firmware said 255.255.255 in QGroundControl and mine was 1.6.0rc. Doh!!! I flashed with v1.4.4 and all issues went away. 

Here is a quick video with the LandscapeMountains executable. Now it's time to attempt to get it setup in the UE4 editor.. [https://youtu.be/6EOXtFl9aGI](url)

lovettchris, 2017-04-20:
Congrats, that looks great !!  Careful, your productivity might plummet now that you have a fun drone to fly :-) now press F1 to see some simulator options.  As for versions, the very latest pixhawk firmware should also work fine now if you build it from the GitHub.com/px4/firmware master branch.

athersec, 2017-04-25:
Thanks!
When I flash with v1.4.4, the drone can take off!!
Now the issue can be closed.

closed 2017-04-25 by athersec