microsoft/AirSim #160: Dependency issue with format.cc (rpclib)

parasange, 2017-04-17:
I'm having a problem of "expected unqualified-id" in format.cc with GCC 6.3.1 (linux410).
[image]

Somehow AirSim just fetches an old commit of rpclib (`a5dfa63ba1e52d637bd147d47e0ee6418acfc4d3`). Building rpclib seperatley worked for me, `git rev-parse HEAD` now gives me then the updated `f3f45d3bf1a72866056c9df4f9da902af1787995`
But when using it to build AirSim (copied rpclib to AirSim/external/rpclib), it still throws that error.
For AirSim, `git rev-parse HEAD` gives me `43baa4f8f514c09fd0b38ff484006d4d408daf74` which should be correct.

sztomi, 2017-04-17:
(as the author of rpclib) This was a breakage that was introduced by a libstdc++ update where a name clash happened. This was fixed in rpclib and I suggest upgrading to the latest release (2.0.0). The upgrade should be easy, the interface if backwards compatible (2 functions were added).

lovettchris, 2017-04-17:
Hey Tamás!

I updated AirSim to your latest rpclib bits. I notice it also builds more cleanly in clang strict mode. Thanks!  The only warning we get now are these:
```
rpclib/include/rpc/config.h:14:11: warning: empty paragraph passed to '\brief' command [-Wdocumentation] 
rpclib/include/rpc/detail/make_unique.h:13:5: warning: 'RPCLIB_CXX_STANDARD' is not defined, evaluates to 0
````

parasange, 2017-04-18:
Thanks, worked for me!

closed 2017-04-18 by parasange

sztomi, 2017-04-18:
@lovettchris Thanks for bringing my attention to this. Is strict mode `-pedantic`?

sytelus, 2017-04-19:
The strict mode is all of these:

`-ggdb  -Wall -Wextra -Wstrict-aliasing -Wunreachable-code -Wcast-qual -Wctor-dtor-privacy -Wdisabled-optimization -Wformat=2 -Winit-self -Wmissing-include-dirs -Wold-style-cast -Woverloaded-virtual -Wredundant-decls -Wshadow -Wstrict-overflow=5 -Wswitch-default -Wundef -Wno-unused -Wno-variadic-macros -Wno-parentheses -fdiagnostics-show-option`

More info at: http://shitalshah.com/p/how-to-enable-and-use-gcc-strict-mode-compilation/