microsoft/AirSim #163: Missing or incompatible modules in AirSim plugin - AirSim on Ubuntu 16.04

ProjectVenom, 2017-04-18:
Hi,

I followed the installation instructions for Linux referenced in your instructions on GitHub, installed Eigen, and the UnrealEngine.

I setup an empty project in unreal, and then ran 

The first error message that appears says:
"Missing or incompatible modules in AirSim plugin -- would you like to disable it?  You will no longer be able to open any assets crated using it"

After saying "NO" - the project tries to load AirSim, but then raises the following errors:

`[2017.04.18-18.11.23:917][  0]LogLinker:Warning: Can't resolve asset name
1001[2017.04.18-18.11.23:917][  0]LogLinker:Warning: Can't resolve asset name
[2017.04.18-18.11.23:917][  0]LogUObjectGlobals:Warning: Failed to find object 'MaterialInterface /Paper2D/MaskedUnlitSpriteMaterial.MaskedUnlitSpriteMaterial'
CDO Constructor (PaperFlipbook): Failed to find /Paper2D/MaskedUnlitSpriteMaterial
[2017.04.18-18.11.23:917][  0]Error: CDO Constructor (PaperFlipbook): Failed to find /Paper2D/MaskedUnlitSpriteMaterial

[2017.04.18-18.11.23:918][  0]Error: CDO Constructor (PaperFlipbook): Failed to find /Paper2D/MaskedUnlitSpriteMaterial

1001[2017.04.18-18.11.23:918][  0]LogLinker:Warning: Can't resolve asset name
CDO Constructor (PaperSprite): Failed to find /Paper2D/MaskedUnlitSpriteMaterial
[2017.04.18-18.11.23:918][  0]Error: CDO Constructor (PaperSprite): Failed to find /Paper2D/MaskedUnlitSpriteMaterial

[2017.04.18-18.11.23:919][  0]Error: CDO Constructor (PaperSprite): Failed to find /Paper2D/MaskedUnlitSpriteMaterial

1001[2017.04.18-18.11.23:919][  0]LogLinker:Warning: Can't resolve asset name
1001[2017.04.18-18.11.23:919][  0]LogLinker:Warning: Can't resolve asset name
[2017.04.18-18.11.23:919][  0]LogUObjectGlobals:Warning: Failed to find object 'MaterialInterface /Paper2D/OpaqueUnlitSpriteMaterial.OpaqueUnlitSpriteMaterial'
CDO Constructor (PaperSprite): Failed to find /Paper2D/OpaqueUnlitSpriteMaterial
[2017.04.18-18.11.23:919][  0]Error: CDO Constructor (PaperSprite): Failed to find /Paper2D/OpaqueUnlitSpriteMaterial

[2017.04.18-18.11.23:919][  0]Error: CDO Constructor (PaperSprite): Failed to find /Paper2D/OpaqueUnlitSpriteMaterial

1001[2017.04.18-18.11.23:920][  0]LogLinker:Warning: Can't resolve asset name
1001[2017.04.18-18.11.23:920][  0]LogLinker:Warning: Can't resolve asset name
[2017.04.18-18.11.23:920][  0]LogUObjectGlobals:Warning: Failed to find object 'PaperTerrainMaterial /Paper2D/DefaultPaperTerrainMaterial.DefaultPaperTerrainMaterial'
CDO Constructor (PaperTerrainComponent): Failed to find /Paper2D/DefaultPaperTerrainMaterial
[2017.04.18-18.11.23:920][  0]Error: CDO Constructor (PaperTerrainComponent): Failed to find /Paper2D/DefaultPaperTerrainMaterial

[2017.04.18-18.11.23:920][  0]Error: CDO Constructor (PaperTerrainComponent): Failed to find /Paper2D/DefaultPaperTerrainMaterial

1001[2017.04.18-18.11.23:921][  0]LogLinker:Warning: Can't resolve asset name
CDO Constructor (PaperTileMap): Failed to find /Paper2D/MaskedUnlitSpriteMaterial
[2017.04.18-18.11.23:921][  0]Error: CDO Constructor (PaperTileMap): Failed to find /Paper2D/MaskedUnlitSpriteMaterial

[2017.04.18-18.11.23:921][  0]Error: CDO Constructor (PaperTileMap): Failed to find /Paper2D/MaskedUnlitSpriteMaterial

1001[2017.04.18-18.11.23:921][  0]LogLinker:Warning: Can't resolve asset name
CDO Constructor (PaperTileMapComponent): Failed to find /Paper2D/MaskedUnlitSpriteMaterial
[2017.04.18-18.11.23:922][  0]Error: CDO Constructor (PaperTileMapComponent): Failed to find /Paper2D/MaskedUnlitSpriteMaterial

[2017.04.18-18.11.23:922][  0]Error: CDO Constructor (PaperTileMapComponent): Failed to find /Paper2D/MaskedUnlitSpriteMaterial

[2017.04.18-18.11.23:922][  0]LogUObjectBase:Warning: -------------- Default Property warnings and errors:
[2017.04.18-18.11.23:922][  0]LogUObjectBase:Warning: Error: CDO Constructor (PaperFlipbook): Failed to find /Paper2D/MaskedUnlitSpriteMaterial

[2017.04.18-18.11.23:922][  0]LogUObjectBase:Warning: Error: CDO Constructor (PaperSprite): Failed to find /Paper2D/MaskedUnlitSpriteMaterial

[2017.04.18-18.11.23:922][  0]LogUObjectBase:Warning: Error: CDO Constructor (PaperSprite): Failed to find /Paper2D/OpaqueUnlitSpriteMaterial

[2017.04.18-18.11.23:922][  0]LogUObjectBase:Warning: Error: CDO Constructor (PaperTerrainComponent): Failed to find /Paper2D/DefaultPaperTerrainMaterial

[2017.04.18-18.11.23:922][  0]LogUObjectBase:Warning: Error: CDO Constructor (PaperTileMap): Failed to find /Paper2D/MaskedUnlitSpriteMaterial

[2017.04.18-18.11.23:922][  0]LogUObjectBase:Warning: Error: CDO Constructor (PaperTileMapComponent): Failed to find /Paper2D/MaskedUnlitSpriteMaterial
`
What am I missing? Is this an AirSim or an Unreal problem?

sytelus, 2017-04-19:
Are using Unreal Engine 4.15?

GPrathap, 2017-05-18:
I am having the same issue. I used [release branch ](https://github.com/EpicGames/UnrealEngine/tree/release). Is there any specific branch which is required for this plugin. 

Thanks,
Geesara

isvogor-foi, 2017-06-01:
Hello. I had a similar issue, and I fixed it by exporting the EIGEN_ROOT (as it says in the guidelines) in the terminal from which you start. /UE4Engine. 
Ivan

lovettchris, 2017-06-10:
closing old issue, please reopen if you are still seeing crashes.

closed 2017-06-10 by lovettchris