microsoft/AirSim #165: Mac Support - still getting errors

hayaalsh, 2017-04-19:
I am still getting this error when I build  `./build.sh` even after pulling the recent version. Does any one knows why? Am I building the wrong file?

```
/Users/aa/Github/AirSim2/external/rpclib/lib/rpc/this_server.cc:7:12: error: thread-local storage
      is not supported for the current target
    static thread_local this_server_t instance;
           ^
1 error generated.
make[3]: *** [rpclib/CMakeFiles/AirSim-rpclib.dir/lib/rpc/this_server.cc.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/Users/aa/Github/AirSim2/external/rpclib/lib/rpc/this_session.cc:7:12: error: thread-local
      storage is not supported for the current target
    static thread_local this_session_t instance;
           ^
1 error generated.
make[3]: *** [rpclib/CMakeFiles/AirSim-rpclib.dir/lib/rpc/this_session.cc.o] Error 1
/Users/aa/Github/AirSim2/external/rpclib/lib/rpc/this_handler.cc:10:12: error: thread-local
      storage is not supported for the current target
    static thread_local this_handler_t instance;
           ^
1 error generated.
make[3]: *** [rpclib/CMakeFiles/AirSim-rpclib.dir/lib/rpc/this_handler.cc.o] Error 1
/Users/aa/Github/AirSim2/external/rpclib/lib/rpc/detail/server_session.cc:11:27: warning: unused
      variable 'default_buffer_size' [-Wunused-const-variable]
static constexpr uint32_t default_buffer_size = 65535;
                          ^
1 warning generated.
make[2]: *** [rpclib/CMakeFiles/AirSim-rpclib.dir/all] Error 2
make[1]: *** [AirLib/CMakeFiles/AirLib.dir/rule] Error 2
make: *** [AirLib] Error 2
```

hayaalsh, 2017-04-19:
Update: I have updarted my xcode and now the error is /Users/aa/Github/AirSim/MavLinkCom/MavLinkTest/main.cpp:901:21: error: use of undeclared identifier 'LoadInitScript'
                    LoadInitScript(fileName);

Do I have to connect the pixhawk before building the file?

lovettchris, 2017-04-19:
Ah, conditional compile error for the case when __cpp_lib_experimental_filesystem is not defined.  I pushed a [fix](https://github.com/Microsoft/AirSim/commit/5eed92b824a32c046063df456343fd3a254864ad).

hayaalsh, 2017-04-19:
Thanks it built successfully 👍

lovettchris, 2017-04-19:
Cool.

closed 2017-04-19 by lovettchris