microsoft/AirSim #166: How to enable AirSim::RpcLibServer on the specified IP and Port?

EliaTarasov, 2017-04-20:
Hi,

There is "RpcEnabled" field within settings.json file, but it's unclear from both the c++ code and the AirSim docs how to enable it on the specified IP and port.

Can anybody give a hint how to do it?
Thanks for the help!

lovettchris, 2017-04-20:
The RpcEnabled switch is just a master switch to enable or disable the RpcLibServer.cpp code inside AirSim (you'll see this in MultiRotorConnector::startApiServer).  When enabled, it uses the "LocalHostIp" as the server address.  The port is hard coded to 41451.  If that is a problem we could add a setting for that too.  If you want to run DroneShell.exe on a different machine has a command line argument "-server <ipaddress>" (no port#) which you can use to connect to AirSim running on another machine. The default is localhost (127.0.0.1).

EliaTarasov, 2017-04-26:
Ok, got it. No, it's not a problem. Thanks for explanation.

closed 2017-04-26 by EliaTarasov

knmitri, 2018-11-09:
Hey  @lovettchris, yeah that would be useful to train in parallel through different ports. 
Can it be changed from the settings file ?