microsoft/AirSim #167: In C++ how to get the value of  Collision Count:

Focke-Wulf, 2017-04-20:
Hello All:
When i am testing the Drone, i am trying to program a simple AI which can control the Drone ignore the Collision with the walls.

I find Plugins already show the Collision Count value which like if the Drone crash and it will be counted.

is any possible for me to get the Collision Count function in my program? 

Thanks for the help!

Focke-Wulf, 2017-04-21:
I am trying to edit based on HelloDrone.  the problem is that in HelloDrone, can we get function of Collision?

Thanks..

lovettchris, 2017-04-24:
This information is currently not being sent via the DroneControlBase API, but it could be added easily enough.

Focke-Wulf, 2017-04-24:
Thanks for the reply! .  I find DroneControlBase API
However I am not familiar with C++ code.... so i still have some problems.

and i find an example here  which from VehiclePawnBase.ccp

[image]

and i also find the code fro DroneControlBase API Which is 
[image]

i am starting to test those code but in DroneControllerBase.cpp, i cann't find the function [getPosistion() ] auto show up ... but the original code works with that...
[image]

So i am trying to edit this file like this...
[image]

Am i at the right way????

If it is right.. do i need to rebuild the Plugins in AirSim ? 

or i could just add these code in the Plugins file which already in my UE4 project???

Thanks for the HElP!!!!!!!!!

sytelus, 2017-04-29:
Search for AVehiclePawnBase::NotifyHit to see the current code that counts collisions. 

If you don't want collisons then search for below statement and set it to true.

```
bool EnablePassthroughOnCollisons = false; 
```

closed 2017-04-29 by sytelus

Focke-Wulf, 2017-04-29:
Hello: 
       Thanks for the reply... What i want is to transfer the collisons info back to the server..
       which is like we can use getPosition() at server side....