microsoft/AirSim #168: Ubuntu 16.04 Error

vatsagandhi, 2017-04-20:
I am getting the following error when trying to build the linux version of Airsim. I had it initially built on UE4 4.15 but because of the "play button issue" I cloned the git and now I am running into errors. I don't know why It spits this out because the files it complains about exist.

[errorUnreal.txt](https://github.com/Microsoft/AirSim/files/943752/errorUnreal.txt)

lovettchris, 2017-04-20:
You might be able to disable those "warnings as errors" by editing the CMAKE_CXX_FLAGS  in cmake/cmake-modules/CommonSetup.cmake.  I don't see any recent changes in there that could break anything and I just tried a clean build on Ubuntu 16.04 and it builds fine, using ~/build.sh.  What version of CLang are you using ?  Could you try "git clean -dfx" just to be sure you don't have any weird state there?

[image]

vatsagandhi, 2017-04-20:
I am using CLang 3.9.1. I deleted everything and recloned the git. I am attaching the output from Build.sh and getlibcxx.sh.  I am gettings warnings I never got before which may be causing the issue. Additionally these were the warnings I got which did not get published into the text files.

[WarningsAirsim.txt](https://github.com/Microsoft/AirSim/files/945698/WarningsAirsim.txt)

[CmakeBuild.txt](https://github.com/Microsoft/AirSim/files/945696/CmakeBuild.txt)

[BuildOutput.txt](https://github.com/Microsoft/AirSim/files/945697/BuildOutput.txt)

lovettchris, 2017-04-20:
Ok, so the AirSim build is successful.  You can safely ignore the new rpclib warnings.  The rpclib guys are working on that, but they are harmless.

I see what is happening, unreal is unhappy with my other compiler fixes, I was trying to avoid "unused parameter" warnings by writing adding this line of code:
> assume_unit_quat; // stop warning: unused parameter.

This works in the AirSim build, but it looks like the Unreal build is unhappy with this for some reason.  I think what we have to do is figure out how to turn off these compiler switches in the unreal build:
> -Werror,-Wunused-value

I'm looking for way to do that, it should be done by adding something to AirSim.Build.cs.

lovettchris, 2017-04-21:
Looking at the windows build configuration which lives here: "D:\Program Files\Epic Games\UE_4.15\Engine\Source\Programs\UnrealBuildTool\Windows\VCToolChain.cs" it looks like they setup an object called CompileAction which has a property called CommandArguments.  They collect these arguments from all over then set this property.  We need to add to that property the arguments "-wno-unused-value".  If you search for "-wunused-value" you might find where they are setting that...

vatsagandhi, 2017-04-21:
So I am looking at the "Engine\Source\Programs\UnrealBuildTool\Linux\LinuxToolChain.cs" and it turns out that the  "-wno-unused-value" property was already set. 
[image]

lovettchris, 2017-04-21:
Ah, but that is in "Shipping" build config, try doing the same in all the other build configurations.

vatsagandhi, 2017-04-21:
So I searched through the entire engine folder for Unreal engine and I could not find "-wunused-value"  which confuses me because that is the error I am getting. Quick question, after making the basic C++ project on Unreal Engine, we modify the .uproject file by adding the airsim plugin dependencies, save it and then build the project with Airsim. Correct? I might try doing a fresh installation of UE4 4.15 again and see if that makes a difference.

[image]

lovettchris, 2017-04-21:
try removing other -w* checks like -wall, -werror, and  -pedantic etc.  These turn on all error messages, 
then try adding lots of -wno*, like -Wno-unused, etc until you find the one that disables this error.

lovettchris, 2017-04-21:
might also want to test that you are editing the right place by adding an invalid switch "-wthis_is_a_test" and make sure you get compile errors as a result.

vatsagandhi, 2017-04-21:
Sorry I keep bothering you but after reinstalling Unreal Engine and trying to create a project file with Airsim, I ran into this issue. I updated the .uproject file to take in Airsim Plugins so not sure how it failed to produce 
"libUE4Editor-AirSim.so."

[AirSimLog.txt](https://github.com/Microsoft/AirSim/files/946186/AirSimLog.txt)

lovettchris, 2017-04-21:
No problem, thanks for pushing on this.  I just [removed my change ](https://github.com/Microsoft/AirSim/commit/d2bdd577fb994e59fe194fa92d4970dfdde21057) that attempted to fix the "warning: unused parameter" warnings since it created a "expression result unused" error on Linux.  This might solve your problems.

Mayankm96, 2017-05-09:
I have been trying to run the AirSim in the Landscape Mountain environment, however, there seems to be an issue in MyProject much similar to this one. Everytime I open it, I am asked to recompile and build the `libUE4Editor-AirSim.so`. On saying yes, the process starts. However, build fails and on checking my log file I found the following error:

```
In file included from /home/mittalm/Documents/UnrealProjects/MyProject/Plugins/AirSim/Source/AirLib/include/common/Common.hpp:16:
/home/mittalm/Documents/UnrealProjects/MyProject/Plugins/AirSim/Source/AirLib/include/common/VectorMath.hpp:146:9: error: 
      expression result unused [-Werror,-Wunused-value]
        assume_unit_quat; // stop warning: unused parameter.
        ^~~~~~~~~~~~~~~~
/home/mittalm/Documents/UnrealProjects/MyProject/Plugins/AirSim/Source/AirLib/include/vehicles/MultiRotorParams.hpp:134:50: note: 
      in instantiation of member function
      'msr::airlib::VectorMathT<Eigen::Matrix<float, 3, 1, 0, 3, 1>,
      Eigen::Quaternion<float, 2>, float>::rotateVector' requested here
            rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r(0...
                                                 ^
2 errors generated.
[517/943] Compile Module.BuildPatchServices.cpp
3 errors generated.
```
I suppose the changes made aren't solving the problem @lovettchris

lovettchris, 2017-05-09:
Try now.

closed 2017-05-09 by lovettchris

Mayankm96, 2017-05-10:
Mhm the new changes worked for me! I am facing some other issue but I'll put it on a relevant thread.

Thank you! :)