microsoft/AirSim #169: Can I run this on a Azure vm ,as I dont have a good system.

Abhi001vj, 2017-04-21:


lovettchris, 2017-04-24:
We have not done it yet, but you would need [Azure VM with a good GPU](https://azure.microsoft.com/en-us/resources/videos/build-2016-introduction-to-nvidia-gpus-in-azure/). And you would need to figure out how to configure Unreal Engine to run in headless mode, and this part doesn't seem to have good answers yet on the unrealengine forums.

sytelus, 2017-04-29:
Azure recently added NC and NV series for VMs that should be able to run AirSim. Check below link for details and let us know how it goes!

https://azure.microsoft.com/en-us/pricing/details/virtual-machines/windows/  (check for NV12 or NC12).

jkuruzovich, 2017-12-09:
I gave this a try on a Windows 10 Pro machine on Azure.  I kept running into an issue indicating DX11 feature level 10.0 is require to run the engine.   
[image]

PHRABAL, 2018-12-17:
@jkuruzovich I have the same problem on an AWS Windows Server 2016 instance. Did you solve this?

ByronDev121, 2020-04-15:
Is there a solution for this?

I get the same error. "DX11 feature level 10.0 is require to run the engine."

I'm trying run AirSim on a Windows server 2019 VM on Google Cloud Platform. The VM configuration is setup with the n1-standard-2 (2 vCPUs, 7.5 GB memory) and Nvidia Tesla K80 GPU.

The latest drivers are installed for the GPUI:

[image]

I also have direct X 12 installed (as I understand it, this should be backwards compatible with 11):

[image]

[image]

 I've tried a number of potential solutions but nothing has worked so far.

guadacasuso, 2020-07-28:
Please check this https://microsoft.github.io/AirSim/azure/

closed 2020-08-28 by jonyMarino