microsoft/AirSim #170: Aircraft can't takeoff with new version of PX4(after ver:1.4.4)

liuli2008212, 2017-04-21:
I used the new version of PX4(after ver:1.4.4),my aircraft can't takeoff when I play. The aircraft is Keep rolling . But it is running normally by the version of 1.4.4 . I have tried the UE_4.15 and UE_4.14.  
Thank you for your help.

lovettchris, 2017-04-24:
Definitely need UE_4.15.  But can you try this version of the PX4 firmware? 
[px4fmu-v2_default.zip](https://github.com/Microsoft/AirSim/files/950543/px4fmu-v2_default.zip)
And make sure you set RpcEnabled to true in your ~/Documents/AirSim/settings.json file.

liuli2008212, 2017-04-24:
HI lovettchris,
Thank you for your help. I tried your version of px4fmu-v2_default.zip . My aircraft can fly within UE_4.14.Your version is 1.6.0,ok? But aircraft can't fly when I try this version of v1.6.0rc1 the PX4 firmware. What did you change at this version of v1.6.0 the PX4 firmware?

liuli2008212, 2017-04-24:
I git this version v1.6.0rc1 of the PX4 firmware.I make and upload,it's ok. But this version of https://github.com/PX4/Firmware/releases/download/v1.6.0rc1/Firmware.zip  can't fly,I don't know why,so I will use this git version.
Thank you for your help and patience.

lovettchris, 2017-04-24:
My version is using latest bits from PX4 master branch, so it is slightly newer. Hopefully the official builds will catch up soon and we won't have this problem.  Glad you are unblocked now.  Can we close this issue?

liuli2008212, 2017-04-25:
Yes,you can close it .Thank you very much.

closed 2017-04-25 by lovettchris