microsoft/AirSim #172: Unable to arm the drone

udaydlsv, 2017-04-21:
My setup : 
i7-3540M CPU @3.00 GHz
8GB RAM
Quadro K3000M
SSD harddrive
(I know my system doesn't match with the recommended setup but would like to play with this if possible)
RC : Taranis Plus & X8R
Flight controller : pixhawk mini 

Using Qgroundcontrol, I flashed px4fmu-v2_default.px4 v1.6 firmware and did select HIL Quadcopter X frame(and rebooted). When I 'play' unreal demo, I see following messages on the screen as can be seen in the png attached. 

[image]
sometimes I see the rotors moving, sometimes they are not. 
The pixhawk mini beeps when I start the demo and blinks with green main LED light. I am assuming this means Unreal finds my pixhawk.

but, I am not able to arm or takeoff the drone.

To test further, I launched droneshell and it says 
waiting for drone to report a valid GPS location..

I have also attached mavlink logs and tried things mentioned on https://github.com/Microsoft/AirSim/issues/138

Did I miss something in the setup ?

P.S : I was able to play with the setup in SITL mode using droneshell commands. it was difficult to control the drone though and it would go crazy almost all the time

lovettchris, 2017-04-24:
Logs show GPS signal is good, so perhaps all you need to do is edit your ~/Documents/AirSim/settings.json and set RpcEnabled to true:

  "RpcEnabled": true

DroneShell requires this.

tklinterg, 2017-04-26:
I have a similar problem, the droneshell states that it is waiting for the drone to report a valid GPS signal via the RPC server. Looking in the settings file I see that this flag is set to 'false', however, by settings this to 'true' the Unreal motor crashes when hitting the play button. Is there another way to get the droneshell to listen.

My aim here is go gain programmatic control over the drone but is the drone cannot listen via the RPC I see this as a potential problem.
Any idea of what I can try ? 

Thanks in advance, 

Best regards

lovettchris, 2017-04-26:
We are actively investigating that crash.  For now what we know works is to use only the "DebugGame Editor" build configuration.  This is the only one that works with RPC enabled.

You can also control the drone via mavlink - see MavLinkTest.exe for examples.  But then you don't get the nice DroneControlBase API, and you will not get images that way.

lovettchris, 2017-05-04:
We fixed a crash bug on Windows build (problem with compiler flags).  So if you were using windows, please try again, and reopen this issue if you still have problems, thanks.

closed 2017-05-04 by lovettchris

tklinterg, 2017-05-09:
Hi Chris,

Just wanted to get back and say thank you. I can now arm the drone and make
it fly by the commands entered in the Drone shell with out the Unreal motor
crashing down.

Thanks again.

All the best!