microsoft/AirSim #173: xcode: Unable to instantiate instance of 'AirSim' object type from compiled assembly

hayaalsh, 2017-04-22:
I keep getting this error when I try to build the file on MacOS using xcode. Anyone knows what is going wrong

```
Setting up Mono
Building MyProjectEditor...
2017-04-22 22:13:55.432 defaults[27282:1372178] 
The domain/default pair of (com.apple.dt.Xcode, IDEBuildOperationMaxNumberOfConcurrentCompileTasks) does not exist
Running command : Engine/Binaries/DotNET/UnrealBuildTool.exe MyProjectEditor Mac Development /Users/alsharha/Documents/Unreal Projects/MyProject/MyProject.uproject
Compiling with MacOSX SDK 10.12
ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'MyProjectModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_eigen.md
  at AirSim.AddEigenDependency () [0x00000] in <filename unknown>:0 
  at AirSim..ctor (UnrealBuildTool.TargetInfo Target) [0x00000] in <filename unknown>:0 
  at (wrapper managed-to-native) System.Reflection.MonoCMethod:InternalInvoke (System.Reflection.MonoCMethod,object,object[],System.Exception&)
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00000] in <filename unknown>:0 
  --- End of inner exception stack trace ---
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00000] in <filename unknown>:0 
  at System.Reflection.MonoCMethod.DoInvoke (System.Object obj, BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00000] in <filename unknown>:0 
  at System.Reflection.MonoCMethod.Invoke (BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00000] in <filename unknown>:0 
  at System.Activator.CreateInstance (System.Type type, BindingFlags bindingAttr, System.Reflection.Binder binder, System.Object[] args, System.Globalization.CultureInfo culture, System.Object[] activationAttributes) [0x00000] in <filename unknown>:0 
  at System.Activator.CreateInstance (System.Type type, System.Object[] args, System.Object[] activationAttributes) [0x00000] in <filename unknown>:0 
  at System.Activator.CreateInstance (System.Type type, System.Object[] args) [0x00000] in <filename unknown>:0 
  at UnrealBuildTool.RulesAssembly.CreateModuleRules (System.String ModuleName, UnrealBuildTool.TargetInfo Target, UnrealBuildTool.FileReference& ModuleFileName) [0x00000] in <filename unknown>:0 
Command /Users/Shared/Epic Games/UE_4.14/Engine/Build/BatchFiles/Mac/Build.sh failed with exit code 5
```

lovettchris, 2017-04-22:
You need to set the EIGEN_ROOT variable to point at the place where EIGEN bits are installed.  If you ran the airsim build.sh script you should have Eigen installed there.  Notice that build.sh does this:

export EIGEN_ROOT="$(pwd)/eigen"

so use the same terminal window to do your unreal build.

hayaalsh, 2017-04-22:
I just tried building it from the terminal window but I got it couldn't find "Engine.h" file.
```
./MyProject.h:5:10: fatal error: 'Engine.h' file not found
#include "Engine.h"
         ^
1 error generated.
```

hayaalsh, 2017-04-23:
I did use the terminal for the unreal plugin build. But i keep getting that error when I compile the c++ code on xcode. I am so frustrated and do not know what to do.

hayaalsh, 2017-04-23:
I get your point.
So here what I did

(1) On terminal: run ./build.sh
(2) On terminal: export EIGEN_ROOT="/Users/aaa/Github/AirSim/eigen", then checked by "echo $EIGEN_ROOT" and got the right directory.
(3) Copied the plugin folder and did all the required changes on .uproject file.
(4) On xcode: compiled the .xcworkspace file to install the plugins (note in make we do not have the refresh option as in VS)

But still I am getting the same error reported above.

hayaalsh, 2017-04-23:
Update: I have edited AirSim.Build.cs to know what exactly is cause the error, and it turned out that xcode is not recognizing environment variables EIGEN_ROOT.

lovettchris, 2017-04-23:
Does this similar discussion help?

[https://discussions.apple.com/thread/1101536](https://discussions.apple.com/thread/1101536)

hayaalsh, 2017-04-23:
Saddly no even after using the way described bellow:
[](http://stackoverflow.com/questions/42323485/set-environment-variables-on-mac-os-x-sierra

hayaalsh, 2017-04-23:
Sadly no. I end up defining $EIGEN_ROOT as `ENGIN_ROOT = "/Users/aa/Github/AirSim/eigen/"`
But now I get the following error.

```
Setting up Mono
Building MyProject2Editor...
2017-04-23 06:19:15.087 defaults[6083:122225] 
The domain/default pair of (com.apple.dt.Xcode, IDEBuildOperationMaxNumberOfConcurrentCompileTasks) does not exist
Running command : Engine/Binaries/DotNET/UnrealBuildTool.exe MyProject2Editor Mac Development /Users/alsharha/Documents/Unreal Projects/MyProject2/MyProject2.uproject
Compiling with MacOSX SDK 10.12
Parsing headers for MyProject2Editor
  Running UnrealHeaderTool "/Users/alsharha/Documents/Unreal Projects/MyProject2/MyProject2.uproject" "/Users/alsharha/Documents/Unreal Projects/MyProject2/Intermediate/Build/Mac/MyProject2Editor/Development/MyProject2Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
Error: UnrealHeaderTool failed for target 'MyProject2Editor' (platform: Mac, module info: /Users/alsharha/Documents/Unreal Projects/MyProject2/Intermediate/Build/Mac/MyProject2Editor/Development/MyProject2Editor.uhtmanifest, exit code: CrashOrAssert (3)).
Command /Users/Shared/Epic Games/UE_4.15/Engine/Build/BatchFiles/Mac/Build.sh failed with exit code 3
```

hayaalsh, 2017-04-23:
Problem solved :D The problem was solved by going to the RecordingThread.h file located at:
`/Users/alsharha/Documents/Unreal Projects/MyProject/Plugins/AirSim/Source`
and correcting the r letter in the line bellow to match the actual file it referencing.
`#include "SimModeWorldMultirotor.h"` to  `#include "SimModeWorldMultiRotor.h"`

hayaalsh, 2017-04-23:
Now when I run the uproject, I get this error message during start up:
```
Default Property warnings and errors:
Error: CDO Constructor (PIPCamera): Failed to find TextureRenderTarget2D'/AirSim/HUDAssets/FpvRenderTarget.FpvRenderTarget'
```
I have tried to find that directory but I couldn't. All I found is FpvRenderTarget.uasset in the Plugin folder.

lovettchris, 2017-04-24:
You are definitely making progress, thanks for the bug fix in RecordingThread.h, I've [checked that one in ](https://github.com/Microsoft/AirSim/commit/a04c7314fed594966ce719674284b8f17c0eae63).

FpvRenderTarget.FpvRenderTarget is defined inside FpvRenderTarget.uasset.  You can see this if you open that asset in the Unreal Editor.  To see it select Show Plugin Content:
[image]

The open this sources panel:
[image]

Select AirSim Content and HUDAssets, and you should see the FpvRenderTarget.

But are you starting with an empty map (i.e. you are NOT manually placing any assets from AirSim content Blueprints) ?  So the only thing you are placing is the PlayerStart object ?

hayaalsh, 2017-04-24:
Thank you for your help.
I checked the HUDAssets, and apparently FpvRenderTarget.FpvRenderTarget is not there. What could be the source of the problem?

hayaalsh, 2017-04-25:
Did anyone figure a solution for this?

lovettchris, 2017-06-10:
closing since original issue was fixed, please open new issue if you are still seeing problems.

closed 2017-06-10 by lovettchris