microsoft/AirSim #174: How to simulate a controller algorithm ?

MeditationG, 2017-04-24:
If I want to simulate a new controller algorithm such as LQR, LQG  for the drone to tracking a specific trajectory,  How should I start with it ?

AlexandreBorowczyk, 2017-04-27:
Currently AirSim utilize the PX4 flight stack for the drone "brain". Either through the Pixhawk in the hardware-in-the-loop simulation or directly through the Software-in-the-loop simulation. You can more information in the [PX4 Development Guide](https://dev.px4.io/en/).

sytelus, 2017-04-29:
We are also working on running full firmware in the same process as AirSim so you won't need external SITL/HITL or PX4. You can find AirLib/include/controllers/rosflight which is all of ROSFlight firmware running from within AirSim. Things are not fully working yet though although all plumbing is done.

closed 2017-04-29 by sytelus