microsoft/AirSim #175: How to run HelloDrone Program

zywOwO, 2017-04-25:
I am new to unreal engine and c++, so please forgive if this question was asked before.

Following the setup video, I can successfully control the drone under "SimGameMode" and now I want to do programmatical control. Where I should run the HelloDrone program or where should I insert that piece of code?

lovettchris, 2017-04-26:
Easiest way to run HelloDrone is on the same machine as the unreal simulation.  The SimGameMode creates the RpcServer which talks to HelloDrone over TCP sockets.  Be sure to first Go to Unreal Editor, Edit/Editor Preferences, select "All Settings" and type "CPU" in the sarch box. It should find the setting titled "Use Less CPU when in Background", and you want to uncheck this checkbox, as per our [FAQ](https://github.com/Microsoft/AirSim/blob/master/docs/faq.md).

zywOwO, 2017-04-28:
I still have a problem connecting the hellodrone.cpp with the unreal project.
I tried to run the .exe file and it prints "Waiting for drone to get a GPS position...".
Then I clicked play in unreal editor. It fails to control the drone by program.
What did I do wrong?

clovett, 2017-04-28:
Try starting unreal "play" before running hellodrone.

closed 2017-04-29 by sytelus

SaiVineethJ, 2018-02-02:
@zywOwO   How did you run the code??
I am getting this error while compiling main.cpp: 
"fatal error: common/common_utils/StrictMode.hpp: No such file or directory
 #include "common/common_utils/StrictMode.hpp""

How to remove this error or is there any other way of running the code??