microsoft/AirSim #177: ./build.sh fatal error <atomic>

spencerh-b, 2017-04-26:
I have followed all of the instructions, on the readme, but I'm getting stuck on the ./build.sh
Running on Ubuntu 14.04

`$ ./build.sh 
putting build in build_debug folder, to clean, just delete the directory...
~/AirSim/build_debug ~/AirSim
Scanning dependencies of target AirSim-rpclib
[  1%] Building CXX object rpclib/CMakeFiles/AirSim-rpclib.dir/lib/rpc/dispatcher.cc.o
In file included from /home/spencer/AirSim/external/rpclib/lib/rpc/dispatcher.cc:1:
/home/spencer/AirSim/external/rpclib/include/rpc/dispatcher.h:6:10: fatal error: 
      'atomic' file not found
#include <atomic>
         ^
1 error generated.
make[2]: *** [rpclib/CMakeFiles/AirSim-rpclib.dir/lib/rpc/dispatcher.cc.o] Error 1
make[1]: *** [rpclib/CMakeFiles/AirSim-rpclib.dir/all] Error 2
make: *** [all] Error 2`

Any ideas as to why this is occuring?

lovettchris, 2017-04-26:
You might be missing libc++.  Try running cmake/getlibcxx.sh.

spencerh-b, 2017-04-26:
@lovettchris is it necessary to have cmake 3.5.1 or do later versions work?

lovettchris, 2017-04-26:
newer versions are fine.

sytelus, 2017-04-29:
I got same error when I ignored executing `./cmake/getlibcxx.sh` instructions in [Linux build](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md) doc. Above error basically means C++ compiler can't find std library where it can find things like <atomic>. To fix it, delete llvm* folders and then follow above doc for Linux build. Make sure the command `./cmake/getlibcxx.sh` doesn't end with warnings or errors.

closed 2017-04-29 by sytelus