microsoft/AirSim #179: AirSim Demo Crashed when starting Landscape Mountains

troiwilliams, 2017-04-26:
Hi,

I installed the AirSim prereqs and I am currently on the following link (https://github.com/Microsoft/AirSim/blob/master/docs/unreal_demo.md) where I need to open the "LandscapeMountains.uproject" in the Unity Editor. But whenever I try to open the Landscape Mountains demo, it crashes as soon as it finishes loading and gives me the following error notification, which contains no error information (image below).

Any ideas on what is causing the project to crash and how I can fix it?

[image]

lovettchris, 2017-04-26:
*Unity Editor* I think you mean *Unreal Editor*, but don't know, perhaps something got corrupted in the download, try downloading again and/or reinstalling Unreal Engine 4.15.  Which operating system are you using?

troiwilliams, 2017-04-26:
Thanks for the response. I am using Windows 10. And, I will reinstall Unreal Engine again and see what happens.

lovettchris, 2017-04-26:
Sounds good.  I'll close the issue for now and cross my fingers for you :-) If that still doesn't work, I also copied my working version of Landscape Mountains to [my onedrive](https://1drv.ms/f/s!AoS-ymEfLuCLo-Vl5wVsj_yv7pcCuA).

closed 2017-04-26 by lovettchris

troiwilliams, 2017-04-26:
I reinstalled _Unreal Engine 4.15.1_ (I got it correct this time 😄) and also copied your LandscapeMountains project into my Unreal Engine Projects directory, but the program still crashes.

lovettchris, 2017-04-26:
Can you get a call stack of the crash using Visual Studio ?

troiwilliams, 2017-04-27:
How can I retrieve a call stack of the crash using Visual Studio?

lovettchris, 2017-04-27:
I use Unreal to generate a visual studio project file:
[image]

Then load it into VS and select the build configuration called "DebugGame Editor":
[image]

Then I hit F5 to run AirSim in the debugger.  This launches the Unreal Editor so I press Play and I'm now "debugging" the running game.  When it crashes VS will show the call stack in the Callstack tool window.  This is also handy way to run the game because you can put breakpoints anywhere and debug any issues that you might be having...

troiwilliams, 2017-05-05:
I did all the steps mentioned above and Unreal Engine has not crashed since I launched it using debugger mode. So, for now it appears okay. But if it does crash in the future, I will let you know.

Thanks so much for the help!

lovettchris, 2017-05-05:
Great to hear.  Thanks for the update.  We have also fixed the crash on Windows so you no longer need to use debug mode.  But since you have it working, keep doing what you are doing :-)

troiwilliams, 2017-05-05:
So, I spoke too soon. :-/ It crashed. Here is the output from Visual Studio: [vs-airsim-crash.txt](https://github.com/Microsoft/AirSim/files/980011/vs-airsim-crash.txt)